MarlinFirmware/Marlin #1911: cant disable zmax endstop for enabling Z2min

kagouraki, 2015-04-14:
hello i want to disable the zmax endstop and in its position to have the Z2 min endstop. so i will have for X min and max, for Y min and max and for the Z i will have two min endstops. i have ramps 1.4 EFB. When i try to verify the folowing error apears : 

Arduino: 1.6.3 (Mac OS X), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

stepper.cpp: In function 'void __vector_17()':
stepper.cpp:166:15: error: 'old_z_max_endstop' was not declared in this scope
         if (!(old_z_max_endstop && (count_direction[Z_AXIS] > 0)) && !locked_z_motor) Z_STEP_WRITE(v); \
               ^
stepper.cpp:701:13: note: in expansion of macro 'Z_APPLY_STEP'
             AXIS ##_APPLY_STEP(!INVERT_## AXIS ##_STEP_PIN,0); \
             ^
stepper.cpp:709:9: note: in expansion of macro 'APPLY_MOVEMENT'
         APPLY_MOVEMENT(z, Z);
         ^
stepper.cpp:166:15: error: 'old_z_max_endstop' was not declared in this scope
         if (!(old_z_max_endstop && (count_direction[Z_AXIS] > 0)) && !locked_z_motor) Z_STEP_WRITE(v); \
               ^
stepper.cpp:704:13: note: in expansion of macro 'Z_APPLY_STEP'
             AXIS ##_APPLY_STEP(INVERT_## AXIS ##_STEP_PIN,0); \
             ^
stepper.cpp:709:9: note: in expansion of macro 'APPLY_MOVEMENT'
         APPLY_MOVEMENT(z, Z);
         ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

in my adv config file i have the folowing : 

#ifdef Z_DUAL_STEPPER_DRIVERS

  #define Z_DUAL_ENDSTOPS

  #ifdef Z_DUAL_ENDSTOPS
    #define Z2_STEP_PIN E2_STEP_PIN           // Stepper to be used to Z2 axis.
    #define Z2_DIR_PIN E2_DIR_PIN
    #define Z2_ENABLE_PIN E2_ENABLE_PIN
    #define Z2_MIN_PIN 19                     //Endstop used for Z2 axis. In this case I'm using XMAX in a Rumba Board (pin 36)
    const bool Z2_MIN_ENDSTOP_INVERTING = false;
   // #define DISABLE_ZMAX_ENDSTOP              //Better to disable the XMAX to avoid conflict. Just rename "XMAX_ENDSTOP" by the endstop you are using for Z2 axis.
  #endif

#endif // Z_DUAL_STEPPER_DRIVERS

I have done this by disabling the Xmax and it compiles fine.

thinkyhead, 2015-04-14:
I believe the fix in #1912 should allow you to compile, and hopefully get the behavior you need.

kagouraki, 2015-04-14:
ok it compiles now. I need to see if it behavies as expected. Thank you

closed 2015-09-25 by AnHardt