MarlinFirmware/Marlin #1918: Google hangout time

boelle, 2015-04-15:
Name             Zone    GMT-7 Offset        GMT-7 awake        thinkyhead overlap       online weekdays      online weekend
@thinkyhead      GMT-7                       9am -  1am         9am -  1am
@clefranc        GMT-4   + 3 hours           6am - 10pm         9am - 10pm
@alexborro       GMT-3   + 4 hours           5am -  9pm         9am -  9pm
@fmalpartida     GMT-1   + 6 hours           3am -  7pm         9am -  7pm
@AnHardt         GMT+1   + 8 hours           1am -  5pm         9am -  5pm
@Wurstnase       GMT+1   + 8 hours           1am -  5pm         9am -  5pm
@boelle          GMT+2   + 9 hours          12am -  4pm         9am -  4pm               1pm - 8pm            ~8am - 10pm
@emartinez167    GMT+8   +15 hours = -9      6pm - 10am         6pm -  1am, 9-10am
@ntoff           GMT+10  +17 hours = -7      4pm - 12pm         4pm -  1am, 9am-12pm

https://plus.google.com/hangouts/_/g2wp5duzb2y6ahikg6tmwao3kua

thinkyhead, 2015-04-16:
I propose we try for this weekend. We should decide on some topics. In general, "the future of Marlin."

boelle, 2015-04-16:
i'm in..... time no matter, anytime friday after 15:00 and until sunday
22:00

maybe also share a google doc with topics and space for how to carry out
things

msutas, 2015-04-16:
Hey Collaborators,

Rather than deciding on the future of Marlin, I believe it would be better if you just define the possible alternatives for future roadmap and ask the community through a series of online polls which to go over and decide afterwards.

scotty1024, 2015-04-16:
Mehmat, if the community wants something they are free to use any of several bounty sites.

Quite frankly this is probably the most effective way to get a specific goal accomplished on an open source project.

msutas, 2015-04-16:
@scotty1024, Maybe because of my poor language skills, you have
misunderstood me. I am also supporting the use of funds for further
development of this project. What I was asking for was the project leaders
to determine some development alternatives and then listen to the community
for deciding on the path. This was not related with any finance issue
rather issues like support for cnc mills etc and these items priority.

boelle, 2015-04-17:
https://docs.google.com/document/d/1GmKjDVOKLj6f6s76ZL99LvCPOwy_PkYK57M88Q2oNdM/edit?usp=sharing

that is link to googledoc with topics....

@thinkyhead you have a G account i can add to it for edit rights?

everyone else have comment rights to it

boelle, 2015-04-18:
is it going to happen ?

scotty1024, 2015-04-18:
The only thing harder than getting bugs fixed is getting people to whom all say "I don't care what time it is" to actually agree to a time. :-)

My suggestion would be to add a fun hangout topic to the agenda so it isn't just all hair shirt topics. Perhaps https://www.kickstarter.com/projects/wr3d/the-diamond-hotend-single-nozzle-multi-color-3d-pr?ref=discovery

boelle, 2015-04-18:
just five me a shout on twitter and i will be there

boelle, 2015-04-18:
what in the link should we look at? i fail to catch it

scotty1024, 2015-04-18:
Check FAQ under clogging? Why is it we always need PTFE for inexpensive hot ends? Is there anything we can do in Marlin to make all metal work better? Are these guys mis-representing their project by not mentioning on their main page that you will no longer be getting that v6 E3D you're looking at in that top most picture?
And no discount for them using the lite6 which costs less than half the v6? Should Marlin have configuration switches to make it easy to configure for hot ends like this, sort of like it does for LCD panels?

Seems like several different conversations possible to me. :-)

scotty1024, 2015-04-18:
Bo,

If you want this to happen, and you are ok with being the person doing this going forward.. Add a fun topic and pick a time eg show leadership.

I've suggested a fun topic, you've seen a nibble from Roxy on that topic. I suggest 1am GMT this Sunday. That's 6PM today in pacific daylight time.

See who shows, wash spin repeat. That's how it gets going.

boelle, 2015-04-18:
I want to...

but 1am is to late for a Sunday, i have to get up for work 8 hours later

What about next Saturday same time?

scotty1024, 2015-04-18:
I think you're better off with weeks advance notice.

If you want a marlinfirmware.org login so you can promote your hang out, you known whom to ask. :-)

nophead, 2015-04-18:
I think what makes all metal hot ends expensive is the need to polish the
bore to stop it jamming. I solved that by simply adding a taper at the
transition zone, so any forward motion separates the plug from wall. See
http://hydraraptor.blogspot.co.uk/2009/02/if-at-first-you-dont-succed.html.
No fancy machining and it worked fine with PLA.

scotty1024, 2015-04-18:
Not to get off topic but I believe the Rep Rap Discount Hexagon has this. The broken PID in Marlin keeps cooking PETG and Taulman and jamming my prints. All I have to do to clear it is manually push it on through. I get a drool of burnt black filament on the bed to clean off. Then restart the print from scratch.

boelle, 2015-04-19:
Next attempt will be 26/4-2015 1:00 GMT

Will try and thinker with some fun topics

scotty1024, 2015-04-19:
OK here's another way to make all metal e pensive: gold plated barrel to reduce friction?

http://deezmaker.com/store/#!/Spitfire-Extruder-Kit-&-Upgrade/p/35538131

I'd do a hangout if Bo can get them to come on and talk about that!

boelle, 2015-04-19:
that one and the diamond are on as subjects at least....

but fire away with ideas.... me printing parts for extra extruders...

nophead, 2015-04-19:
Gold plate the barrel and use a real diamond nozzle to allow abrasive
filaments. Extruder bling!

boelle, 2015-04-19:
and mount on a Mendel90.... 3D print-gasm :-P

boelle, 2015-04-21:
i can be on the hangout in about 3 hours and most of tomorrow

thinkyhead, 2015-04-25:
Good to have a time, I will set an alarm!

boelle, 2015-04-25:
@thinkyhead how can i give you edit rights to the google doc? have an G account i can add?

boelle, 2015-04-25:
@everyone who will be there?

AnHardt, 2015-04-25:
I'll try. But 3:00 in the morning? Be loud - i'll sleep in front of my desk.

boelle, 2015-04-25:
depends where you are? for me its 2:00

AnHardt, 2015-04-25:
No DLS (Summertime) in Denmark? I'm IN at the moment.

boelle, 2015-04-25:
Yep summertime

Where in DK are you?

boelle, 2015-04-25:
http://www.timeanddate.com/worldclock/

london right now is 14.22 and DK 15.22... so tonight when london is 1:00 dk should be 2:00 right?

AnHardt, 2015-04-25:
Mh, yes. But London has also DLS. So - as far i understand they are 1 hour different to GMT. Or am I confused wit GMT and UTC? I'm IN the Hangout - not Denmark. I'm located in the Ruhr-District Germany close to Dortmund.

boelle, 2015-04-25:
I think you mix UTC (which does not have DLS - its the same year round) and GMT (which has summertime) that is why GMT is one hour ahead of UTC during summer, but in winter GMT and UTC are the same. 

The Most ideal time to use is UTC as its the same no matter where you are and what date

AnHardt, 2015-04-25:
Please have a look at: http://www.timeanddate.com/worldclock/uk/london

AnHardt, 2015-04-25:
BST != GMT

boelle, 2015-04-25:
yes i see... but hmm....

boelle, 2015-04-25:
one thing is for sure.... they are wrong about UTC/GMT..... UTC and GMT is not the same....

AnHardt, 2015-04-25:
Ok. Lets remember: "Next time we will make our date using UTC."

boelle, 2015-04-25:
Yep.... i can agree to that... :-D

boelle, 2015-04-25:
to make the confusion better: http://coloringpagesjos.net/wp-content/uploads/2015/192542-gmt-time-zone-map-world.jpg

GMT and UTC is the same.... GMT is outdated :-P

boelle, 2015-04-25:
the text in the middle near buttom.... formerly GMT....

boelle, 2015-04-25:
i will correct all the times and clean up the list now that we have figured UTC is the most sure way

boelle, 2015-04-25:
@fmalpartida where are you located? UTC -1 seems strange... portugal?

AnHardt, 2015-04-25:
Ok. So You, @Wurstnase and me are at the moment UTC + 2.
To make things perfect use 1:00 am in the title.

boelle, 2015-04-25:
need to clean up arround me.... we are getting kitchen renovated next week so i a lot of crap arround

boelle, 2015-04-25:
@AnHardt what do you think about the topics i have put in the google doc so far?

fmalpartida, 2015-04-25:
@boelle, I amat GMT+1 DLS in Spain.

boelle, 2015-04-25:
hehe... makes much more sense now :-D

fmalpartida, 2015-04-25:
Has the hangout happen or is it being planned for this evening?

boelle, 2015-04-25:
for this evening.... 1am...

boelle, 2015-04-25:
ohh.

boelle, 2015-04-25:
2 am for us euro people

fmalpartida, 2015-04-25:
Just had a quick chat with Andreas and talk to to folks this evening.

boelle, 2015-04-25:
me just cleaning up table here and getting my usb microscope adjusted so it can be used as webcam...

AnHardt, 2015-04-25:
@boelle 
Your topics are ok. But maybe we'll take a bit of time to introduce ourselves.

fmalpartida, 2015-04-25:
Yes! By the way, here is 16:36, uct+1 + dls, so the call at 11 europeant time, right?

fmalpartida, 2015-04-25:
Topics look good indeed.

AnHardt, 2015-04-25:
@boelle 
Don't clean up to much. I will show only the nice Chinese, font i have got this afternoon, on a display. Else i had to clean up. :-)
Try to play with Google Effects. The 'background feature' is great to hide everything - including your selves.

boelle, 2015-04-25:
the hangout is planned for 1:00am UTC.... so it should be 2am CET (central european time)

boelle, 2015-04-25:
will be here after dinner... 2 hours approx

need to print some spare part for my printer anyway

nophead, 2015-04-25:
I think GMT is still correct when referring to time zones.
http://www.timeanddate.com/time/gmt-utc-time.html

[15 more replies]

closed 2015-05-07 by boelle