MarlinFirmware/Marlin #1920: MINTEMP error, even though reports show normal thermistors values

dattas, 2015-04-15:
Arduino 1.6.3, RAMBO, single extruder and heated bed. Commit tested: d0b1f70118104b3bf45bba4ba7edce1ec803c1dd

I've tested the thermistors, and the display I have connected shows normal temperatures (around 22 celsius) for both the extruder and the heated bed. If I revert to older commits (such as f47fc879fe47c3597ad47cb4026888c028c1c55e ) this issue doesn't exist. I've also tried decreasing and increasing the min temp in the configuration but the issue persists.

AnHardt, 2015-04-15:
Please check if all you unused TEMP_SENSORs are at 0 and check if EXTRUDERS  is 1.

dattas, 2015-04-15:
@AnHardt EXTRUDERS is for sure 1, one of the unused temp_sensors may have been at a non-zero value, but that was when I was playing with a dual extrusion (and I had this error with dual extrusion which is why I also tried to simplify it to ensure it wasn't a dual extrusion problem). I'll try again in a few hours (after work) and ensure the configuration for the TEMP_SENSORs for the unused is 0. I would expect Marlin to ignore anything after the first non zero TEMP_SENSOR if EXTRUDERS is 1.

AnHardt, 2015-04-15:
The use of EXTRUDERS shoud be avoided where we can. Ideally we should remove it at all.
EXRUDERS is neither a valid number for involved sensors nor steppers nor heaters. An extruder can be a system of: n sensors, m temperatures, o heaters, p fans and q motors - and likely i forgot something. All, N to Q = 1 is more a common exception than the norm.

thinkyhead, 2015-04-16:
@AnHardt There is a pending PR that replaces `EXTRUDERS` with `HOTENDS` but it's pretty stale now.

Ultimately more indirection will help. An Extruder object should be defined that has as its traits: the number of hot-ends it has, the number of motors it has, the number of sensors it has (and what types), etc. A HotEnd will be another object, with a heater and a sensor...

AnHardt, 2015-04-16:
@thinkyhead 
:+1:  that will be a mayor step in the right direction. With a accentuated mayor - so after bugfix.

A 'temporary' fix here cold be, to make the HAS_TEMP_x dependent on EXTRUDERS <= x in 'Conditionals.h'. Or maybe a error at compile-time like 'You defined more temperatures than extruders.' in 'SanityCheck.h' But I'm not sure if we will break an other valid configuration with this.

By the way. Where we are at mayor changes in 'temperature.cpp'.
I feel that the bed is a normal heater like the others and noes not merit an extra handling. This could save a lot of code and exceptions.

AnHardt, 2015-04-16:
@thinkyhead 
More an uncaught bad configuration than a bug.

I have seen the MINTEMP error for a combination of:
#define MOTHERBOARD BOARD_RAMPS_13_EEB
#define EXTRUDERS 1
#define TEMP_SENSOR_1 1

For  RAMBO i expect the pis for TEMP_SENSOR_1  as defined like in EEB.

thinkyhead, 2015-04-16:
@AnHardt I agree that all the array variables that include heaters and thermistors for extruders, but which keep the bed as a separate variable… They all should be changed to include the bed as the last element of the array.

AnHardt, 2015-04-28:
@dattas 
We'll keep an eye on it when restructuring the temperature code.

closed 2015-04-28 by AnHardt