MarlinFirmware/Marlin #1921: Acceleration fails to reach plateaux speed by the end of the ramp

nophead, 2015-04-15:
The acceleration in Marlin seems to be broken at high speeds. I have a logic analyser hooked up to the E step signal. It is executing G1 E1 F1800 with 565.08 steps per mm, jerk 10 and acceleration 5000. It should accelerate smoothly from 5mm/s to 30mm/s in 5ms. It actually accelerates from 5.8mm/s to 16.7mm/s in 7.9ms and then jumps to double stepping at 29.74mm/s. 

[image]

This shows the point where it jumps and the timing markers show the last single step period. It then starts double stepping at the nearly twice the speed. The deceleration looks roughly the same. It should actually accelerate much faster in single step mode and then carry on accelerating in double step mode.

The comments in the code say that it uses the Leib ramp method but I don't think it does. The code looks much simpler. I think the bug is that the acceleration rate is wrong so at the end of the ramp it is not going the speed it thinks it is but then it switches to the constant speed plateaux, which is the correct speed, hence the jump.

Wurstnase, 2015-04-15:
Can you replace https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L833
```block->acceleration_rate = (long)(acc_st * 16777216.0 / (F_CPU / 8.0));```
by
 ```block->acceleration_rate = (long)(acc_st * (16777216.0 / (F_CPU / 8.0)));```

and test it again?

nophead, 2015-04-15:
I think they both produce the same result because all the operands are floats, so the order makes very little difference. My best guess is this line goes wrong https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/stepper.cpp#L717 when acceleration and speed are both high.

Wurstnase, 2015-04-15:
If it's that you could verify it with ```#define MultiU24X24toH16(intRes, longIn1, longIn2) intRes = ((uint64_t)(longIn1) * (longIn2)) >> 24```.

Wurstnase, 2015-04-16:
Well, I think I got it.

acc_st should not be bigger than 2.000.000. So the max acceleration for 565 is ~3500. If you are bigger than this the number is > 24bit.

In planner.cpp you will find this: 
```
block->acceleration_rate = (long)(acc_st * 16777216.0 / (F_CPU / 8.0));
```

acc_st is acceleration * step_rate. The acceleration_rate should be smaller than 2^24 and when you restructure the equation you get F_CPU/8 = acc_st.

Wurstnase, 2015-04-16:
Another workaround:

Replace the 16777216 with 2097152 and make under the MultiU24X24toH16 a acc_step_rate <<= 3. Then you can go up to 28000 acc with 565 steps/mm.

nophead, 2015-04-16:
Yes but that would loose some accuracy of course. The planner assumes that after the number of steps it has calculated have elapsed the speed will equal the plateaux speed. The stepper keeps calculating v = u + a.t each ramp step, calculating t by accumulating the timer periods. It then calculates the next timer period from the speed using a table to avoid division. Any error at the end causes a jump in speed when it switches to the plateaux after a fixed number of steps, regardless of whether it got there or not.

This doesn't have anything to do with the Leib ramp algorithm at all, which uses a Taylor series to calculate the next period from the previous one using multiplies and adds.

I think a better fix is to change MultiU24X24toH16 to be MultU32x24toH16 by adding two extra multiply instructions. I will have a go at that as soon as I have deciphered gcc avr assembler. 

Another issue is the fact that when double and quad stepping I think it can overshoot the end of the constant speed part by up to three steps and thus start the deceleration late, so not reach the final speed. Not sure how much that matters yet.

Wurstnase, 2015-04-16:
> Any error at the end causes a jump in speed when it switches to the plateaux after a fixed number of steps, regardless of whether it got there or not.

No it can't jump, because the speed is limited by plateaux-speed.

> Another issue is the fact that when double and quad stepping I think it can overshoot the end of the constant speed part by up to three steps and thus start the deceleration late, so not reach the final speed.

When it's to late for deceleration it just missed the last 3 steps for the speed, which should not a big problem. We use jerk, which is similar to that step of speed. Or do I'm missing something or missunderstand you?

Anyway, U21x24toH16 should be realy good enough. The precision is not that important as it seems. Other firmwares calculates the speed only every 1ms or less. Marlin is much more accurate in that part.

nophead, 2015-04-16:
Yes it is limited to the plateaux speed but the effect of dropping bits without rounding is to make the final speed too low, so it will jump to plateaux speed. Not good to have jerks at high speed as that is when torque is low.

On the deceleration it will violate the jerk setting by not quite reaching the jerk speed before it stops or not quite reaching the planners target junction speed.

The overall precision is not that important but the fact that the planner calculates the acceleration steps using accurate maths, assuming perfect acceleration, and then the stepper uses different maths and approximations means there is discontinuity at the end of each ramp.

Looking at MultiU24X24toH16 is seems to have one 8x8 multiply missing as I count 8 and it takes 9.

Wurstnase, 2015-04-16:
Yes, you could be right. I'm just thinking loud.

> Looking at MultiU24X24toH16 is seems to have one 8x8 multiply missing as I count 8 and it takes 9.

Maybe this is R27? But I'm far away from understanding asm.

nophead, 2015-04-16:
I have fully decoded the asm.

A full multiplication of 24x24 would produce 48 bits. The bottom 24 are discarded when it does the implicit divide by the 16777216 scaling factor. The top 8 are not needed because it only returns 16 bits.

MultiU24X24toH16 doesn't bother calculating the two LSBs or the MSB of the 48 bits. Not only does it not multiply the two LBSs it actually throws away the bottom 8 bits of the MSB1 x LSB2 and MSB2 x LSB1 products. It keeps the fractional 3rd byte of the answer in r27 and uses it to round to nearest the two byte result. However it is mathematically incorrect because the lower order multiplications can cause caries that would affect the upper bytes if it bothered to calculate and sum them. However, the returned result will only be out by one at the most I think.

To make it do 32x24 it only needs two more MULs and three adds. I think that adds 7 cycles, which I assume is 7/16 uS. I will give it a go.

closed 2015-04-18 by thinkyhead

Wurstnase, 2015-04-20:
Hi @nophead 
I try to understand the asm-part. Maybe you can told me if I'm right.

We have now a 32bit (A) and a 24bit (B) value.

Bit 4 | Bit 3 | Bit 2 | Bit 1
------|------|------|------
A4 | A3 | A2 | A1
- | B3 | B2 | B1

to solve our equation we need

Bit 7 | Bit 6 | Bit 5 | Bit 4 | Bit 3 | Bit 2 | Bit 1
----|----|----|----|----|----|----|----
MSB | A4*B3
 |MSB|A3*B3|
 |MSB|A4*B2|
 | |MSB|A2*B3|
 | |MSB|A3*B2|
 | |MSB|A4*B1|

The rest is not required. The result is Bit 7 and Bit 6. Am I right?

nophead, 2015-04-20:
No, completely wrong register notation, see the comments I added. 
```
// B0 A0 are bits 24-39 and are the returned value
// C1 B1 A1 is longIn1
// D2 C2 B2 A2 is longIn2
```

The multiplies are done in this order:

         | B0      | A0    | r27   |         |       |
--------| --------| -------|-------|--------|-------|
47-40 | 39-32 | 31-24| 23-16 | 15-8 | 7-0 |
         |          |          |  MSB  |A1*B2 |  |
         | MSB  |B1*C2|           |           | |
MSB |C1*C2|           |          |          | |
        |MSB   |C1*B2|           |           | |
        |          |MSB  | A1*C2 |           | |
        |          |MSB  | B1*B2 |           | |
        |          |MSB  | C1*A2 |           | |
        |          |         | MSB    |B1*A2 | |
        | MSB  |D1*A2|          |           | |
MSB |D1*B2|          |          |           | |

The last two multiplies added by me. 

* Bits 7-0 are never generated.
* Bits 47-40 and 15-8 are discarded as soon as they are calculated. 
* Bits 39-16 are accumulated as a 24 bit number and then bit 23 is used to round 39-24, which is the two byte final result.

Wurstnase, 2015-04-20:
Well, this are my first steps into asm. Thanks!

nophead, 2015-04-20:
It was my first step into AVR asm as well but I first learned assembler code nearly 40 years ago on an INS8060. That hardest part was working out gcc's weird system for allocating 8 bit register names to parts of larger arguments.

Wurstnase, 2015-04-20:
I'm searching for the same thing of arm-asm. But didn't find anything about it. 
My first steps are actually to build some code on my raspi2. Python is much simpler ;)

Maybe it's not necessary on a 32bit system, but my hobby is the arm-port. So I'm looking to port as much as possible.

nophead, 2015-04-20:
A 32 bit system should be able to do with a single multiply, a rounding addition, and a shift. Probably no point in writing in assembler. Just
```
intRes = ((uint64_t)longIn1 * longIn2 + 0x800000ULL) >> 24;
```

Wurstnase, 2015-04-20:
Ok, I will add that rounding. How big should the acceleration_rate should be?

block->acceleration_rate = (long)(acc_st * 16777216.0 / (F_CPU / 8.0));

On the arm-port I replaced F_CPU/8 (2M) to HAL_TIMER_RATE (42M). I could extend the 2**24 to maybe 2**28 to be close to the avr-values.

nophead, 2015-04-20:
The F_CPU/8 term is to convert the accumulated timer values in acceleration_time to seconds because acc_st is in steps per second^2. So yes it should equal your timer rate. It avoid a division when doing v = u + at where t is in timer units.

The 16777216.0 is to allow a 24 bit fraction which is later compensated for by the 24 bit shift. Making the shift bigger gets your more precision but 24 bits is about as accurate as a float.

The size of the result needs to be big enough to hold the biggest change in step frequency during the acceleration. In the AVR case step frequency is capped at 40kHz, which fits in an unsigned 16 bit int.

If your maximum step rate needs 3 bytes you should be able to have up to 5 bytes of fraction on a 32 bit system. I think I would just use 2^32 as the scaling factor as a clever compiler or some hand assembler can then do a 32x32=64 multiply, add 0x80000000 with a double length add and then just return the most significant 32 bits without needing a shift.

Wurstnase, 2015-04-20:
> If your maximum step rate needs 3 bytes you should be able to have up to 5 bytes of fraction on a 32 bit system.

What did you mean with the 5 bytes of fraction? Can't follow you.

The asm-part should be really easy. umul (32x32->64) + add and that's it.

nophead, 2015-04-20:
Because if your multiply generates a 8 bytes and you need a 3 byte result you can shift away 5 bytes. However I was neglecting the fact that the acceleration rate needs to fit in 32 bits so the maximum shift is determined by your max acceleration, steps per mm and the timer rate.

Wurstnase, 2015-04-20:
Ok. So if I understand this:
timer rate is fixed to 42M
acc * step_mm / timer_rate * 2^xx should be fit in 2^32.
While acc * step_mm should anytime below 42M xx could be 32.

So I'm save to use 2^32 instead of 2^24 and can use a 32bit mul with the 0x80000000 round.

nophead, 2015-04-20:
Yes that looks right.

Wurstnase, 2015-04-20:
:+1:  thanks.

Wurstnase, 2015-04-20:
Sry, I'm back. I tried some stuff and stock.

Why is
```
#define MyMul(res, val_a, val_b) res = ((uint64_t)val_a * val_b + 0x80000000) >> 32
```
this ~3 times faster than
```
  #define MyASM(res, val_a, val_b) \
    asm volatile ( \
      "eor %0, %0 \n\t" \
      "mov r0, #0x80000000 \n\t" \
      "umlal r0, %0, %1, %2" \
      : \
      "=&r" (res) \
      : \
      "r" (val_a), \
      "r" (val_b) \
    )
```

When i compile this with gcc on my raspi, the asm-part is 2 times faster. On the Due the asm-part is 3 times slower...

nophead, 2015-04-21:
Compilers often optimise out loops that don't do anything. Declaring result and value_a volatile should force it to evaluate the function.

To find out why it is slower you need to look at the code generated by the compiler. In addition to the asm lines you write it might add code to save clobbered registers and code to get the arguments into registers, etc.

Wurstnase, 2015-04-21:
So it could be faster when I 'mov' them manually? I will check this.

PR #1926 (2015-04-18): Fixed jump in speed at high accelerations on axes with lots of steps
I.e., when acceleration * steps per mm > 2,000,000.
This was done by changing MultiU24X24toH16 to take a 32 bit operand.
Removed the claim that stepper.cpp uses the Leib algorithm.

Fixes #1921