MarlinFirmware/Marlin #1929: max_e_jerk is ignored for E only moves if max_xy_jerk is less

nophead, 2015-04-17:
This line https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L877 assumes max_xy_jerk is relevant regardless of whether the move involves those axes. The consequence is the extruder jerk used for retract can not be bigger than the xy jerk. This is a limitation because generally the extruder motor sees far less inertia than the XY axis motors.

thinkyhead, 2015-04-25:
@nophead Chris, I'll just add some comments here, and perhaps you can point out in more detail what the problem is. I believe I understand the issue, but I'm not sure how to detect an e-only move within this block.
```
  float csz = current_speed[Z_AXIS],
        cse = current_speed[E_AXIS],
        vmax_junction_factor = 1.0,
        mz2 = max_z_jerk / 2, me2 = max_e_jerk / 2,
        vmax_junction = max_xy_jerk / 2; // Start with a safe speed assumption

  // If Z speed is over half the z jerk, limit speed to half the z jerk
  if (fabs(csz) > mz2) vmax_junction = min(vmax_junction, mz2);

  // If E speed is over half the e jerk, limit speed to half the e jerk
  if (fabs(cse) > me2) vmax_junction = min(vmax_junction, me2);

  // Limit speed to the block nominal_speed
  vmax_junction = min(vmax_junction, block->nominal_speed);

  // A safe speed to use for the exit_factor (up to the block nominal_speed)
  float safe_speed = vmax_junction;

  // If moves are queued and we're already moving...
  if ((moves_queued > 1) && (previous_nominal_speed > 0.0001)) {
    float dx = current_speed[X_AXIS] - previous_speed[X_AXIS], // difference in X speed
          dy = current_speed[Y_AXIS] - previous_speed[Y_AXIS], // difference in Y speed
          dz = fabs(csz - previous_speed[Z_AXIS]), // difference in Z speed
          de = fabs(cse - previous_speed[E_AXIS]), // difference in E speed
          jerk = sqrt(dx * dx + dy * dy); // The magnitude of the difference in XY speed

    // Use the block nominal_speed
    vmax_junction = block->nominal_speed;

    // Limit to the max xy jerk, if needed
    if (jerk > max_xy_jerk) vmax_junction_factor = max_xy_jerk / jerk;
    // Limit for the max z jerk, if needed
    if (dz > max_z_jerk) vmax_junction_factor = min(vmax_junction_factor, max_z_jerk / dz);
    // Limit for the max e jerk, if needed
    if (de > max_e_jerk) vmax_junction_factor = min(vmax_junction_factor, max_e_jerk / de);

    // The new speed can be no more than the previous nominal speed
    vmax_junction = min(previous_nominal_speed, vmax_junction * vmax_junction_factor);
  }
  block->max_entry_speed = vmax_junction;
```

thinkyhead, 2015-04-25:
I assume it's the initializer that needs to be changed – `vmax_junction = max_xy_jerk / 2;`

nophead, 2015-04-25:
I did this to the version I have:
```
float vmax_junction = max(max_xy_jerk, max(max_z_jerk, max_e_jerk))/2;
  if(fabs(current_speed[X_AXIS]) > max_xy_jerk/2 || fabs(current_speed[Y_AXIS]) > max_xy_jerk/2)
	vmax_junction = min(vmax_junction, max_xy_jerk/2);
```
to make it symmetrical. However the code doesn't feel correct to me because I don't see how vmax_junction and safe_speed can represent the max_jerk of multiple axes without storing which axis it refers to. I also don't get the divide by 2. Why does the allowable jerk double when there are buffered moves?

The code inside if(moves_queued ...) makes more sense to me and I don't understand why that isn't used all the time. That compares the jerk of each axis to its own jerk limit and if it is exceeded slows works out how much to slow all the axes to keep each one within its jerk and finds the slow down factor that is greatest. That seems to be the correct logic and should at apply at any time.

I also moved float vmax_junction_factor = 1.0; inside the if(moves_queued ... block as that is only place it is used.

thinkyhead, 2015-04-25:
@nophead I think your code is correct, and will merge it into the codebase.

As for the divide-by-two stuff, that is mysterious, but maybe it makes some sense. The first calculation is only for the `exit_factor` parameter (unless there are no new moves, in which case it also applies to the `max_entry_speed`). So, in the first calculation it looks at the current speed, and then only applies one jerk limit: half of the smallest limit from only the moving axes.

Looking at `calculate_trapezoid_for_block` these inputs are clamped, so `nominal_rate * factor` is always >= 120.

I'll have to study `calculate_trapezoid_for_block` some more. I'm not sure I understand how the entry and exit speeds are applied, and their relationship to jerk. Jerk is the maximum non-accelerated change in speed, and so I can see why it applies at the beginning and end of accelerated moves. It seems to also imply that at every junction between accelerated moves there are always non-accelerated "jerky" moves. So the idea is to make sure those junctions meet up where the end speed (of each axis) is as close to the start speed as possible, or at least under the jerk limits…

Must continue to contemplate this stuff….

nophead, 2015-04-25:
>So, in the first calculation it looks at the current speed, and then only applies one jerk limit: half of the smallest limit from only the moving axes.

Applies one jerk limit to what though? Each axis could be moving at a different speed. Each of those speed changes should be compared to the jerk for that axis. If it is exceeded that speed should be capped and all the others reduced by the same factor to keep the axes in sync. That is exactly what happens when there is more than one move in the buffer. I don't see why it isn't always the case. 

For example, suppose you have an XYZ move with X and Y moving much faster than Z. If Z is slightly over its jerk value the vmax_junction would be set to its jerk value but the travel speed, which is the vector sum of XYZ speeds, only needs to drop a little to bring Z in range, not be capped to the Z jerk speed.

The only difference is the move has to full stop if the there is no more in the buffer so the jerk is equal to the speed, rather than the difference between speeds.

Also the divide by two make no sense because if jerk is defined as the start speed it will be halved for single moves but not if there are more moves in the buffer. Why would that be?

nophead, 2015-04-25:
Here is a very old version of the code https://github.com/MarlinFirmware/Marlin/commit/094afe7c1065d5663628b389f27687a5f465abb8#diff-34334e15f7432bad0e9e354453604423L1231 which has some comments. Not sure why nice commented functions got replaced with variables. Presumably safe_speed is half to allow you to double back on yourself without exceeding the jerk, so you can run at safe speed without ever considering what the next move does. 

However I still think it is wrong that if I do a move stand alone it starts slower than if another move is buffered up. Only the exit speed should half until it knows what comes next.

nophead, 2015-04-25:
It's effect is you can't test jerk settings with manual commands or jogs because it gets halved compared to running gcode from a file.

thinkyhead, 2015-04-26:
I see the full history of that line is interesting.
```
git log -L886,886:planner.cpp | grep "commit "
commit 4ca2f336f093d824bae1aa460744ab92255b7144
commit e45e5c68bb67ddec39b9bda25cf96396c12fe10b
commit a9c7da06e398101517acc5b7a07e8dcb544df294
commit 36958ee30576dd282a373c8b1da6ad8d9d22eb7b
commit 72ace55e6ae964a5984a13e09492d1d92295f098
commit 79d1bfb5c02793c88f554002e04b5263d81703b7
commit 1d171e9e52012c8129dfec2b7f2640e696678f2e
commit b9ea0651ec52e0c7c20761c0409cd59259b984bf
    This reverts commit 8bcdb9f5f062fe68d90e31752453e1d662b3f0e2.
commit 8bcdb9f5f062fe68d90e31752453e1d662b3f0e2
commit 40e80816235739064fcc7fd74228c9c225387caf
commit 00674af3a82e6b6b517a12ae3e77803df4f4892d
commit 094afe7c1065d5663628b389f27687a5f465abb8
```
All the planner code was replaced just 3 commits ago.
```
4ca2f336f093d824bae1aa460744ab92255b7144
-  if ((moves_queued > 1) && (previous_nominal_speed > 0.0)) {
+  if ((moves_queued > 1) && (previous_nominal_speed > 0.0001)) {

e45e5c68bb67ddec39b9bda25cf96396c12fe10b
- if ((block_buffer_head != block_buffer_tail) && (previous_nominal_speed > 0.0)) {
+ if ((moves_queued > 1) && (previous_nominal_speed > 0.0)) {

a9c7da06e398101517acc5b7a07e8dcb544df294 - bkubicek authored on Nov 15, 2011
Huge replacement of planner.cpp code
- if (previous) { ... }
+ if ((block_buffer_head != block_buffer_tail) && (previous_nominal_speed > 0.0)) {
```
I think it's worth looking at the planner code of other firmwares to see what their approach looks like, and borrow liberally.

closed 2016-04-23 by jbrazio