MarlinFirmware/Marlin #1930: Delta Print from SD not working?

ProjectEnder, 2015-04-17:
Been having some issues printing, and figured i would make sure it was not computer controlling it problems by printing from SD, but cants get the system to even start printing from the SD.

Its loads up the file, heats the Bed, then the Hot end, and Home all the Axis. but never gets farther. 

are any other Delta Printers are getting the same issue.

Notes: the screen only says "M105" in the lower left.
bellow are the first few lines of the G-Code

G21 ; set units to millimeters
M107
M190 S90 ; wait for bed temperature to be reached
M104 S230 ; set temperature
G28 ; home all axes
M106 ;
G1 Z5 F5000 ; lift nozzle

M109 S230 ; wait for temperature to be reached
G90 ; use absolute coordinates
G92 E0
M82 ; use absolute distances for extrusion
G1 F3300.000 E-4.00000

Marlin 1.0.3 DEV
downloaded 16/04/2015

ntoff, 2015-04-17:
My cartesian does that, I've found pressing the encoder button sets it off and running again.

epatel, 2015-04-17:
Does the hot end reach 230 degrees? As it should do a home and raise the nozzle a little and then wait for the hot end to reach 230 degrees. My pid values seem to be a little off though I did a M303 so it will overshoot the target level and M109 will not accept it until the heat has stabilized close to the target temperature.

ProjectEnder, 2015-04-17:
@ntoff  i will give the button press a shot.
@epatel Yes it reaches and hold 230, sat at 230 for 5 min with no action.

Forgot to add this strange thing, while printer from SD i was monitoring with USB so i could keep a eye on the temps via computer, Printrun stopped receiving temp updates on both bed and hotend before the bed hit temp

alexborro, 2015-04-18:
This is an old bug.. I have a Marlin version one year old with same bug..
Never had time to investigate, as I ever print from SD with no USB
connection.

Cheers.

Alex.

thinkyhead, 2015-04-18:
Make sure the SD card lock slider isn't too loose. Also make sure the SD card is seated firmly. Finally, tighten all cables to the controller, bending pins a little to increase grip. Make sure to use good quality USB cables with ferrite cores. And, check the board itself for potential interference with the serial UART, then see if you can cause it to fail by manipulating cables.

alexborro, 2015-04-18:
Scott, this is a known bug. You can reproduce it.

Reset the printer, connect the USB cable and then connect the Host software
to the printer.

Then start any printing. After initial heating, it will wait for the lcd
button press in order to go on.

Cheers.

Alex.

nophead, 2015-04-18:
Is it just a delta bug? Works fine for me with OctoPrint and a Panalolu2.

ntoff, 2015-04-18:
@nophead nope, happens to me on my cartesian printer. Doesn't happen all the time though and I'm not sure what it is that makes it  happen. It kind of seems to be random for me.

thinkyhead, 2015-04-19:
There's not some kind of `M0` or 'M1` command happening from the host is there? Apart from that, what else can cause the LCD to wait for a button press? And, does it show a message on the LCD? I would test this, but I've got to run!

ntoff, 2015-04-19:
I just messed about before and although it's not the latest dev version it happened after I did some calibration prints while connected to the laptop, and then a print from SD. It sat there heating up and once heated it displayed the M105 in the LCD and I had to hit the encoder button (twice this time) to make it print.

nophead, 2015-04-19:
Could you post the start of your gcode up to the first printing move please.

Are you saying it displays "M105" on the display. That would imply there is a M117 going wrong.

ntoff, 2015-04-19:
I'm not the OP but: http://pastebin.com/i7TvLVE3 have at it. Gcode is the same for both printers.

>Are you saying it displays "M105" on the display. 
Yep.

nophead, 2015-04-19:
Try removing the M117s. My guess is they are somehow getting mixed with the host sending M105 and somehow it see M1 as a command. Might be a host problem with M117/M105 or a Marlin problem parsing M117.

ntoff, 2015-04-19:
OP posted their gcode and there aren't even any M117's in it.
It's printing from SD when it happens to me so no host program involved. 

I'll try it later, 10pm and I'm fed up with fiddling with the printers. One of them's gonna get chucked out a window if I try anything more tonight.

nophead, 2015-04-19:
Did you use a host to put the code on the SD card with Marlin or did you write it from a PC?

Was the listing you posted taken from the SD or was it the original file from the slicer?

Are you printing stand alone from SD or do you have a host attached as well?

* I think the only reason Marlin waits for a button press is M0 or M1. 

* M105 is only sent by hosts, it don't think it is ever written by slicers.

* The fact you see M105 on the screen looks like the gcode stream seen by Marlin was a jumbled mixture of the file you posted and M105. I.e. the M117 message got an M105 in it. That would not cause a pause but if M105 somehow gets seen as M1 that will cause the pause.

It looks very much like Marlin was seeing M105 injected into random places of the file stream. If you place M105 in the body of M117 it will display but if you place it a few characters sooner you can get M1M10517 and that probably pauses.

I never see this problem but I never use M117 and I only use OctoPrint or a very old version of Pronterface. So the bug seems like it must be M117 or host specific.

thinkyhead, 2015-04-19:
A host that sends line numbers and checksums should not allow the stream to get corrupted. Maybe there is something happening with the ring buffer getting corrupted, but then I would expect to see more serious problems.

AnHardt, 2015-04-19:
Tested uploads to SD with Pronterface (2014.03.10) and Octoprint 1.1.1.
Pronterface adds some M105, but in separate lines.
Octoprint did not add M1005s.

Do you see any messages on the serial line - if connected - during the pause?
What boards do you use. Are these 'small' processors? Do we see a heap stack collision?

ProjectEnder, 2015-04-26:
Sorry for not getting back to update this issues sooner.

So yes the clicking the LCD button gets the M105 on the screen the clear and the print starts. i am not at my system currently so i can not give a updated GCode. all my GCode file are placed on the SD card via USB SD Reader.

When i found this issues, i did not see any message on PrintRun from the serial line, but the issues happens both with computer monitoring and with out. I Can give it another shot later. I will pastebin one of my GCodes as well if that helps.

ProjectEnder, 2015-05-01:
I have look over my GCode many time and there is no M105 in it, but the printer still seems to stall. and everytime and encoder wheel press will clear it and move on with the task. 
I am doing all my slicing with Slic3r.

I dont see this a huge issue myself, i just was worried the first time i sow it happen

thinkyhead, 2015-05-02:
@ProjectEnder Have you downloaded the latest code and tried it? I can find no instance of the string `M105` except as an error message, which is only printed when a bad `M105` command happens, and i should be displaying even more than that. But, try the latest code, and if the problem pops up again, please post a photo. Maybe it will make more sense once I have seen it.

ProjectEnder, 2015-05-04:
I updated by code, and then did many prints yesterday with no issue. i will close this issue and open a new one if it comes back with screen caps and serial debug info next time.

closed 2015-05-04 by ProjectEnder

thinkyhead, 2015-05-05:
:boom: