MarlinFirmware/Marlin #1932: Enable Thermal Runaway Protection by default

harerama, 2015-04-17:
It saved my printer once (many thanks !):
I really think security features should be enabled by default and the default configuration seems ok to me.

thinkyhead, 2015-04-18:
I think it was enabled by default at some point, and I forget why it was disabled. But it makes sense to enable it by default, as a safety feature.

AnHardt, 2015-04-28:
@thinkyhead 
The history tells me. @alexborro added 'Thermal Runaway Protection' on Jun 30 2014 with commented out #defines. This is untouched until now.
Should be save to be enabled by default.

thinkyhead, 2015-04-29:
#1991 enables it for extruders. Do you think it ought to be enabled for the bed as well?

AnHardt, 2015-04-29:
Yes, i think so. A heater like the others.

harerama, 2015-04-29:
I think all should be enabled yes.
It saved my printer because I had a cable connection ON THE BED partially cut, so the few remaining wires became really hot (with smoke going out !!!) - as the bed could not maintain its temperature, the protection activated.
As a rule, all safety protections should be enabled by default - letting power users the ability to disable them if they want. We are discussing potentially burning homes here, so the safer the better...

Thanks again for saving mine ;-)

thinkyhead, 2015-04-30:
Done!

closed 2015-04-30 by thinkyhead