MarlinFirmware/Marlin #1934: Dangerous homing behavior on Delta with Z-Probe triggered

Natealus, 2015-04-17:
I was testing the height of my new design slowly stepping the effector down until it triggers the z-probe and I know what the nozzle to z-probe offset should be. Once I've done that and the Z-probe is still triggered, I hit the home button to send G28 and ONLY the X tower's motor moves and bashes the effector into the tower skipping the stepper motor like mad. 

This is a Very bad behavior that can cause physical damage as it busted the clip that holds the diagonal rods on the effector. Luckily I have 1 spare. So catastrophe averted, but we need to look at why it did that.

ProjectEnder, 2015-04-18:
is that Z-Probe on Zmin or the Zprobe Pins?

thinkyhead, 2015-04-18:
@Natealus Yes, lots of debugging around bed leveling right now, but unfortunately I'm not equipped to test, so this feedback is invaluable. I'm half-asleep, so remind me, did you already submit a debug log from my debug_leveling branch?

Natealus, 2015-04-18:
@ProjectEnder Same thing on my board. I thought Z-min was the common place to put the Z-probe? 

@thinkyhead I haven't tried that yet...is it up to date and you need more debugging logs? This isn't an action During the G29 E but when I was manually moving the effector down to get a reading on how far the nozzle Z is from the probe. While it was still triggered I hit home and that happened.

thinkyhead, 2015-04-20:
@Natealus The behavior you're seeing might not be easy to decipher from the logs, but give it a try anyway, because it does do logging around `G28` in addition to `G29`. Of course, be prepared to hit the power switch when it goes crazy.

I'm keeping my debugging version separate, and updating it as the head of the Development branch changes. The current will always be at this location:

- https://github.com/thinkyhead/Marlin/tree/debug_leveling
- https://github.com/thinkyhead/Marlin/archive/debug_leveling.zip

You can compare the branch to Development to see where the debug logging has been added. Please feel free to add more logging and pass it along. Maybe we will add similar debug logging throughout the code for the future.

Tonight I'm going to go through the logs we have so far and scan the code for causes of problems, so hopefully there will be some patches to test soon!

Wurstnase, 2015-04-20:
This could be the G28 X thing. The host sends G28 X0 and then this will crash. We should make a filter when max end stops are used that the 0 go away.

Natealus, 2015-04-23:
Yeah Mattercontrol sends G28 X0 Y0 Z0. Which sends the towers up simultaneously towards Max until touch and then Z is at it's set height and down she goes from there to bed center 0,0,0 so technically X and Y should be center but that happens after each touches in sequence. Looks like it goes X, Y, and then Z. When it crashed though only the X tower moved...the others stayed exactly where they were before I sent the homing command.

sandyhello123, 2015-04-28:
ok

thinkyhead, 2015-04-28:
Where are things at now? Some changes have been made just today to fix a couple of homing-related issues, and you can also those latest changes also in my [debug_leveling](https://github.com/thinkyhead/Marlin/tree/debug_leveling) branch, for the most thorough debugging. I've also added initial support for the `M111` command, allowing debug output flags to be changed at run-time, which may help with some future diagnostics.

thinkyhead, 2015-05-09:
The current Development code still has a Z positioning problem with `G29`. One of the changes in that commit is responsible for the weird behavior. But I've narrowed down the current leveling issues to a small handful of recent changes, and rolled back the leveling code to around March 26 at the branch https://github.com/thinkyhead/Marlin/tree/before_delta_merge

The differences between that code and the current leveling / homing code are embodied in commit https://github.com/thinkyhead/Marlin/commit/3f0a990360d04a45c11ba6ebdfeff2b16d26bf34

So, I suggest starting with [before_delta_merge](https://github.com/thinkyhead/Marlin/tree/before_delta_merge) to see if it behaves properly. Then we can try to determine which of those recent changes is responsible for any remaining problems.

a4jp-com, 2015-10-02:
Has any work been done on this or has this problem fixed?

Natealus, 2015-10-07:
I can confirm this still happens even with a capacitive proximity sensor. It must have something to do with the Z-probe or min stop code that might be appropriate on another machine other than delta.

clefranc, 2015-10-07:
I just tried to reproduce this issue on my delta without success. The G28 run flawlessly. It was a problem I've also detected, a few months ago. But It's been fixed since: https://github.com/MarlinFirmware/MarlinDev/commit/e2d69f66b29aa0957ff4610ab1a6644fe8393fca

You'll have to provide more information on your setup, configuration files and pin.

Natealus, 2015-10-08:
If I can reproduce it again I'll let you know. I'm working on the RC right now but I'm mostly doing production work right now. When I get some free space I'll work on the ABL system again. I think it was happening when we were having pattern problems with the grid and when it was still "triggered" down there. Doesn't happen during manual movement commands.

AnHardt, 2015-11-06:
Fiyed by https://github.com/MarlinFirmware/MarlinDev/pull/65

closed 2015-11-06 by AnHardt