MarlinFirmware/Marlin #1936: Home offset not applied

ntoff, 2015-04-18:
M206 values are not applied. 

Tested on latest dev version M206 Z-0.3 is my usual home offset, now it does nothing. Homing the printer tells it that it's at Z0 and if I type G1 Z0 it doesn't move .It's physically below the bed at home position so if it thinks Z0 is touching the bed, oh boy.

Prior to this, homing the printer would display Z-0.3 and typing G1 Z0 would move it upwards by 0.3mm.

Same for X and Y.

thinkyhead, 2015-04-18:
Strange, I'll check it out later today.

ntoff, 2015-04-18:
I looked at a bit of the code last night but couldn't see what got messed up, then got  distracted in an attempt to automate my configuration setting up procedure.

The code has changed dramatically vs the version I have that works so I couldn't even use that to compare.

ntoff, 2015-04-19:
https://github.com/MarlinFirmware/Marlin/issues/1942

It's linked to what's talked about in there. G28 Z homes and sets the homing offset, G28 Z0 does not.

So who is going to go and re-write the reprap wiki to add the fact that it now takes notice of the value you send it? http://reprap.org/wiki/G-code#G28:_Move_to_Origin_.28Home.29 says it ignores any value.

So does this mean I have to manually write into my Gcode at the start G28 Z-0.5? So that's now two places (start code and M206) where this value is used, and can be potentially different.

lcfm1, 2015-04-23:
https://github.com/MarlinFirmware/Marlin/issues/1953

daid, 2015-04-23:
I think the change where "G28 X0" is something different then "G28 X" is really really stupid. As this breaks a lot of current assumptions. It also breaks GCode as general that you have "key" + "value" pairs.

Wurstnase, 2015-04-23:
#1945 PR is waiting for merge

thinkyhead, 2015-04-24:
What should `G28 X22.5 Y117 Z12` do? If the specification was that these values should be ignored, well, these values have been used for a while, apparently. You will also find precedent for places where the presence of `E` (for example) matters, and it doesn't require `E1` or something like that. So, should this be corrected throughout the code? And, should the parameter values to `G28` be ignored?

ntoff, 2015-04-24:
I think a lot of host programs probably send G28 n0 (n being x y or z). I don't know why you would ever have G28 applying to an extruder, unless maybe a paste extruder? Does Marlin even have a setup for having end stops on an extruder?

The issue is (here anyway) the host program sends G28 Z0 and it homes there, and then tells the printer that's where it is and it's physically not at 0. Sure, I could re-write the home thing in pronterface to not send the 0 but it's not limited to pronterface.

It's an issue for me because I have my nozzle homed to below the bed so if I use the current version and hit the home button in pronterface and start printing, the printer thinks it's already at z0 when it's not so when it goes up to 0.1 for the first layer *SMACK* well, there goes the nozzle, slammed straight into the side of the bed.

To me it makes more sense to have G28 Z0 home to the physical home switch, and then MOVE to Z0, and not just tell the printer "hey, you're now at zero, good luck if you're not" because that's specifically what G92 is for.

daid, 2015-04-24:
The G28 values should be ignored. If you want to set the position, that's what G92 is for.

Rule 1: Don't break backwards compatibility.

thinkyhead, 2015-04-25:
#1965

closed 2015-05-09 by thinkyhead