MarlinFirmware/Marlin #1939: AUTO_BED_LEVELING not using correct first point

fitch22, 2015-04-19:
This is from recent build, source dated today.  Using Printrboard on Simple Metal that has Y endstop at max travel.

Executing G29 just uses what appears to be the last position as its first point, then moves the correct delta amount to the 2nd and 3rd points, but the absolute coordinates are off since the first point is off.  In my case, the 3 points are supposed to be (25, 140) (25, 10) and (125, 10).  It looks like it is instead using (25, 150), (25, 20) and (125, 20).  This worked as of my previous build with source dated 4/4/2015.

thinkyhead, 2015-04-24:
Does the firmware think it's at a different point before it starts `G29`, or does it correctly know the current coordinate? Does `G28` work correctly for you? The behavior of `G28 XYZ` should be the same as `G28` but `G28 X0 Y0 Z0` will screw up a delta, or any machine with max endstops. For delta machines, the correct syntax would be `G28 X0 Y0 Z180` (for a machine with `Z_HOME_POS` set to 180). At least, that's the idea. Can you confirm that `G28` works as I describe on your machine?

fitch22, 2015-04-24:
@thinkyhead, This bug showed up and is fixed on what code I am using by #1945.  I think it was all related to homing Y where my endstop is at max Y but the position was reset to 0.  The first point of the auto level wanted it to move -10 in Y and it could not.  Subsequent moves for the other 2 points must have been relative.  I believe it is fixed with #1945.

thinkyhead, 2015-04-25:
#1965 goes even further – _All values_ to XYZ are ignored, they are just flags to home a certain axis. This is consistent with the [documentation for G28](http://reprap.org/wiki/G-code#G28:_Move_to_Origin_.28Home.29) on the RepRap wiki. So, give the latest code a try and see if it's doing any better!

fitch22, 2015-04-25:
I have confirmed that #1965 fixes my original issue (so did #1945).  I have changed all startup code to use "G28 X Y" with no positions.  All is good.  Hope it is OK to close this issue.  Thanks.

closed 2015-04-25 by fitch22

thinkyhead, 2015-04-26:
:hamster: