MarlinFirmware/Marlin #1940: G28 Y0 from unknown position starts with full speed move

fitch22, 2015-04-19:
With recent build (dated today), a home in the Y direction on my Simple Metal with Printrboard, starting from unknown position, starts out with what appears to be a full speed move in negative Y direction to get off the endstop, then smoothly approaches the endstop, then smoothly leaves the endstop and returns to the stop.  A subsequent home for Y does not do this, nor does a home in X.  However, my printer homes X at minimum X and homes Y at maximum Y.

This is very repeatable, if I disconnect and reconnect (from Repetier), and then home Y, it really jerks off the endstop, then completes the homing with smooth moves.  From the same condition, a home X has all moves smooth.

Also, my jerk value is set to 10 and all my acceleration values for X and Y are set to 1000.

Printing from this version of code seems smoother than before, that may be related to the acceleration changes contributed by nophead.  The sound is noticeably better.

fitch22, 2015-04-21:
I found some more behaviors that might be helpful.
1. Tracing through the code for G28, I could find no difference between how a home in X is different from a home in Y, so I repeated various tests.  Indeed, the very first move in X is also rather jerky (technical term).  I suspect that there is no acceleration on that first move.
2.  It does not matter whether Y is at the endstop or not.  The first move is jerky, but second and subsequent moves are not.
3.  If I do a manual move in Y prior to the home Y, then home Y is smooth.  It is possible that my manual move would have been jerky but since it is done slowly I don't hear it.
4.  After homing Y, I cannot manually move in -Y position (my endstop is at max Y).  Yet, unfortunately I can manually move in +Y direction which chews up my belt since it is already at the stop.  This sounds like #1945.  This also may explain #1939 where after homing, the G29 does not move Y in the - direction to go to the first point.

fitch22, 2015-04-22:
At the moment I am lost in plan_buffer_line().  Maybe someone more familiar with this code can help.  What it looks like to me, is that when you power up the printer, and perform a home Y (or home X), it artificially sets its position as 0, then sets a target of 1.5 x max travel in the direction of the endstop to ensure it can hit it.  However, it does not really know exactly where it is and I think the trapezoid calculation gets messed up, thus the initial speed is too high.

This happens for the 1st move toward the endstop, or if it is already at the endstop it happens on the first move away from the stop.  It never happens for that axis ever again.

I can even disconnect from Repetier and reconnect, so Repetier has no knowledge of the position, but as long as power has not been cycled, all moves are smooth.

Wurstnase, 2015-04-22:
Repetier sends a G28 Y0, allways, when using the buttons. You could home with G28 Y200 or whatever your max-endstop is and test it again.

nophead, 2015-04-22:
Surely that should be G28 Y to use the home position set in the firmware?
You don't want to have to duplicate that information in the host config.

nophead, 2015-04-22:
Actually until recently Mmarlin ignored a value of 0. Changing that has
broken compatibility with hosts.

Wurstnase, 2015-04-22:
That's right, If informed Repetier allready. In the past every firmware only homed by G28 X or Y independend what numbers were behind X or Y.

#1945 will fix this. But most host programms needs to know that. There is a patch request also for Pronterface. But keep in mind, that this will break other firmwares.

nophead, 2015-04-22:
The code used to look like this https://github.com/nophead/Marlin/blob/Marlin_v1/Marlin/Marlin.ino#L662. Zero was ignored, probably because hosts send it.

#1945 still won't work on my machines even though the X and Y axis homing direction is -1, the home position is -101, not 0.

Wurstnase, 2015-04-22:
Yes, again. Don't think about other min/max-solutions. I will change this.

nophead, 2015-04-22:
Although ignoring  zero seems odd, I can't see any use case for needing to force home to be zero when it is set to something else in the firmware. I can see setting it just off from zero would be useful to people that home to Z = 0.

Wurstnase, 2015-04-22:
Any value in G28 are not necessary. If you need a different value for your min or max endstop you can configure it allready. Or you can use a G92. So I don't see any benefit for this. Just messing this command.

fitch22, 2015-04-22:
Sorry for the late reply, but I have incorporated #1945 into my code and it had no effect on the abruptness of the very first move.

fitch22, 2015-04-23:
OK, I have done my best to debug what I hear.  I have put print statements for all variables that might possibly be involved in the homing and have compared the differences between the very first home and a subsequent home.

In the X direction, the results are identical.  There appears to be no variable differences between what gets initialized and how things are the 2nd time through.

In the Y direction, there are 2 differences.  The initial current position is 0,0 which makes sense because the printer wakes up and has no idea where it is.  The 2nd time, it thinks the current position is 0,150 which is correct as my printer homes at max Y of 150mm.  There is a destination that also shows up as 0,0 first and 0,150 second, but I think that is just copied from the current_position.

So, I am at a loss to find anything wrong with the code.  I am now wondering if the sounds I hear are related in some way to Repetier turning on the motors simultaneously with the first move.

I would appreciate anyone else's input on this, but also am sorry I wasted anyone's time.  This does not seem to be a bug.

closed 2015-04-23 by fitch22

thinkyhead, 2015-04-24:
@fitch22 I've got a build with lots of debugging code added for `G28` and `G29` at https://github.com/thinkyhead/Marlin/tree/debug_leveling if you'd like to help debug this further!