MarlinFirmware/Marlin #1942: Homing to max position sets the initial coordinate to 0 instead of the max coord.

aljscott, 2015-04-19:
I've been having various issues with the old Marlin release running a dual extruder so I decided to try and upgrade. After all the various issues with the new releases e.g. LCD not working etc. I've pulled the main branch and got it running on my machine.

So it appears that if you have the homing direction set to +1 instead of -1 so that it homes to the max position it ends up setting the coordinate to 0 instead of the defined max coordinate.
So for example I home my X to the right i.e. x-max. After it is homed the position is 0 instead of 200mm. This was working fine on the old marlin release. Same settings in configuration.h

// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR 1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

#define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.

// @section machine

// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 175
#define Z_MAX_POS 183

What I noticed was that if set MANUAL_HOME_POSITIONS and selected this option as the first thing I did after the machine was turned on.. Then did an auto-home, the machine would end up with the correct start coords after the homing.

ntoff, 2015-04-19:
I wonder if this is related to my issues, even when homing to the min endstops, it still says 0,0,0 and doesn't apply any home offsets (my home is actually -2,-10,-0.5).

Wurstnase, 2015-04-19:
#1904

aljscott, 2015-04-19:
https://github.com/MarlinFirmware/Marlin/commit/ac2b23f5743a822e04296b2e7fbbb6d8fd855128

Don’t add `home_offset` when setting XYZ explicitly in `G28`

I suspect this isn't quite doing the right thing.

orangeudav, 2015-04-19:
Hah, got same issue with newest Printrun. It send G28 X0 instead of G28 X.

thinkyhead, 2015-04-24:
I hate to have to work around bad host behavior, when `G28 X0` _clearly makes no sense_ for a max endstop. Up until recently `G28 X0` would not set the X coordinate, and that was a workaround, treating zero as a special case. Any other value would set X, but 0 would be ignored. But, make it work like it's supposed to work and it exposes heretofore unseen problems with hosts and slicers.

I would like to keep the proper behavior, but as a compromise add a new configuration option and maybe just make it enabled by default:

    #define G28_IGNORE_ZERO

ntoff, 2015-04-24:
You may or may not want to work around bad host behaviour but how many hosts out there actually send "broken" code because that's how it's been for so long?  I honestly do not ever see a reason to use G28 to tell the printer where it is when we have G92 specifically for that reason. IMHO G28 should just ignore any and all numbers after it, and we use G92 for setting where the printer *think* it is.

thinkyhead, 2015-04-25:
#1965 applies the correct behavior in `G28`. The extension where current coordinates are set by including values with `XYZ` has been removed.

closed 2015-04-25 by thinkyhead

FootStank, 2017-08-22:
I had the same problem and found the line in the Marlin setup, that sets the manual home to (0,0,0)
The line is like this:
#define MANUAL_Z_HOME_POS 0
I changed it to this:
#define MANUAL_Z_HOME_POS Z_MAX_POS