MarlinFirmware/Marlin #1947: Stop a M190

MoonshineSG, 2015-04-21:
Is there no way to stop a M190 (Wait for bed temperature to reach target temp) ? I tried to send a M140 S0 (Bed Temperature (Fast) , but there seams to be no response from the printer...

thinkyhead, 2015-04-24:
@MoonshineSG We've had this request before, and it doesn't seem to be going anywhere: #685. I'm not sure if the new `M410` command will help, but give it a try.

closed 2015-04-24 by thinkyhead

nophead, 2015-04-24:
The problem is the host will typically send the gcode following the M190, so the M410 will queue behind those and not be processed until the M190 has finished.

thinkyhead, 2015-04-25:
I've proposed that there be some commands allowed to skip ahead in the queue, so long as they come from the serial host. Currently it seems as though `M112` (`kill()`) might go through immediately. @MoonshineSG See if `M112` does the trick.

nophead, 2015-04-25:
If the command queue is full it stops reading commands from them serial buffer, so it will only be seen if the host doesn't fill the queue.

thinkyhead, 2015-04-26:
Does the host continuously re-send (during a print) until the line is acknowledged, while the firmware ignores commands while the buffer is full? Or does the firmware send "ok" when it's ready to accept a new command? Depending on the handshaking, it might still be possible to accept commands on the serial line, even when the buffer is full. But has this already been attempted?

MoonshineSG, 2015-04-27:
@thinkyhead M112 doesn't work (nor does M410)

thinkyhead, 2015-04-28:
@nophead @MoonshineSG I've been looking at `M23` and see that it would benefit by blocking any new commands going into the buffer until the buffer is empty. (Currently, if you put more than one `M23` in a row, they all get executed before any new file is started.) Perhaps if the buffer was blocked from accepting any new commands (pretending it is full) until `M190` released it, this would do the trick too.

povlhp, 2018-06-09:
I found one solution - Goto to printer and select cooldown, then the job terminates.
Would still be better with a way to do it from OctoPrint. And I do not consider the Arduino reset an option. I use that from command line in real emergencies only (long print going bad and seen on video).

thinkyhead, 2018-06-10:
@povlhp — With `EMERGENCY_PARSER` enabled `M109` and other blocking commands can be interrupted with `M108`.