MarlinFirmware/Marlin #1948: FILRUNOUT

Wurstnase, 2015-04-21:
Is there a reason for
```
#if HAS_FILRUNOUT 
    if (card.sdprinting && !(READ(FILRUNOUT_PIN) ^ FIL_RUNOUT_INVERTING)) 
    filrunout(); 
#endif 
```
the card-part?

[Marlin_main.cpp#L5820](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L5820)

nophead, 2015-04-21:
Presumably if it was printing from a host the host would get upset when it stopped responding during the M600 filament change procedure.

Wurstnase, 2015-04-21:
Could an M400 solve this?

nophead, 2015-04-21:
I don't see how. To work it would need to continue to reply to the host and somehow indicate it was busy to stop the host timing out and giving up. 

Easier for the host to read filament pin occasionally and handle the filament change itself.

Wurstnase, 2015-04-21:
good idea! for repetier host it's easy. We can make a MSG_FILRUNOUT and save this in the eeprom or something like this. When the pin change it can send a '@pause' for repetier and something else for other hosts.

ntoff, 2015-04-21:
>Presumably if it was printing from a host the host would get upset when it stopped responding during the M600 filament change procedure.

I print from Pronterface and it plays nicely with M600 filament change commands

Wurstnase, 2015-04-21:
Also while printing?

ntoff, 2015-04-21:
I print using pronterface, so I open the gcode, scroll to where I want the filament change, insert an M600, load it up in pronterface, and hit print.

When it gets to the filament change the printer goes home, beeps like crazy at me (sounds a bit like the alarm from Ghost Busters) until I change the filament and hit the button. Then off it goes on its merry way again.

I don't know how long I could leave it sit for because the heaters are on and I sit & watch it when it's about to come up to the filament change. I've not had pronterface time out though.

Granted this is version 1.0.2 "stable" release, not the latest dev version so something may have changed.

Wurstnase, 2015-04-21:
Ok, when this works, we could also remove ```card.sdprinting && ```

nophead, 2015-04-21:
Putting M600 in the gcode and sending from the host is a different case to
the runout switch triggering the firmware to inject M600. In the former
case the host could see the M600 in the stream and expect to wait. In the
later case the host would be unaware unless M600 sends something to it. It
would just see the firmware become unresponsive.

Wurstnase, 2015-04-21:
Good argument. I will go forward with your idea.

MagoKimbra, 2015-04-21:
You're forced to check sd.printing, otherwise at any time, even when the printer is switched on there is the thread running the command fw m600.

Wurstnase, 2015-04-21:
That could be made by a gcode. Some sign for the firmware which activates the sensor.

MagoKimbra, 2015-04-22:
Yes Wurstnase. I create gcode M11, this is placed at the beginning of a gcode to indicate the start print and from this moment can be controlled the switch filrunout.... But if tou print from sd the card.sdprinting is useful for this...

Wurstnase, 2015-04-22:
I wouldn't take an other M-code. M600 S1/S0 for activation/deactivation should make that job. So keeping the functions close together will make the job easier for the people.

MagoKimbra, 2015-04-22:
Yes this is a best solution.... I use M11 because one software (i do not rember name) use m11 when start transfer gcode to printer for print, and i decide use M11 for filrunout, but your solution is better...

closed 2016-05-04 by jbrazio