MarlinFirmware/Marlin #1949: Compile Error: Y Dual Stepper Drivers

gegiarmo, 2015-04-21:
I get this error when uncommenting (enabling) #define Y_DUAL_STEPPER_DRIVERS. I get a very similar when if enabling dual Z steppers (not at the same time as dual Y, but did so just as a test for the error).

Arduino: 1.6.3 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

stepper.cpp: In function 'void __vector_17()':

stepper.cpp:127:42: error: expected ';' before ',' token

   #define Y_APPLY_DIR(v,Q) Y_DIR_WRITE(v), Y2_DIR_WRITE((v) != INVERT_Y2_VS_Y_DIR)

                                          ^

stepper.cpp:392:7: note: in expansion of macro 'Y_APPLY_DIR'

       Y_APPLY_DIR(INVERT_Y_DIR,0);

       ^

stepper.cpp:127:42: error: expected primary-expression before ',' token

   #define Y_APPLY_DIR(v,Q) Y_DIR_WRITE(v), Y2_DIR_WRITE((v) != INVERT_Y2_VS_Y_DIR)

                                          ^

stepper.cpp:392:7: note: in expansion of macro 'Y_APPLY_DIR'

       Y_APPLY_DIR(INVERT_Y_DIR,0);

       ^

In file included from Marlin.h:21:0,

                 from stepper.cpp:24:

fastio.h:44:24: error: expected primary-expression before 'do'

 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)

                        ^

fastio.h:67:23: note: in expansion of macro '_WRITE'

 #define WRITE(IO, v)  _WRITE(IO, v)

                       ^

stepper_indirection.h:70:29: note: in expansion of macro 'WRITE'

 #define Y2_DIR_WRITE(STATE) WRITE(Y2_DIR_PIN,STATE)

                             ^

stepper.cpp:127:44: note: in expansion of macro 'Y2_DIR_WRITE'

   #define Y_APPLY_DIR(v,Q) Y_DIR_WRITE(v), Y2_DIR_WRITE((v) != INVERT_Y2_VS_Y_DIR)

                                            ^

stepper.cpp:392:7: note: in expansion of macro 'Y_APPLY_DIR'

       Y_APPLY_DIR(INVERT_Y_DIR,0);

       ^

fastio.h:44:24: error: expected ';' before 'do'

 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)

                        ^

fastio.h:67:23: note: in expansion of macro '_WRITE'

 #define WRITE(IO, v)  _WRITE(IO, v)

                       ^

stepper_indirection.h:70:29: note: in expansion of macro 'WRITE'

 #define Y2_DIR_WRITE(STATE) WRITE(Y2_DIR_PIN,STATE)

                             ^

stepper.cpp:127:44: note: in expansion of macro 'Y2_DIR_WRITE'

   #define Y_APPLY_DIR(v,Q) Y_DIR_WRITE(v), Y2_DIR_WRITE((v) != INVERT_Y2_VS_Y_DIR)

                                            ^

stepper.cpp:392:7: note: in expansion of macro 'Y_APPLY_DIR'

       Y_APPLY_DIR(INVERT_Y_DIR,0);

       ^

stepper.cpp:1169:1: error: expected '}' at end of input

 }

 ^

stepper.cpp:1169:1: error: expected '}' at end of input

stepper.cpp:1169:1: error: expected '}' at end of input

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

AnHardt, 2015-04-21:
@gegiarmo 
Please give us a bit more information about your setup. I can't reproduce this error.

AnHardt, 2015-04-21:
It looks a bit as if there are no pins defined for your electronics board. What board?

gegiarmo, 2015-04-21:
Ramps 1.4. Heated bed pin moved to pin 4 for standalone transistor control. I've since downloaded a fresh copy of Marlin, enabled dual stepper drivers, and am re-enabling my settings and compiling as I go to see if there's a conflict somewhere. I'll let you know what I find.

gegiarmo, 2015-04-24:
Not sure what the problem was. Starting with a fresh copy of Marlin, with dual steppers enabled from the start, and then adjusting to previous settings, I found no compile errors or other issues.

closed 2015-04-24 by gegiarmo