MarlinFirmware/Marlin #1950: x axis  works incorrectly & Setup steps / mm

pantileionut, 2015-04-21:
I purchased the following configuration -
Mega 2560 R3 development board
1.4 Ramps card drivers
LCD12864 SD-Card reader
A4988 stepper motor driver

-marlin 1.0.0 - the first day went well and everything nice ... recently driver on the x axis behaves strange: if you try to move the cursor on the x axis motor does not react (the driver enable pin 5V), but if Y or Z axis move any driver x (enable pin 0V) receives the order previously set number of steps executed by the Arduino board restarts after 30 seconds when connected with uSB cable (only with the firmware). I reversed the order of the pins pin axis x with y and y do the same driver set as x ... (indicating that the problem would be soft). (I downloaded the new firmware, I installed on another PC and everything just made).
-1.0.3 -marlin -axis functioning normally but no steps engines exective not correct, or that you 10mm 1mm or 0.1mm engines rotate just as moving as much as does a full rotation encoder .... and if you change stepp / mm do not notice any change .... (version 1.0.0 settings step / mm go ok to YZ axes and E) with firmware 1.0.3 with  not restart arduino.

with another firmware (0.92 Repetier) everything works perfectly when and how it works drivers and settings step / mm are also ok but I would prefer to be able to use marlin.

thinkyhead, 2015-04-24:
Can you post your `Configuration.h` somewhere so we can look at it? It sounds like perhaps the "manual feed rate" might be set too high.

pantileionut, 2015-04-24:
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Scara printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// @section info

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.3 dev"
#define STRING_URL "reprap.org"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

// @section machine

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
// :[0,1,2,3,4,5,6,7]
#define SERIAL_PORT 0

// This determines the communication speed of the printer
// :[2400,9600,19200,38400,57600,115200,250000]
#define BAUDRATE 250000

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_13_EFB
#endif

[... 711 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.0 //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT         3.3 //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT         1.9 //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY       20   //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

pantileionut, 2015-04-24:
I found the manual feed rate, we changed and everything works normal. I think it's from the selection X0.1, X1, X10 because the number of steps was made same trap as the encoder

pantileionut, 2015-04-24:
update pe axele x y si z motoarele nu ajung la pozitia setata pe manual din meniu si nici daca comanzi din printun. indiferent de pasul de miscare 0.1mm 1mm sau 10mm driverele realizeaza acelasi numar de pasi se opresc din miscare dupa ce ai luat comanda neajungand la pozitia corecta ( cred ca  nu memoreaza in buffer). 
p.s. extruderul functioneaza ok...

pantileionut, 2015-04-24:
sorry
xy and z axes update engines do not reach the position set manually from the menu and even if you order in printun. regardless of 1mm or 0.1mm up to 10mm drivers move the same number of steps made stop motion after taking command of not reaching the correct position (I think not stored in the buffer).
P.S. extruder works ok ...

thinkyhead, 2015-04-25:
Thank you, my Romanian is very rusty. :smile:

Try lowering the `DEFAULT_MAX_FEEDRATE` for your Z axis. Instead of 100 make it something between 2 and 5. You can probably raise the `DEFAULT_MAX_FEEDRATE` for your X and Y axes however. I use 500 for each of those without any trouble.

Your `DEFAULT_AXIS_STEPS_PER_UNIT` values seem reasonable. For a GT2 belt with a 20 tooth pulley and 16 microsteps the value would be 80 (`3200 steps / (2mm * 20 teeth) = 80 steps-per-mm`). Your values of 66.67 indicate you are using a GT2 belt with 24 tooth pulley, or maybe a 2.4mm pitch belt with 20 tooth pulleys….

pantileionut, 2015-04-25:
I modified feed rate and did not solve anything
after some tests I found these
-at coordinates moving with a unit (x0,1, X1, X10) engine run just one step regardless of the setting Stepp / mm or move up
- Motor does not lose steps
I think somewhere in the firmware have to multiply between the desired displacement and number of steps to give the correct positioning, and that be the problem

pantileionut, 2015-04-25:
using trapezoid screw M12 with 3mm step

nophead, 2015-04-25:
With x16 microstepping you have 3200 steps per revolution, which is 3mm, steps per mm should be 3200.0/3 or 1066.67, not 66.67.

pantileionut, 2015-04-25:
start
echo:Marlin 1.0.3 dev
echo: Last Updated: Apr 25 2015 22:03:43 | Author: (none, default config)
Compiled: Apr 25 2015
echo: Free Memory: 2994  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X66.67 Y66.67 Z66.67 E66.67
echo:Maximum feedrates (mm/s):
echo:  M203 X25.00 Y25.00 Z25.00 E1000.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X5.00 Z5.00 E10.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Filament settings: Disabled
echo:SD init fail
echo:echo:endstops hit:  X:0.01
echo:echo:endstops hit:  X:0.01
echo:echo:endstops hit:  X:0.00
echo:echo:endstops hit:  X:0.01
echo:echo:endstops hit:  X:0.03
echo:echo:endstops hit:  X:0.04
echo:echo:endstops hit:  X:0.06
echo:echo:endstops hit:  X:0.07
echo:echo:endstops hit:  X:0.06
echo:echo:endstops hit:  X:0.07
echo:echo:endstops hit:  X:0.09
echo:echo:endstops hit:  X:0.10
echo:echo:endstops hit:  X:0.12
echo:echo:endstops hit:  X:0.13

regardless of which control movement step 0.01 mm Says it all makes

nophead, 2015-04-25:
Correct your steps per mm settings and move the axis clear of the end stop.

boelle, 2015-04-25:
What machine do you have?

boelle, 2015-04-25:
And have you checked backlash?

pantileionut, 2015-04-25:
I corrected step / mm but all the same ... do not use micro-stepping driver set from driver
what you mean when you ask what type of machine....
arduino Mega 2560 R3 board
1.4 Ramps card drivers
LCD12864 SD-Card reader
A4988 stepper motor driver
I do not know how to check backlash

boelle, 2015-04-25:
the list is just the electronics side of things.... what about the mechanical side?  

backlash is the amount of movement the stepper can do without anything moving... 

one example is if the belts are very loose... or if the belt has its teeth worn down

pantileionut, 2015-04-25:
the mechanics:
moving on 12mm steel rods with linear bearings and to use movement as I said trapezoidal screw M12 3mm step ..

currently motors are removed so there is no issue of transmission movements
I try to make a little video to see how it behaves ...
problem manifests Y and Z axes X. The extruder everything goes ok steps and others

nophead, 2015-04-25:
echo: M92 X66.67 Y66.67 Z66.67 E66.67

I repeat: FIX YOUR STEPS PER MM: 1066.67, not 66.67!!!!!!!!!!!1

pantileionut, 2015-04-25:
Let's calculate together:
engine has 200 steps to a complete rotation
3mm trapezoid screw has a full rotation
steep / mm = 200 steps / 3 mm = 66.67 steps / mm
wrong somewhere?

pantileionut, 2015-04-25:
I tried with firmware repetier ... everything works perfectly ..... set with 66.67

nophead, 2015-04-25:
Only if your A4988 drivers are in full step mode.

pantileionut, 2015-04-25:
drivers are in full mode ..... I do not know if maybe the settings to be something set

nophead, 2015-04-25:
Apologises then.

Your accelerations look high for lead screws and the fact your X end stop is triggered is suspicious.

pantileionut, 2015-04-25:
please watch the video to better understand the problem ....
https://www.wetransfer.com/downloads/ccdf87f05d88d9bda1afbe400b278fec20150425231338/999ac41921391317dce8cfcf002befc220150425231338/c4e035

AnHardt, 2015-09-25:
error404 for the video link -> user not longer interested.

closed 2015-09-25 by AnHardt

WillSmithInChina, 2016-04-19:
@pantileionut  I also met the problem, when i changed feedrate of x/y , it didn't work, so how do you solve it ?

thinkyhead, 2016-04-19:
Please test with latest `RCBugFix` to see if the issue exists in the current code. If it does, start by checking endstop settings. Then check movement rate settings. If the problem continues, we can look further.

WillSmithInChina, 2016-04-19:
@thinkyhead thanks for your answering