MarlinFirmware/Marlin #1953: DELTA Printer: M666 for z tower not working

lcfm1, 2015-04-21:
for the other towers works

thinkyhead, 2015-04-24:
To help debug leveling issues, please download and try this branch, which prints lots of debug output during leveling. I just added debug output to show the value of `endstop_adj` during `homeaxis`, the only place where it is applied.

https://github.com/thinkyhead/Marlin/tree/debug_leveling

thinkyhead, 2015-04-24:
#1945 might help this also, so give that fix a try. Some hosts send `G28` with `Z0` and this is wrong for max endstops. #1945 ignores zero.

thinkyhead, 2015-04-25:
#1965 is merged (or soon will be). It fixes `G28` so that it will completely ignore any numeric values included with XYZ. This should fix any problems with the coordinates being set wrong with max endstops and deltas, but confirm whether the latest code is working better. There is still a little more fixing up to do around leveling and homing.

lcfm1, 2015-04-25:
@thinkyhead here is my debug log:
http://pastebin.com/Nzgy56dY

thinkyhead, 2015-04-26:
@lcfm1 Well, I'm not sure what's going on, so I have added logging in `M666` to trace when the values are set. If it shows that the values are actually being set, maybe something is over-writing the last one. https://github.com/thinkyhead/Marlin/tree/debug_leveling

lcfm1, 2015-04-26:
@thinkyhead http://pastebin.com/QTJnxYj4

lcfm1, 2015-04-27:
https://www.youtube.com/watch?v=tDLbqLve128&lc=z13iwrqgstbxg5xzg04cdt25nm2qh32zhm40k

Thanks alot! This completely solved my endstop and bedleveling issues. A reminder to those using Marlin! If changing X and Y offsets works but changing Z offset doesn't the add this to your code in Marlin_main.cpp under the function static void homeaxis (int axis){}

    enable_endstops(false);  // Ignore Z probe while moving away from the top microswitch.
    current_position[axis] = 0;
    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
    destination[axis] = -home_retract_mm(axis) * axis_home_dir;
    plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
    st_synchronize();
    enable_endstops(true);  // Stop ignoring Z probe while moving up to the top microswitch again.

Then under #ifdef DELTA add the following

if (endstop_adj[axis] * axis_home_dir < 0) {
      enable_endstops(false);
      plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
      destination[axis] = endstop_adj[axis];
      plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
      st_synchronize();
      enable_endstops(true);

The two pieces of code should allow the Z offset to be confirmed :)

lcfm1, 2015-04-27:
or it's works here https://github.com/RichCattell/Marlin

thinkyhead, 2015-04-28:
#1987

closed 2015-05-09 by thinkyhead

Riekusvdwest, 2017-10-25:
Does any one still have this file?

https://github.com/ErikZalm/Marlin/tree/Marlin_v1/Marlin/example_configurations/delta

It has been removed from the site?

I am stuggeling with the same isseus. Can’t get my m666 to work with z_axis and cant get my auto leveling to work