MarlinFirmware/Marlin #1956: Configurable Homing order of G28

sjkelly, 2015-04-22:
It was mentioned in #1079 that the homing order of G28 could be fixed by splitting the command into three: `G28 Z\nG28 Y\nG28 X\n`. However this leaves the G28 command unsafe to use on many corexy printers than have stationary endstops for X and Y. 

I would like to make a PR for making this a Configuration.h option so we don't have to reorder the execution in the code. Any feedback on this idea would be appreciated!

nophead, 2015-04-22:
What are "stationary endstops" and why is corexy different to other Cartesian machines?

sjkelly, 2015-04-22:
See below.

[image]

nophead, 2015-04-22:
Sorry,  I still don't understand. The picture seems to show a moving opto
and a stationary tab. How is that different from a stationary opto and a
moving tab?

sjkelly, 2015-04-22:
Sorry it was not clear. Both endstops are at (0,0). In this case Y needs to be at 0 to home X, otherwise the flag will miss the endstop.

nophead, 2015-04-22:
I see, it's because the X axis moves with Y but the tab for its limit switch is not mounted on the X axis and stays at 0,0. 

Is there a reason why this is the case for corexy or just a design choice?

ntoff, 2015-04-23:
why not just put your preferred homing methods in your startup gcode? I have mine raise the Z, home x and y and then home Z, it just sits there in my start gcode file and gets added on every time I slice something.

sjkelly, 2015-04-23:
@ntoff We currently do exactly that. The issue is sending a `G28` from a user interface like Repetier or pronterface will home in the traditional fashion which will crash the axes.

thinkyhead, 2015-04-24:
When #1957 is ready to merge it should help with this. It's good to have this option (in one form or another) for those rare designs that depend on it.

sjkelly, 2015-04-29:
I think this can be closed with #1957

closed 2015-04-29 by sjkelly