MarlinFirmware/Marlin #1961: M117 & G4

Salandora, 2015-04-24:
Hello,

Today I played a bit with M117 and G4, wanted to display a specific information x seconds before the Print starts so I thought using M117 for the Message and G4 to let it wait a bit.

Problem now is G4 has this line in it:
LCD_MESSAGEPGM(MSG_DWELL);

This is going to destroy any attempt to put a message on the Display followed by a G4 command.

thinkyhead, 2015-04-25:
I'm merging #1966 which will prevent showing the "Sleep..." if another message is already set. Check and see that it works for you.

AnHardt, 2015-04-25:
Interesting here is that we came up with three different solutions. All of us felt her we should do something. After using a system for a while we get blind for the little nags, like this.

Salandora, 2015-04-25:
Actually I'm not home will update firmware when home and check if it works.

thinkyhead, 2015-04-25:
@AnHardt So true. At the moment I'm chasing down a "little nag" where the LCD messages and LCD click aren't always behaving quite right, but so close… I also notice that I discover and fix more things when I'm actually _using_ the printer, so I must keep finding more things to use it for.

Salandora, 2015-04-26:
Works great, can be consideres as fixed.

closed 2015-04-26 by Salandora

thinkyhead, 2015-04-27:
:muscle::alien: