MarlinFirmware/Marlin #1971: total print time ( again )

morgan-it, 2015-04-26:
Nice Feature Request:
should be nice to have a counter for total print hours of a printer, it should be displayed in lcd or on serial via an Mcode
i do not think is too difficult, and i think could be very useful

AnHardt, 2015-04-26:
@morgan-it 
What is your definition of print time? 
First activity - last activity?
Switching the power off is a bad choice for the last activity - no power to write the value away..

daid, 2015-04-26:
https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/lifetime_stats.cpp
Just leaving this here.

morgan-it, 2015-04-26:
i mean:
how many days, hours, minutes, seconds of printing time has my printer
yes Daid i think you you are right

morgan-it, 2015-04-26:
Daid i have tried to compile ultimaker2marlin firmware but gives me some error

CoderSquirrel, 2016-04-23:
Sorry to reopen it after a long time. But has it been managed?

thinkyhead, 2016-04-24:
Before I evaluate this concept I need to understand better:
- How many write cycles can the EEPROM handle before it fails?
- Can writing to one "cel" in the EEPROM affect the lifetime of other "cels" in the EEPROM?
- Is this even something I should concern myself with?

daid, 2016-04-24:
@thinkyhead see the link I posted. As I've implemented this for the Ultimaker2, and did the checks and math on it.

jbrazio, 2016-04-24:
I can't understand why this is so demanded.. but ok it won't hurt adding it to our feature list.

@daid do you know if only the offset being bytes written degrades or all the EEPROM space degrade at the same pace ? I can see the print counter accumulator to be written much more than once an hour.

And I can also see the angry mob with torches and pitch forks the next time we upgrade the EEPROM layout.. as this may become become a "pride token" among the users.

daid, 2016-04-24:
@jbrazio the EEPROM degrades per byte.

bobc, 2016-04-24:
I did some EEPROM life tests for work. We use a conservative figure of not more than 1 write per hour to stay within 100,000 cycles over a 10 year life. We just write the EEPROM on an hourly timer.

In destructive tests, I got to several million cycles before EEPROM started to show wear, but that was only one sample under particular conditions (voltage/temperature).

I guess if you do once per hour, or after "end of print" (maybe 15 minutes inactivity?) that should be fine for a 5 year life.

To avoid losing hours, I suggest a "flush" command that can be use at end of print. Otherwise, if the user prints for 50 minutes and powers off every time, no hours get recorded.

That is pretty much the best you can do with standard EEPROM, without going to FRAM or a large NOR Flash.

jbrazio, 2016-04-24:
Thanks for the insightful comment.

CoderSquirrel, 2016-04-24:
@daid @thinkyhead @jbrazio  i just asked it because i was looking the gcode file that Cure makes to sd and i saw that it has a line which is writen the time of the print, and rather than making math wouldn't be better to just use this variable?

jbrazio, 2016-04-24:
Could you post the gcode ?

CoderSquirrel, 2016-04-24:
The machine setting was a mender, and this gcode i've made just for test. For look what was inside.

> 
> M190 S70.000000
> M109 S195.000000
> ;Sliced at: Sat 23-04-2016 16:46:59
> ;Basic settings: Layer height: 0.15 Walls: 0.8 Fill: 20
> ;Print time: 6 minutes
> ;Filament used: 0.266m 0.0g
> ;Filament cost: None
> ;M190 S70 ;Uncomment to add your own bed temperature line
> ;M109 S195 ;Uncomment to add your own temperature line
> G21        ;metric values
> G90        ;absolute positioning
> M82        ;set extruder to absolute mode
> M107       ;start with the fan off
> G28 X0 Y0  ;move X/Y to min endstops
> G28 Z0     ;move Z to min endstops
> G29;
> G1 Z15.0 F9000 ;move the platform down 15mm
> G92 E0                  ;zero the extruded length
> G1 F200 E3              ;extrude 3mm of feed stock
> G92 E0                  ;zero the extruded length again
> G1 F9000
> ;Put printing message on LCD screen
> M117 Printing...
> 
> ;Layer count: 79
> ;LAYER:0
> M107
> ;LAYER:1
> M106 S127
> ;LAYER:2
> M106 S255
> G1 F2400 E-1.00000
> G0 F9000 X88.311 Y97.323 Z0.600
> ;TYPE:WALL-INNER
> G1 F2400 E0.00000
> G1 F1020 X88.611 Y97.321 E0.00748
> G1 X88.611 Y98.416 E0.03480
> G1 X88.311 Y98.417 E0.04228
> G1 X88.311 Y97.323 E0.06957

After this was the rest of the gcodes printing, i don't thinks is necessary in this case

jbrazio, 2016-04-24:
You speak about the "print time" comment ? That's not accurate, it's what the slicer thinks it will take.

This is an easy implementation, we are just concerned with the ERPROM durability in the micro controller as it has a limited write cycle count. But has explained before we should be ok with it.

CoderSquirrel, 2016-04-24:
@jbrazio yeah i know that its not accurate, its only estimated, and as you guys was worried about it thougth it was a good way.

thinkyhead, 2016-04-25:
@jbrazio This would be independent of the EEPROM layout. We would store this info either before EEPROM index 100 (that range is unused currently) or at the very top of the EEPROM.

jbrazio, 2016-05-04:
This has been implemented in PR #3625

PR #3625 (2016-04-29): Print job statistics
Fixes #1971

This PR introduces Print Statistics, by default they are disabled, to turn them on the user must `#define PRINTCOUNTER` in `Configuration.h`.

```cpp
//
// Print Counter
//
// When enabled Marlin will keep track of some print statistical data such as:
//  - Total print jobs
//  - Total successful print jobs
//  - Total failed print jobs
//  - Total time printing
//
// This information can be viewed by the M78 command.
```

A sample output of the M78 command:
```
M78
Print statistics: Total: 3, Finished: 2, Failed: 1, Total print time: 3h 2min
ok
```

Note: Sti [... cut]
closed 2016-05-04 by jbrazio