MarlinFirmware/Marlin #1976: Move Axis options (feature request)

ntoff, 2015-04-27:
I don't know why the options are limited depending on which axis you choose but it would be nice to have the 10mm option available to the Z and E axis, as it is now it's only available to X and Y. Also it would be nice to have 0.1mm and 0.05mm options available to all axis for people trying to calibrate / level the bed manually without having to plug a laptop in.

NarimaanV, 2015-04-27:
https://github.com/MarlinFirmware/Marlin/pull/1424 For the Z it can apparently cause problems since it buffers a lot of movement quickly since the Z moves much slower than the X and Y, and for those that don't have software endstops enables, you risk crashing the Z axis. The E I'd say isn't dangerous, but some people might get upset if they forget they're in the 10mm move section and move 40-50mm of material instead of 4-5mm (can't be wasting filament now can we?). I agree about having 0.05mm movement options at least for the Z (don't know it's application in the X and Y). If you really want the 10mm option for Z and E (because I doubt it'll get added since I also got shot down for asking this a while back), just go into ultralcd.cpp, find these lines:
https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L690-L700

and change it from this:
```
static void lcd_move_menu_axis() {
  START_MENU();
  MENU_ITEM(back, MSG_MOVE_AXIS, lcd_move_menu);
  MENU_ITEM(submenu, MSG_MOVE_X, lcd_move_x);
  MENU_ITEM(submenu, MSG_MOVE_Y, lcd_move_y);
  if (move_menu_scale < 10.0) {
    MENU_ITEM(submenu, MSG_MOVE_Z, lcd_move_z);
    MENU_ITEM(submenu, MSG_MOVE_E, lcd_move_e);
  }
  END_MENU();
}
```
to this:
```
static void lcd_move_menu_axis() {
  START_MENU();
  MENU_ITEM(back, MSG_MOVE_AXIS, lcd_move_menu);
  MENU_ITEM(submenu, MSG_MOVE_X, lcd_move_x);
  MENU_ITEM(submenu, MSG_MOVE_Y, lcd_move_y);
  MENU_ITEM(submenu, MSG_MOVE_Z, lcd_move_z);
  MENU_ITEM(submenu, MSG_MOVE_E, lcd_move_e);
  if (move_menu_scale < 10.0) {
  }
  END_MENU();
}
```
Did that for my own personal printer (just for the Z) so I know it works. Just remember the warning about buffering up too many moves in the Z and risking crashing it if you give it to many 10mm moves before realizing where it should stop.

thinkyhead, 2015-04-28:
I've often thought that the Z axis movement shouldn't begin until a half second (or so) after you have stopped rotating the controller. And, if you start a new move, the Z movement should stop right away, and wait until you pause again before moving to the new Z position. Perhaps this can be implemented with some ease...

NarimaanV, 2015-04-28:
I like that idea! That way, you can turn it as much as you want, then as soon as it approaches an end, just turn the knob the other way to basically tell it "Ok stop now". Just as long as nothing glitches and makes turning the other way fail to stop it, that sounds like a pretty solid idea. :D

ntoff, 2015-04-28:
>(because I doubt it'll get added since I also got shot down for asking this a while back), just go into ultralcd.cpp, find these lines:

I've had it on mine for a while now which is why I thought I might suggest it. I use it quite a bit.

Also is there an option to manually adjust the home offsets via the LCD? I know you can press a button to *set* them, but once set you can't change them without connecting via a host program, can you?

NarimaanV, 2015-04-28:
I personally have no idea, I've actually never had to use homing offsets (yet), so somebody else will probably have to answer that for you. Have you actually tried setting it in the LCD then trying to change it later via the LCD?

thinkyhead, 2015-04-29:
@ntoff The __Set Home Offsets__ menu option subtracts `current_position` from the `home_offset` array. So, first you do an __Auto Home__ command (which sets `current_position` to negative `home_offset`.) Then, move the axes with the LCD options or host, and then use "Set Home Offsets" from the LCD to subtract the `current_position` from the `home_offset` and set `current_position` to X0 Y0 Z0. To clear `home_offset` you just __Auto Home__ then __Set Home Offsets__ right away.

thinkyhead, 2015-04-30:
I'm adding a new gcode, `M428` that will set the home offsets based on the current position. I'm not sure how I want to implement it, because it's currently set up only to consider offsets from (0, 0, 0) and I feel that it should be set up to use offsets from (X_HOME, Y_HOME, 0) instead. Z would be offset from 0 because it's the offset from the bed that matters.

Some of the homing code could be made much simpler if the behavior of these printers was to always go to (0, 0, 0) after hitting their end-stops!

AnHardt, 2015-04-30:
To step back a bit further in this discussion.
How about turning the nob - and noting happens - until you push the nob - for the absolute movement?

How about different absolute and relative movement-menus to move the axis - per push?

    Move relative
      move 10mm
        x +10mm
        x -10mm
        y +10mm
        y -10mm
        z +10mm
        z -10mm
        e +10mm
        e -10mm
      move 1mm
        x +1mm
        x -1mm
    ...

or alternative

    Move Relative
      x +100mm
      x +10mm
      x +1mm
      x +0.1mm
      x -0,1mm
      x -1mm
      x -10mm
      x -100mm
      y +100mm
      y +10mm
    ...

besides @NarimaanV : the if statement can be omitted if empty.

thinkyhead, 2015-04-30:
Movement only on knob-click is an interesting idea. It makes it a lot more controllable. Another idea is to just ignore knob rotation until the current move is actually finished.

NarimaanV, 2015-04-30:
@AnHardt Yeah I know, I just thought to leave it in their in case whoever wanted to use that decided to remove Z or E from the 10mm move options again. Of course if it was something like a pull request I'd be sure to remove an empty if statement. Also in terms of the idea rotating the knob and the printer moving after you press in the knob sounds nice, the only thing it makes me worry about is back to the idea of accidentally making the printer move too far without realizing it (similar to how the Z buffers movements now).

Another idea to just throw out there, this would probably only be applicable to the Z axis, but what if there was just a "move up" or "move down" option where once selected, the axis would move in that direction until the knob was pushed in. Of course that wouldn't be for any precise movements, but just if one wanted to move the Z up a "bunch" just to clear something or if they wanted to raise their Z all the way to the other end of the home without having to turn the knob like mad. Of course the user would run into some serious problems if they forget to push the button again or walk away for a moment thinking they have time to do something else while they wait only to come back a little too late. We could build in some sort of timer that determines the absolute longest time an axis (probably Z mainly) would take based on the speed and the distance and kick in to stop that axis if the timer ends, though some Z-axes move so slow that it could be a while before the timer kicks in like if you started half way between ends and things wouldn't turn out so well then.

ntoff, 2015-04-30:
To move up a "bunch" I just go to the 10mm menu and move Z up by around 50mm or so. That's why the 10mm Z option is so handy for me. Having to move it up & down in 1mm increments is an absolute pain when working on the printer. I mean we're all so gung ho about SD printing and removing the PC connection to the printer for standalone usability but the LCD menus lack quite a few basic features and some features it does have are clunky and a pain to navigate (like the power on / off being buried in a sub menu).

I'm happy with the menu being the way it is now with move on rotate, not really sure you'd need a 100mm option though. I like being able to just quickly turn the knob to get a whole bunch of movement out of it when testing for friction or whatever. Soft limits (or in my case physical min / max switches) should limit the printers movements if it tries to move outside the physical boundaries.

I wonder if there should be some kind of warning about moving it before homing. Like if the printer has just been turned on, hasn't been homed and you try to move it, should it have a message that says "Warning: Printer not homed" in the move menu? In those cases if it's already at say Z100 and you try to move to z max, the soft limits won't stop it because it thinks it's 100mm lower than it already is.

thinkyhead, 2015-05-01:
Since these manual movements cannot occur during printing, it should be fine to "take over" the steppers and stop them whenever it makes sense. For example, if you spin Z up to 50mm and then change your mind and spin the controller back down to 0, the motor should reverse the moment that the displayed Z value is less than the current Z position.

dposea, 2015-09-06:
How about add a serial terminal to the lcd, so you can G1 Z100 and skip the knob

thinkyhead, 2015-09-07:
@dposea A serial terminal on the LCD would be cool, but of course entry would be by knob so it would be pretty tedious to type in gcode!

HerrWeber, 2015-12-20:
Hi, im just starting with marlin and 3d printers. i added some menu items to the move axis menu. 

  MENU_ITEM(submenu, MSG_MOVE_50MM, lcd_move_menu_50mm);
  MENU_ITEM(submenu, MSG_MOVE_10MM, lcd_move_menu_10mm);
  MENU_ITEM(submenu, MSG_MOVE_1MM, lcd_move_menu_1mm);
  MENU_ITEM(submenu, MSG_MOVE_01MM, lcd_move_menu_01mm);
  MENU_ITEM(submenu, MSG_MOVE_001MM, lcd_move_menu_001mm);

i added the MSG_MOVE_* to the language_en.h and created the move functions.
they work perfectly, but when using the lcd_move_menu_001mm the second decimal won't show

static void lcd_move_menu_001mm() {
  move_menu_scale = 0.01;
  lcd_move_menu_axis();
}

when i choose this option in the full graphics smart controller i have only 1 decimal place so i cant see the change in the lcd although the printer it is moving correctly. can you tell me how to add that second decimal place?

KiteLab, 2015-12-20:
On the INFO screen there is no additional space to display that additional digit.
On the edit screen you could use a display function with more digits.
In _lcd_move()
```
  if (lcdDrawUpdate) lcd_implementation_drawedit(name, ftostr31(current_position[axis]));
                                                        ^^^^
 ```
If you can't find one (below line 1920) you can try dtostr().

HerrWeber, 2015-12-21:
oh wow, many thanks @KiteLab ! im now using ftostr32. now as i reed the ftostrXX functions i wonder why aren't you using the stdio.h sprintf function, or the snprintf function. only reason i could find is that they could require much more processing power and the arduino is not capable of such thing. if not, why is that?

Wackerbarth, 2015-12-21:
The stdio routines are much more general than the few specific operations that we require.
With that generality comes an "expense" in computational complexity, requiring both more mpu cycles and more memory.  Since both of these are at a premium in our application, we gain by having a less general, but more efficient, function to meet our requirements.

closed 2016-05-04 by jbrazio