MarlinFirmware/Marlin #1978: Improve G29 topology report to include topology after corrections.

CptanPanic, 2015-04-27:
I understand that the auto bed correction can only handle corrections in a flat plane.  I would find it useful if as part of the topology report (G29 T) not only displayed raw topology but also the calculated topology utilizing the bed level correction matrix. This way you could easily see if there is not flat plane problem with bed that wouldn't be fixed by matrix, and also the magnitude of the issue.

Is this how it work? You would to loop through each x,y,z position in the topology report, and call apply_rotation_xyz and report the z?

brainscan, 2015-04-27:
G29 T can already be used to see how unflat/warped/bent/curved the bed is! Just increase the probe points and you can easily see the variation in the output. I don't understand how seeing the correction matrix would help anymore than what is already available?

nophead, 2015-04-27:
It's hard to separate the tilt from the curvature just by looking at
numbers. The post correction Z offset would be useful in showing how
un-flat your bed is.

Roxy-3DPrintBoard, 2015-04-27:
> G29 T can already be used to see how unflat/warped/bent/curved the bed is! Just increase the probe points and you can easily see the variation in the output. I don't understand how seeing the correction matrix would help anymore than what is already available?

I tend to agree.  But this is making the assumption that the bed is fairly well leveled and flat.  In that case, you get the points with the mean height subtracted off and you can see how much higher or lower a point is from the mean height.

In the case where the bed is not very well leveled and/or flat, I can see where this would have value.    But I almost wonder if it shouldn't be a separate command.  The reason is G29 generates the correction matrix (that is its real purpose in life).   To do what is being talked about very well may require many many more points to be probed.  In fact, it may be you don't want to do just a grid but instead have the logic to map topographical lines.

I don't know...   But it certainly worth some discussion.  And it would be an interesting piece of code to write if it really had some value.

CptanPanic, 2015-04-27:
So I tried to do an example in matlab, but am not getting numbers that look correct.  What am doing wrong?  I was expecting something around 0.  

% Matlab Code
left = 58;
right = 200;
front = 12;
back = 187;

num = 5;

x_pos = left:(right-left)/num:right; % Create array of x positions.
y_pos = front:(back-front)/num:back; % Create array of y positions

topology = [+0.72043 +0.14788 -0.41823 -0.96191 -1.49031; ...
            +0.95981 +0.30659 -0.24449 -0.75714 -1.32516; ...
            +1.04907 +0.48535 +0.03427 -0.53303 -1.08864; ...
            +1.11064 +0.62210 +0.15408 -0.33947 -0.68363; ...
            +1.24167 +0.86601 +0.41828 +0.08916 -0.36334];
        
correction_matrix = [+0.999898 +0.000000 -0.014282; ...
                     -0.000069 +0.999988 -0.004851; ...
                     +0.014282 +0.004852 +0.999886];

topology_cor = topology;

for y = 1:num
    for x = 1:num
        point = [x_pos(x) y_pos(y) topology(x,y)];
        correction =  [x_pos(x) y_pos(y) topology(x,y)] * correction_matrix;
        topology_cor(x,y) = correction(3);
    end
end

% Here is result.
topology_cor =

   -0.1662   -0.9085   -1.6443   -2.3577   -3.0558
   -0.3325   -1.1554   -1.8762   -2.5586   -3.2963
   -0.6488   -1.3823   -2.0031   -2.7401   -3.4654
   -0.9929   -1.6512   -2.2889   -2.9522   -3.4661
   -1.2675   -1.8129   -2.4303   -2.9292   -3.5514
`

nophead, 2015-04-27:
As well as the rotation matrix there should also be a Z offset. This is where Marlin gets vague. However, the result doesn't look right even with an offset, but I don't know Matlab. Do array indices start at 1?

Roxy-3DPrintBoard, 2015-04-27:
> Do array indices start at 1?

Yes, in MatLab they thought that made sense...

Roxy-3DPrintBoard, 2015-04-27:
I don't have a way to write this formatted correctly.   But I think your point[] matrix should be 1x3 meaning 1 across and 3 down.   If you change point = [x_pos(x); y_pos(y);  topology(x,y)];  that might fix the problem.   (I'm not a MatLab expert!  But I think you want to get point into a 1x3 instead of a 3x1 format)

CptanPanic, 2015-04-27:
Thanks for the suggestions.

1) So matlab will not allow multiplying 1x3 matrix by 3x3.  So have to use 3x1.

2) Also I tried doing the multiplication like vector_3.cpp does in vector_3::apply_rotation()
like 
topology_cor(x,y) =  x_pos(x) * correction_matrix(1,3) + y_pos(y) * correction_matrix(2,3) + topology(x,y) * correction_matrix(3,3); 
I get the same numbers.  

3) I also tried inverting the signs of the diagonals, but that didn't give me better looking numbers, just different.  Tonight when I get home I will try with the Marlin code.

Roxy-3DPrintBoard, 2015-04-27:
> 1) So matlab will not allow multiplying 1x3 matrix by 3x3. So have to use 3x1.

This doesn't sound correct.   I think it is 'correct' to multiply a [1x3] * [3x3] just as it is valid to multiply a [3x1] * [3x3]    Please double check this because this doesn't 'correct'.

> 3) I also tried inverting the signs of the diagonals, but that didn't give me better looking numbers, just different. Tonight when I get home I will try with the Marlin code.

No!  Do not invert the sign of the numbers going down the diagonals.  Try inverting the sign of all the numbers off of the diagonal.  If you notice, the matrix is symmetrical with the signs of the numbers flipped.

CptanPanic, 2015-04-28:
So I was thinking about this more, and I think I was thinking about it wrong.  What what help would be seeing the difference between the topology and the calculated flat plane.

CptanPanic, 2015-04-29:
So I got things working in matlab, and generated a couple of cool plots.  The image below shows the difference in mm between the topology and the flat plane.  As you can see there is about .4mm difference between highest and lowest points, so I will definitely need to work on this, especially pulling down the front right edge. Second plot is shows flat plane, and the topology points.

Next I will try adding this calculation to Marlin

[image]

[image]

Roxy-3DPrintBoard, 2015-04-29:
That is cool!

CptanPanic, 2015-04-29:
I had 2 things wrong; First off I had the x/y values were from the front/left even though the topology was from back/left.  Also I no longer applied correction to topology, but just to a constant z and then calculate the difference between this calculated z and topology.

brainscan, 2015-04-29:
Yeah I stand corrected that is cool.

alvinavr, 2015-05-01:
@CptanPanic 
Very good graph! Much easier to visualise than a table of numbers.

I have just modified  Marlin to output a table of numbers showing the residual error after bed levelling. This helps with improving the mechanics.

Looking at your picture, your Y-bars are not parallel. I know because I had the same problem.

Roxy-3DPrintBoard, 2015-05-01:
> I have just modified Marlin to output a table of numbers showing the residual error after bed levelling. This helps with improving the mechanics.

If we had a web site where we could automatically run the code to make the chart that would be very helpful.  What I'm thinking is Marlin would output:   http://some.website/some_page #1 #2 #3 #4 #5 #6 #7 #8 #9 etc.

The user could copy that long line of output and paste it into a web browser.   And then the page would magically get displayed with a detailed graph.

thanosd, 2015-05-02:
I don't remember where I came across this, but I think it does what you want, though perhaps not with the exact URL "API" you showed:

http://www.maui-3d.com/cgi-bin/plotG29?plotG29ID=772

thinkyhead, 2015-05-02:
I've been playing with THREE.js a bit, but if there's some already-built graphing lib that's your best bet. You can ask @Scotty1024 or @boelle to add it to marlinfirmware.org, which is the appropriate place for all this stuff.

AnHardt, 2015-07-17:
Played a bit with #2444
and realized that the map with the lowest point at zero is nice for graphical purposes but not for statistics.

Made some additions in https://github.com/AnHardt/Marlin_O/pull/11 to print out the Max, Min, Error_sum and Error²_sum of the topology blocks.

The exiting part of this is the calculation of the heights, to minimize the errors in the "Corrected Bed Height vs. Bed Topology:"-map.
```
diff = (eqnBVector[ind] - mean) - (z_tmp - avg2);
```
what means:
```
diff = (measured height[point] - mean_of measured_heights) -
(calculated_corrected_height[point] - mean_of_calculated_corrected_heights);
```

closed 2015-09-25 by AnHardt