MarlinFirmware/Marlin #1979: Converting from old marlin

boelle, 2015-04-27:
So had time to get the latest marlin installed

copy and pasting settings from the 2-3 year old copy of marlin got me going but now i'm a bit stuck

this old one:

#define X_MAX_LENGTH 221
#define Y_MAX_LENGTH 221
#define Z_MAX_LENGTH 390.65

// The position of the homing switches. Use MAX_LENGTH * -0.5 if the center should be 0, 0, 0
#define X_HOME_POS X_MAX_LENGTH * -0.5
#define Y_HOME_POS Y_MAX_LENGTH * -0.5
#define Z_HOME_POS Z_MAX_LENGTH

is no where in the new config....

should i use these?

#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 221
#define Y_MAX_POS 221
#define Z_MAX_POS 390.65

and how do i set them so that center of bed is 0,0 ?

nophead, 2015-04-27:
You make MIN_POS -110.5 and MAX_POS 110.5 to get 0 in the middle.

boelle, 2015-04-27:
mostly a note for self, but when i ran pid auto tune for the bed (not an option in old marlin) it ended up with this:

PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
#define  DEFAULT_Kp 176.72
#define  DEFAULT_Ki 13.59
#define  DEFAULT_Kd 574.51

have it not forgotten bed after the underscore ?

boelle, 2015-04-27:
almost there.... 

i can do G28 and it homes.... but it will try and move past the endstops.... 

config as of now: http://pastebin.com/zPG4Pefw

cant figure if something should ignore the endstops

boelle, 2015-04-27:
maybe this in configuration_adv:

#define ENDSTOPS_ONLY_FOR_HOMING // If defined the endstops will only be used for homing

?

nophead, 2015-04-27:
For Mendel90 the endstops are outside the print volume, so once homed the
soft limits should prevent them ever being hit again.

boelle, 2015-04-28:
oh... forgot to set soft limits :-/

thinkyhead, 2015-04-28:
#1989

boelle, 2015-04-28:
:-D the man is on fire... fast fix

boelle, 2015-04-28:
btw.... will prop be able to test auto level to late tomorrow or weekend. only need to rebuild my extruder and do some minor mod's to my electronics

closed 2015-05-07 by boelle

thinkyhead, 2015-05-08:
@boelle Great. Maybe we can do some video-based leveling debugging…. :beetle:

boelle, 2015-05-08:
yep, just need to wire up the servo and touch switch

boelle, 2015-05-08:
home at last....

Q: are there a M119 equilant for testing the touchprobe?

i have changed the pins file for my board so it now looks like this:

http://pastebin.com/HXz4bAPy

i know i use  M280 PX SX to set the pos of the servo

thinkyhead, 2015-05-08:
M119 might work for the probe. If not we need to add it in.

boelle, 2015-05-08:
oki.. btw my config: http://pastebin.com/mrXadjJ1

boelle, 2015-05-09:
first problem... servo does not move

sure its a software problem  but not sure where

boelle, 2015-05-09:
M119 does not report status on the z probe either... only XYZ

boelle, 2015-05-09:
hmm... doing autotune for the bed and hotend

bed was ok

but not during hotend autotune the extruder fan led flashes together with the hotend

boelle, 2015-05-09:
M106 and M107 work fine thou... so not a wiring issue

bed cooling fan can also be controlled manual

will check if everything else works but assume so far that they do

nophead, 2015-05-09:
Does your fan share the 12V wire with the heater, as per Mendel90? If so
maybe the heater is wired with the wrong polarity, in which case turning it
on will turn the fan off.

boelle, 2015-05-09:
Yes, i have chekced it thou and work fine when i turn things on and off in pronterface.

will check again thou

boelle, 2015-05-09:
doooh.... yes wired in wrong :-( *slap in face*

boelle, 2015-05-09:
nope it was right... me not reading schematics right

boelle, 2015-05-09:
*slap in face* turned out check i double checked it was wired in wrong

still wondering why the servo will not move

fmalpartida, 2015-05-09:
I'll try our tonight the new de branch during the hangout. On Thursday the
M851 command was broken.

boelle, 2015-05-09:
well i have just downloaded the lastest copy but have no clue why the servo is not moving

boelle, 2015-05-09:
fresh copy on my config: http://pastebin.com/9hXCtrku
pins file: http://pastebin.com/2BYjaQj8

boelle, 2015-05-10:
as promissed an update.... 

i updated to latest IDE and figured how to get the old MCU/AVR in there... then did a compile
then i found out that in fact i only needed to add the bootloader file as the fuse settings turned out to be the same. it took at lot of compare using an online fuse calculator. first entering everything for the old MCU... then selecting the new one and efter the same settings... what a relief.

then i only had to edit my pins file (pins_ omca .h something) so that it would not say upps if i picked the new MCU or the old one...

now i only have to replace the chip itself and burn bootloader etc

thinkyhead, 2015-05-11:
Good news! I haven't gone anywhere near the bootloader stuff (yet) so I still find it quite mysterious.

boelle, 2015-05-11:
now bad news:

avrdude: verification error, first mismatch at byte 0x1f000
         0xff != 0x0c
Error while burning bootloader.
avrdude: verification error; content mismatch

boelle, 2015-05-11:
scratch it... i went back to ide 1.0.6

at least i can burn bootloader for the 644 there and upload a sketch...