MarlinFirmware/Marlin #1980: main fan stops at low PWM speeds

CONSULitAS, 2015-04-27:
The main fan stops running at low PWM speeds (K8200 at about 35 to 40 of 255), even if kickstarted with FAN_KICKSTART_TIME.

I would say this is a bug (even if hardware dependent) because the user - and host software or slicer - would expect the fan to run if set to speeds higher than 0.

So the fan should be driven to run at least at minimal possible speed when fan speed is set to 1 or other "to slow" value.

Can be easiely reproduced: Start fan manually, lower speed to small values greater 0, fan will go slower and slower until stop. You can hear PWM sounds from fan indicating that it is still powered.

AnHardt, 2015-04-27:
@CONSULitAS 
It's quite normal for a fan to stop some when he does not get enough power.
My best fan still turns with 15, an old worn one needs 120. It also depends on your power supply and your MOSFETS. Since we do not support (until now) fans with pulse feedback, we have no chance to detect if a fan is turning or not. So, where to set a minimum value?
I think here the user is responsible to search for a value that works for his system. And when a value is found - don't go below.
Kickstart may help to let your fan turn for a while with 35. But when kickstarttime is over it will stop.
Without kickstart it may not even start to turn with 45. KickSTART only helps to start the fan.

CONSULitAS, 2015-04-27:
@AnHardt Sleepless in Ruhrgebiet? Me sleepless in Würzburg, today... :-)

Correct. Please have a look at the PR #1981 mentioned above. This solves the problem. Simple solution, configurable, working.

You are right: It depends on your hardware, but jerk, acceleration and so on, too. I would suppose giving a reasonable standard and let the user fit if necessary.

PR #2118 (2015-06-07): add FAN_MIN_PWM for slow fan stopping issue (V2.1 remade)
replaces #1981 (and #2117 :smile: )
closes #1980

The main fan stops running at low PWM speeds (K8200 at about 35 to 40 of 255), even if kickstarted with FAN_KICKSTART_TIME.

If fan speed is lower than FAN_MIN_PWM (and not 0) it is set (dynamically) to FAN_MIN_PWM to run at least on minimal possible speed.

@thinkyhead Thanks for the great idea with linear scaling speed beetween FAN_MIN_PWM and 255!
closed 2015-09-25 by AnHardt