MarlinFirmware/Marlin #1990: Mulitple extruders on one stepper motor

dglassinc, 2015-04-28:
I'm switching from an older Marlin to this newer version. Previously we have run E0 and E1 using the same output pins without issue. I've been tinkering with this new version and have found that when more the one E has the same pin setup neither stepper motor will function. I switched the pins back to different outputs and I was able to drive the stepper motor as normal.

emartinez167, 2015-04-28:
Exactly what I had been dealing with Brian previously on the HPX2-MAX. Have
you guys resolved this then?

dglassinc, 2015-04-29:
No we are still unable to call out the stepper motor pins as we did in the older version of this firmware. We are hoping for some support on this issue.

thinkyhead, 2015-04-30:
I'd personally rather see formal support for these setups added rather than just have the pin numbers be hacked and hope it works (as you're very lucky it did in the first place). Simply setting the pins the same is not really supportable in the long-term.

There are extruders, similar to the HPX2-MAX, that flip a servo to switch the feed, keep the same stepper motor, but reverse its direction. All of those actions are doable within the current code-base, but someone who makes and/or owns these extruders will need to help Marlin support them as a real feature. Which means, Marlin would know to use the same extruder motor (by index) for the second extruder, rather than having no formal notion that it is doing so.

In the long term I would like to see Extruders as formal objects in Marlin, where each extruder has zero or more stepper motors, zero or more nozzle+heaters, zero or more servos, zero or more thermal sensors, zero or more measurement sensors, and zero or more binary sensors.... Then when you switch "Extruder" it might pick one of two extruders that share the same stepper motor, but the Extruder object would know how to activate itself – perhaps switching a servo and setting a flag to reverse the motor direction, for example.

MagoKimbra, 2015-04-30:
thinkyhead unfortunately I'm not very good with github and how well you understand even myself understood with English. But it is time that I am telling you if you want you can see my marlin where the number of extruders is independent of the number of drivers for extruder and the number of hotend. I currently printing with 4 extruders, but with a single driver and a single hotend.

emartinez167, 2015-04-30:
MagoKimbra, would you mind sharing the code so it can be incorporated into the main code?

Regards,

Ernesto

thinkyhead, 2015-05-01:
@MagoKimbra I have of course reviewed your changes and I have been looking forward to incorporating them once this current bug hunt is over. It's a good start in the right direction!

closed 2016-05-04 by jbrazio