MarlinFirmware/Marlin #1994: EEPROM needs structure

atunguyd, 2015-04-29:
I have been looking through the ConfigurationStore.cpp and in my opinion the structure of the EEPROM  should  to be reconsidered.
Currently the structure is defined by the order that variables are written and the size of each variable and the presence of each variable. While care has been made to add dummy variables to ensure the size and location is constrained it opens up the chances for error.

I suggest that the EEPROM structure is maintained using a C structure with adequate padding for future variables, would also be a good idea to implement a CRC routine to detect bad EEPROM values so that they are not loaded blindly into the firmware.

atunguyd, 2015-04-29:
Hi Dev team,, just to comment on my comment above, I do not mean to come across as bullish so apologies in advance if the comment does sound this way, I have been involved in embedded development for over 15 years working on set top boxes for large  pay-tv customers. In this area we always used EEPROM to store settings for many years and I have learnt first hand what errors can occur with a badly managed eeprom structure.

I am happy to put my hand up to attempt to help clean up this area of the code if needed.

Wurstnase, 2015-04-29:
Do it when you think it's good and make a PR. Than we can discuss, merge or throw it away :) That's my way.

atunguyd, 2015-04-29:
No problem Wurstnase, let me try to take the time to see what I can do, I must admit though that I cannot get my head around git as a version control system (I have only ever used soursesafe and then svn) so it will take me some time to figure out how to do this (inbetween my real work :) )

clefranc, 2015-04-29:
@atunguyd I've played with the EEPROM functions some time ago, I totally agree with you, the current design is prone for too much errors. I'll definitively take a close look at your PR.

atunguyd, 2015-04-29:
Hi Clefranc,

Ideally the way I see it is that all modules should not use global variables for their settings and rather all use a global structure. For example:

typedef struct {
   float Kp;
   float Ki;
   float Kd;
} stPIDSettings;
typedef struct {
   float retract_feedrate;
   float retract_zlift;
   float  retract_recover_length;
} stRetractSettings;  
   
extern struct {
    stPIDSettings ExtruderPidSettings[4];
    stPIDSettings BedPidSettings;
    stRetractSettings  RetractSettings;
    .
    .
    .
    .
    uint8_t spare [150];
    uint8 CRC;
} stEEPROM_Settings;

Then individual modules use this external structure stEEPROM_Settings and deference each parameter they need.   If they change a setting they update the structure and call configuration_store.cpp who is responsible for recalculating the CRC and storing this into eeprom. It is also responsible at startup to read the structure from eeprom and check the CRC, then if there are any issue to reseed the structure with valid default values. 

This however is a big change and will require changes to a lot of modules so it would need to be done with caution.

You will notice I added a spare array, this allows one to add extra fields (decrementing the spare) without breaking the CRC of existing settings, if spare is always set to a known value (say 0xff) it will be possible to detect whether a new value needs seeding at startup or not. One can also get clever with the use of a union to force the CRC to the end of the structure and ensure byte alignment of other structure elements.

thinkyhead, 2015-04-30:
A structure is fine, but consider stealing the EEPROM code from Repetier instead, which uses a human-readable format and thus is able to remain completely backward-compatible without a lot of length-checking and so forth.

As for a CRC, data corruption hasn't really been a problem. If the EEPROM fails to finish writing it leaves the version set to "000". It is extremely rare for EEPROM writing to fail that badly.

Most recently I've made changes to `configuration_store.cpp` so that all the values that can be stored in EEPROM can also be set with GCode. So, if you take the report output from `M503 S0` you can run it back and reproduce all your EEPROM settings. Naturally, I recommend using the same approach going forward, as it also makes settings portable to other firmwares.

This would also allow the `Config_PrintSettings` and `Config_StoreSettings` functions to use mostly shared code.

atunguyd, 2015-04-30:
Thinkyhead, certainly using the M503 S0 values as is is an options but would that not end up using double your variable stack size (which means more ram usage), because you would need to create a variable or array the size of the M503 S0 parameters to read and write to EEPROM and thereafter convert these to the external global variables (which in turn are taking up heap space).This would also not be a very efficient use of eeprom space since you are storing your variables as ASCII encoded.

I also think this would result in more opcodes (ie more program space) whereas if all external variables are in a fixed structure the copy and read routines are all a block copy.

While write failures are rare on eeproms it is still prudent to provide some sort of error detection, when you have a million or more of your products out there the chances of a power failure during an eerom write is largely magnified and you would be surprised to see what a badly initialized variable can do to some code (after all no one can claim that they write code to take into account every eventuality).

thinkyhead, 2015-05-01:
@atunguyd I can think of a few ways to reduce the storage space required. For example, just off the top of my head, first modify `Config_PrintSettings` to "sprintf" each line into a single-line buffer before sending it to serial. Next, structure `Config_PrintSettings` so that you can request just a single line of output (as a pointer), and if you request a line higher than the number of settings, you get back `NULL`. Then, `Config_StoreSettings` can loop from 0, getting each line and writing it to EEPROM, until it receives NULL.

thinkyhead, 2015-05-01:
The output `M503 S0` on my Prusa i3 with a few options comes out to only 220 bytes.
```
M92 X80.00 Y80.00 Z4000.00 E1258.14
M203 X500.00 Y500.00 Z2.25 E45.00
M201 X9000 Y9000 Z100 E300
M204 P1500.00 R1500.00 T3000.00
M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
M206 X0.00 Y0.00 Z0.00
M301 P20.83 I1.04 D104.71
```

closed 2016-05-04 by jbrazio