MarlinFirmware/Marlin #1996: Code Sprints - Anyone?

boelle, 2015-04-29:
This idea was vented by @thinkyhead a long time ago

Thought i would bring this up again

I think the idea was to do sprints once a month

While changing thermistors on my printer i thought why not do these every 2 weeks and then alternate between a google hangout and code sprints? so that we have 2 hangouts a month and 2 codesprints 

Is the idea total bonkers?

@Wurstnase @AnHardt @nothinman @thinkyhead @alexborro @tmilker @stv0g @galexander1 @daid @ErikZalm

any of the above that have time to spare for this?

people with only stiring up an endless debate without contributing do no need to reply

thinkyhead, 2015-04-30:
_A sprint is a get-together of people involved in a project to give a focused development on the project. Sprints are typically from one week up to three weeks. Sprints have become popular events among some open source projects. For example, the PyPy project is mostly developed during regularly held sprints where most of the international developer team gathers.[1]

The sprints are often held near conferences which most of the project team attend, but they can also be hosted by some involved party at their premises or some interesting location.

Sprints are organized around the ideas of the Extreme Programming discipline of software development. The sprint is directed by the coach, who suggests tasks, tracks their progress and makes sure that no one is stuck. Most of the development happens in pairs. A large open space is often chosen as a venue for efficient communication.

Sprints can vary in focus. During some sprints people new to the project are welcomed and get an intensive hands-on introduction pairing with an experienced project member. The first part of such sprints is usually spent getting ready, presenting the tutorials, getting the network setup and ensuring that configuration/source control software and processes are installed and followed.

A significant benefit of sprinting is that the project members meet in person, socialize, and start to communicate more effectively than when working together remotely._

boelle, 2015-04-30:
The Meet in person might be a problem on this one...

I live at least a days constant drive from @AnHardt  and 2 days drive from @fmalpartida 

I was more thinking if we could do it with a google hangout in some way.... ie talk on the hangout and figure who takes what bug and for those who want to code in piece can part from the hangout... those that dont mind can stay

some bugs might need 2 persons who might benefit from the hangout

fmalpartida, 2015-04-30:
Hi folks,

I think that we can do "sprints" (lets call them iterations) to get this going. I must be one of the few folks at work that use agile processes with my teams and they are not necessary linked to "extreme" programming. Extreme Programming (XP) is just an agile process. 
Sprints (iterations) are applicable to agile development cycles such as scrum, kanban, AUP, XP (is just one more agile process), ... Anyway, enough engineering talk. As our objective should be to agree to achieve incremental, evolutionary change/objectives and new feature introduction.

To define a "sprint" (iteration) we have to set up an objective or set of objectives which I think will help focus the effort.

What we have to recognise is that we are starting to have an unhappy community and as such we need to organise what we publish and how we publish it. We also have to communicate well what's the contents of each branch has, its short comings and purpose.

I do think that we should be running at least two separate public branches. A development branch where we merge in the work and do a "code" freeze to test. Nothing goes in that branch that is not bug fixing derived from the testing activity. Once "cleared" (iteration is complete) the code is merged/pull-in to the release branch. This release branch is then labeled and published. I would also suggest to publish or zip it for the public. 

Normally, in large developments there are 3 main branches: release, integration and development, including a sand-box for trying out "high risk features". I don't think we need so much formalism here though.

What do you think?

For me the first sprint's objective should be to iron out anything that may be broken with autoleveling and deliver that into the release branch. I will do some testing over the weekend on the current development branch and let you know how that is going.

@boelle if you drive 1 day and I drive another one we can meet round about Paris. However, we are starting summer here so this weekend is beach or swimming pool time. Perhaps you may want to drive 2 days :-)

boelle, 2015-04-30:
hehehe....

i can agree to 2-3 branches... that is the only reason i got the owner stamp back so i can create those

I can also agree that the first goal must be to iron out autolevel. I might go and re-arrange the milestones so that those comes first (if i have not allready done so)

fmalpartida, 2015-04-30:
Thumbs up!

Can we create the "release" branch where we will pull in the release as soon as we get the autoleveling polished? This should be our main focus in the very short term. We can then focus on additional milestone/sprint/iteration items.

By the way, where are the milestones?

boelle, 2015-04-30:
https://github.com/MarlinFirmware/Marlin/milestones

and yes, to create a branch is a few clicks for me

fmalpartida, 2015-04-30:
Cool Bo, thanks! I will take a look at the whole picture to try and get my head around it. 

Perhaps one boundary condition. No new features or code refactoring pulled into the development branch till we get the first release out. If not we will never finish and will have a load of people pissed off.

boelle, 2015-04-30:
that is why a lot has the label "on hold..... "

fmalpartida, 2015-04-30:
Thumbs up!

thinkyhead, 2015-05-01:
One big effort that would benefit from a sprint is the object-izing of the code-base, wrapping all the main entities in simple classes and using access privileges to clean up the global namespace.

fmalpartida, 2015-05-01:
I agree, another one is the HAL.

thinkyhead, 2015-05-02:
Ah yes, we very much need that [Hardware Abstraction Layer](http://en.wikipedia.org/wiki/HAL_(software)) too.

closed 2015-06-29 by boelle