MarlinFirmware/Marlin #2002: Move documentaion to webiste (marlinfirmware.org)

boelle, 2015-04-30:
I got this idea of having all documentation in wiki format on the site and use github for the code only....

who has any constructive objections ?

@scotty1234 did you get my request for an account?

Scotty1234, 2015-04-30:
I think you have the wrong person!

boelle, 2015-04-30:
yep... it was meant for @scotty1024

scotty1024, 2015-04-30:
Bo,

You know that combination of letters and numbers you use with an @ on the front? Move the @ to the end and add me dot com. I will get you setup via email through that address.

boelle, 2015-04-30:
oki... did a request for a login via the page and had to figure a 50 word bio... 

but will send a mail anyways

scotty1024, 2015-05-01:
What? Should have been a 100,000 word bio!

I sent you a ssh account. If you need a wiki web account I will need to set that up too.

AnHardt, 2015-05-09:
I'm not sure if this is the way to go.
With this restrictive polity the idea of an wiki is not matched anymore.
I expect more contributions on the GitHub-Wiki.

boelle, 2015-05-09:
the goal is to have all documentation in one place and not a mix between github and reprap.org

and everyone that wants to contribute can get an pass/user from @scotty1024 

it was the whole idea behind getting the domain name in the first place

boelle, 2015-05-09:
see you all tonight on hangout

closed 2015-05-11 by thinkyhead