MarlinFirmware/Marlin #2003: Need a pause button

columbus1231, 2015-04-30:
I am an artist hobbyist and I use a 3d printer running on Marlin firmware to create my art. 
I am trying to hack Marlin to enable me to add a pause button to pause the printer while printing. 
I need to be able to edit my workpiece anytime I want during the print job, and then resume the printing after I'm done editing.
I will do this using a while loop that tell the printer to do nothing while the pause button has been put to high, but I don't know where to put the loop.  
Can anyone help?

Roxy-3DPrintBoard, 2015-04-30:
This sounds much more complicated than you seem to think it is.  First, there are many GCode commands buffered in Marlin's memory.   But then you are going to pause the print so you can edit the work piece and send a new GCode file to Marlin?   

How do you propose you know where you currently are in the new GCode file?

columbus1231, 2015-04-30:
Thanks for your response. To answer your question, I don't intend to do any changes to the g-code. 
The editing I intend to do on the workpiece are all done manually by hand, because they are too complex for the printer to perform.
The printer will need to be paused while I do these edits so my hand doesn't bump with the printers moving parts.
So the printer will continue with the g-code left off when I hit the pause button after I release the pause button.

Roxy-3DPrintBoard, 2015-05-01:
OK...   IF that is the case, it should be fairly tame to do.   And it should be possible to ignore the buffered up GCode commands by just not processing them.   I'm thinking you would just grab control of things and at the end release the printer to go.   With that said, if you are planning to do things from an LCD panel, they jam commands in the GCode buffer.   So, either the buffer would have to be run dry first, or another scheme would have to be done to leave those in place waiting until you are done.

thinkyhead, 2015-05-01:
I recommend editing the GCode and inserting `M0` commands where you want it to pause. `M0` allows you to set a custom message on the LCD controller, so you can give yourself cues.

    M0 Bend the thing
    ...
    M0 Insert the widget

Roxy-3DPrintBoard, 2015-05-01:
Yeah M0 makes a lot of sense.    I might have misunderstood:

> I need to be able to edit my workpiece anytime I want during the print job, and then resume the printing after I'm done editing.

Are you thinking you want to 'edit' the print by manually controlling the printer via the LCD panel?   That is much harder to do.   And M0 won't allow that because the printer is already in the middle of a command.

columbus1231, 2015-05-01:
Thanks. I agree M0 seems to be the one I will try that could work. I will have to find a way to put M0 on my g-codes.

Roxy-3DPrintBoard, 2015-05-01:
> Thanks. I agree M0 seems to be the one I will try that could work. I will have to find a way to put M0 on my g-codes.

The easiest way is to pull the GCode into a text editor.  And then search for something like:

G1 Zx.x.x     when it switches levels.   Presumably, at some level height you know what you want to do.  So when it switches to that level, you add in a line that will pause the machine.  This is the same strategy used to do multiple colors on a single extruder machine.   You tell it to pause at a specified level, do the filament change, and then tell it to continue.

brainscan, 2015-05-01:
I haven't used repetier host in a while but you can do that easily as it has a built in search for layers to insert code.

closed 2015-05-02 by thinkyhead

columbus1231, 2015-05-02:
Thanks for the good ideas. I will try repetier and see.

artbody, 2017-04-05:
Hy to all
I think this "pause button" is in some cases more than important.
Example:
I'm planing a printer which prints with a fluid (70°C hot) and another which should print with a special plaster.
For that I have a reservoir which is filled cyclically
2 valves control the flow
A Stepper pushes this out of the reservoir through the printer nozzle
Two electrical switches should work fine
one switch to tell the Printer to PAUSE and start to refill the reservoir
the other to tell the pump to stop refilling and resume from pause

Example 2
so an other Problem will occur when the **filament** at a **normal printer** reaches his **end**.
if you accidentally pass the moment to change the filament ... :-1: 
10 to 20 hours printjob To throw away ??? looks very professional

Sensor :+1:  IF filament empty GOTO PAUSE !!!!!!
PAUSE :save actual position: lift Z and goto xy 0 ; cool down the nozzle;  WAIT
Sorry if i don't understand why this is not part of marlin .

thinkyhead, 2017-04-07:
In the current `RCBugFix` there's an option to lift the head and move aside in response to "Pause SD Print" from the LCD (or `M25` from the host). There's no option to pause prints from the host in this manner (but hosts could supply that behavior) unless the `SDSUPPORT` option is disabled, and then you can use `M125` to pause a print with lift-and-move.

This new feature is a sub-option of `FILAMENT_CHANGE_FEATURE` called `PARK_HEAD_ON_PAUSE`. It doesn't currently have cool-down and re-heat as part of its behavior, but that behavior does exist in the `M600` handler, so it should be easy to duplicate in park-head-on-pause.

The filament sensor feature has an option to call a command on run-out. The default command is "`M600`."

lf-, 2017-07-20:
@thinkyhead Please clarify if there is a button on the LCD interface to pause the print for any reason and pick up later. For example, I might not have the time to watch the printer through a long job and want to continue it the next day. Is that possible?

fiveangle, 2017-07-20:
https://github.com/MarlinFirmware/Marlin/releases : 

#6407 : Unified M600 / M125 as ADVANCED_PAUSE_FEATURE