MarlinFirmware/Marlin #2004: Development Marlin compile error with Manual and Mesh Bed Leveling

Fixx, 2015-05-01:
When enabled, Manual Bed Leveling and/or Mesh Bed Leveling makes error at compile.

Arduino: 1.6.3 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from configuration_store.cpp:91:0:

configuration_store.cpp: In function 'void Config_PrintSettings(bool)':

Marlin.h:95:77: error: call of overloaded 'serial_echopair_P(const char*, int)' is ambiguous

 #define SERIAL_ECHOPAIR(name,value) do{ serial_echopair_P(PSTR(name),(value)); }while(0)

                                                                             ^

configuration_store.cpp:680:5: note: in expansion of macro 'SERIAL_ECHOPAIR'

     SERIAL_ECHOPAIR(" X", MESH_NUM_X_POINTS);

     ^

Marlin.h:95:77: note: candidates are:

 #define SERIAL_ECHOPAIR(name,value) do{ serial_echopair_P(PSTR(name),(value)); }while(0)

                                                                             ^

configuration_store.cpp:680:5: note: in expansion of macro 'SERIAL_ECHOPAIR'

     SERIAL_ECHOPAIR(" X", MESH_NUM_X_POINTS);

     ^

Marlin.h:97:6: note: void serial_echopair_P(const char*, float)

 void serial_echopair_P(const char *s_P, float v);

      ^

Marlin.h:98:6: note: void serial_echopair_P(const char*, double)

 void serial_echopair_P(const char *s_P, double v);

      ^

Marlin.h:99:6: note: void serial_echopair_P(const char*, long unsigned int)

 void serial_echopair_P(const char *s_P, unsigned long v);

      ^

Marlin.h:95:77: error: call of overloaded 'serial_echopair_P(const char*, int)' is ambiguous

 #define SERIAL_ECHOPAIR(name,value) do{ serial_echopair_P(PSTR(name),(value)); }while(0)

                                                                             ^

configuration_store.cpp:681:5: note: in expansion of macro 'SERIAL_ECHOPAIR'

     SERIAL_ECHOPAIR(" Y", MESH_NUM_Y_POINTS);

     ^

Marlin.h:95:77: note: candidates are:

 #define SERIAL_ECHOPAIR(name,value) do{ serial_echopair_P(PSTR(name),(value)); }while(0)

                                                                             ^

configuration_store.cpp:681:5: note: in expansion of macro 'SERIAL_ECHOPAIR'

     SERIAL_ECHOPAIR(" Y", MESH_NUM_Y_POINTS);

     ^

Marlin.h:97:6: note: void serial_echopair_P(const char*, float)

 void serial_echopair_P(const char *s_P, float v);

      ^

Marlin.h:98:6: note: void serial_echopair_P(const char*, double)

 void serial_echopair_P(const char *s_P, double v);

      ^

Marlin.h:99:6: note: void serial_echopair_P(const char*, long unsigned int)

 void serial_echopair_P(const char *s_P, unsigned long v);

      ^

Error compiling.

closed 2015-05-02 by thinkyhead

Fixx, 2015-05-05:
Still not fixed.

Arduino: 1.6.3 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Marlin_main.cpp: In function 'void gcode_M421()':

Marlin_main.cpp:4523:41: error: 'select_x_index' was not declared in this scope

     if (!err) mbl.set_z(select_x_index(x), select_y_index(y), z);

                                         ^

Marlin_main.cpp:4523:60: error: 'select_y_index' was not declared in this scope

     if (!err) mbl.set_z(select_x_index(x), select_y_index(y), z);

                                                            ^

Error compiling.

thinkyhead, 2015-05-06:
#2027 patches the code to fix the method namespaces