MarlinFirmware/Marlin #2009: No move after homing

prophetmaster, 2015-05-02:
hello i have uploaded the latest version of marlin in my delta and after G28 no move except extruder, anyone have encoutered this problem ?

Marlin firmware 1.0.3 dev

boelle, 2015-05-03:
there are no ver 1.0.3... only 1.0.2...

where did you get 1.0.3 from?

fmalpartida, 2015-05-03:
It seems that there have been some boards being updated that have version 1.0.3 dev on them.

thinkyhead, 2015-05-03:
@boelle The version reported currently from `Configuration.h` is "1.0.3 dev" to distinguish the Development HEAD version from the 1.0.2 tagged release.

@prophetmaster The code for `G28` has been undergoing some changes lately. For the most reliable delta behavior you may want to stick with the @jcrocholl fork of Marlin for everyday use. If you want to help track down potential problems with homing (and leveling) I have [a logging version of Marlin](https://github.com/thinkyhead/Marlin/tree/debug_leveling) for that purpose. The output logs from that version may help pinpoint where homing is getting stuck.

prophetmaster, 2015-05-03:
ok good news, it is a mess can manage firmware between development and stable we do not know or seek anything ...

KerwoodDerby, 2015-05-06:
Discovery: going from vendor-supplied old Marlin firmware to current version, I didn't realize at first that the max endstops were now being heeded (mine aren't installed), so they always were appearing triggered by the firmware. I grounded those connections, and then got the desired behavior out of G28 (and G29 later).

thinkyhead, 2015-05-06:
@KerwoodDerby You could just change the `X`, `Y`, and `Z_MAX_ENDSTOP_INVERTING` settings so that it considers those non-existent end-stops to _never_ be triggered.

KerwoodDerby, 2015-05-06:
@thinkyhead Indeed so, but I was just trying to illuminate the problem for the original poster. It had me scratching my head for hours because I had gone from a working setup to one which appeared broken, and I had nearly convinced myself something was wrong with the firmware.

thinkyhead, 2015-05-06:
@KerwoodDerby That's actually good to know, too. I wouldn't expect my endstops to change between Marlin versions. We might want to clear that up before the official 1.0.3 release.

thinkyhead, 2015-05-09:
@prophetmaster Still seeing this problem?

3machine, 2015-05-11:
Hello, I have the same issue with Marlin 1.0.3 dev, Last Updated: May 10 2015
I use a rumba board with drv8825 and 3 min endstops 
After a g28, all axes move +5mm , no homing 
it is impossible to go in the negative direction

3machine, 2015-05-11:
currently retrying with marlin-release

3machine, 2015-05-11:
changing the X, Y, and Z_MAX_ENDSTOP_INVERTING settings worked fine

thinkyhead, 2015-05-12:
@3machine Are you saying that by changing the `X`, `Y`, and `Z_MAX_ENDSTOP_INVERTING` settings fixed your issue with 1.0.3 dev? (The HEAD of marlin-release is at 1.0.2).

prophetmaster, 2015-05-15:
your solution not resolving my problem...

thinkyhead, 2015-05-15:
@prophetmaster What do you get with `M119`? Please test with the latest code. https://github.com/MarlinFirmware/Marlin/tree/Development

prophetmaster, 2015-05-15:
after G28:

the max endstop is triggered by the homing

Send: M119
Recv: Reporting endstop status
Recv: x_min: TRIGGERED
Recv: x_max: TRIGGERED
Recv: y_min: TRIGGERED
Recv: y_max: TRIGGERED
Recv: z_min: open
Recv: z_max: TRIGGERED
Recv: z_probe: open

with no homing :

Send: M119
Recv: Reporting endstop status
Recv: x_min: TRIGGERED ???
Recv: x_max: open
Recv: y_min: TRIGGERED ???
Recv: y_max: open
Recv: z_min: open -> z-probe
Recv: z_max: open
Recv: z_probe: open

G29 : no move

Recv: G29 Auto Bed Leveling
Recv: -75.40 -45.40 -35.40 -25.40 14.60 -5.40 4.60 14.60 4.60
Recv: -15.40 4.60 24.60 34.60 44.60 54.60 64.60 64.60 64.60
Recv: 54.60 74.60 84.60 94.60 104.60 114.60 124.60 134.60 134.60
Recv: 134.60 144.60 154.60 164.60 174.60 184.60 194.60 204.60 214.60
Recv: 214.60 224.60 234.60 244.60 254.60 264.60 274.60 284.60 294.60
Recv: 294.60 304.60 314.60 324.60 334.60 344.60 354.60 364.60 374.60
Recv: 374.60 374.60 384.60 394.60 404.60 414.60 424.60 434.60 454.60
Recv: 444.60 444.60 444.60 454.60 464.60 474.60 484.60 504.60 524.60
Recv: 504.60 494.60 504.60 514.60 494.60 534.60 544.60 554.60 584.60

thinkyhead, 2015-05-16:
@prophetmaster If you have no min endstops then try `DISABLE_MIN_ENDSTOPS` so the machine will ignore them. Or, change the `[XYZ]_[MIN/MAX]_ENDSTOP_INVERTING` settings for any endstops that are showing up as triggered when they should not. In your case, change both `X` and `Y_MIN_ENDTOP_INVERTING`.

prophetmaster, 2015-05-16:
I have activated #define Z_PROBE_ENDSTOP and changed pin 32 to 18 for me all are right in my configuration...

on G28 the Z cariage go down no it the endstop ???

thinkyhead, 2015-05-16:
@prophetmaster Trial and error will perhaps get it figured out. Keep us posted if you find the cause.

AnHardt, 2015-09-25:
Probably something wrong with the configuration. Maybe homing direction.

closed 2015-09-25 by AnHardt