MarlinFirmware/Marlin #2010: Extruder motor eventually stops working

diegotrap, 2015-05-03:
My extruder motor works for a while (several prints), but then invariably it stops moving. I suspect it may be related with issues #1378 and #1414.

Contrary to the previous issues, this happens after several succesful prints. It's a firmware problem since all hardware components have been switched. It is solved by changing the Arduino board (I use RAMPS) with a fresh Marlin instalation (even of the same version).

thinkyhead, 2015-05-03:
The EEPROM issues from earlier this year were largely cleared up. Does the output of `M503` look different when it's working versus when it isn't?

diegotrap, 2015-05-03:
I didn't want to make it work again to try to gather information about the problem, so I can only get the not-working report. The current M503 output is

  00:53:09.280 : echo:Steps per unit:
  00:53:09.288 : echo:  M92 X80.00 Y80.00 Z4000.00 E104.00
  00:53:09.288 : echo:Maximum feedrates (mm/s):
  00:53:09.289 : echo:  M203 X200.00 Y200.00 Z5.00 E25.00
  00:53:09.289 : echo:Maximum Acceleration (mm/s2):
  00:53:09.289 : echo:  M201 X3000 Y3000 Z100 E10000
  00:53:09.293 : echo:Accelerations: P=printing, R=retract and T=travel
  00:53:09.297 : echo:  M204 P3000.00 R3000.00 T3000.00
  00:53:09.302 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum   E jerk (mm/s)
  00:53:09.305 : echo:  M205 S0.00 T0.00 B20000 X15.00 Z0.40 E5.00
  00:53:09.305 : echo:Home offset (mm):
  00:53:09.309 : echo:  M206 X0.00 Y0.00 Z0.00
  00:53:09.309 : echo:PID settings:
  00:53:09.309 : echo:  M301 P22.20 I1.08 D114.00
  00:53:09.313 : echo:Filament settings: Disabled
  00:53:09.313 : echo:Z-Probe Offset (mm):
  00:53:09.313 : echo:  M851 Z-0.90

AnHardt, 2015-09-25:
As the OA is pretty alone with this problem - closing.

closed 2015-09-25 by AnHardt