MarlinFirmware/Marlin #2014: [Request] Stop changing serial commands and messages on a whim

foosel, 2015-05-04:
Case in point, [this specific change](https://github.com/MarlinFirmware/Marlin/commit/d43cc2dd5f52c9b71b486e7778d7fb44ea67fad6#diff-744b3dbd56ea076f353b4ed94eb96271L125). Also the recent change to the behaviour of `G28` that thankfully got rolled back again.

We do not have a proper transport protocol in place here (the GCODE RepRap Wiki page is NOT a well defined protocol definition!). That means that as a firmware you have to be very very careful when making backwards incompatible adjustments to your external interface, which your serial interface is. Host software has to work with what is there in terms of a "protocol". Changing messages like the one in the commit linked above (especially if hidden like that in an unrelated commit) can have consequences beyond Marlin. 

"But the hosts do very ugly things they shouldn't be doing!" - I agree! And I hate it too! But do you know the reason? Because there is no proper transport protocol so host developers have to constantly depend on workarounds and dirty tricks to make the more sophisticated functionality requested and expected by the users work at all without them having to understand G and M commands and even touch a serial terminal (and btw - make all that work across firmwares, so not only with Marlin but also with Repetier, Smoothie, etc). Changing commands or messages on the serial interface out of the blue like this does NOT help in making our job easier, so please be a bit more considerate! At the *very least* stop hiding stuff like this in the mix and start documenting it properly. 

Thanks in advance!

daid, 2015-05-04:
I'm with @foosel here.

@thinkyhead Do NOT. EVER. EVER change the error messages. I know the current situation sucks, but if you change things without talking things over with the host software developers you will cause chaos.

@foosel Shall I give you rights on this fork so you can revert stupid changes like this and wack people on their heads when they do this?

foosel, 2015-05-04:
@daid I'm tempted to say "Yes please" just to reduce the amount of headaches the recent events have been causing me.

Booli, 2015-05-04:
+1 vouching for control by @foosel from a host perspective.

Roxy-3DPrintBoard, 2015-05-04:
> +1 vouching for control by @foosel from a host perspective.

+2 !

clefranc, 2015-05-04:
Is there such open source protocol for serial communication with a 3D printer? Can it be based on a existing RFC or ISO? If not, who should define the protocol for communication between the open source firmwares and proprietary hosts? The firmwares coders? Hosts coders? An independent committee? Will he ever had any effort to create a standard serial messaging. If yes, what can we learn from it?

I'm just thinking out loud...

foosel, 2015-05-04:
FWIW, I got so fed up with the lack of a proper protocol and protocol definition and the constant problems that are caused by that situation that I recently started working on a baseline to get one crowdsourced, together with @jnesslr. Hoping we'll have something to go public with soon so that we can all make the world a better place in that regard inside the RepRap community. Focus is the open source world though. IMHO proprietary projects can go play in their own sand box for all I care.

daid, 2015-05-04:
Proprietary hosts can figure it out on their own. But we are all FOSS devs here I think. And we don't want a gazillion different versions of the protocol to support.

wolfmanjm, 2015-05-04:
+1 for @foosel  FWIW the "spec" does say debug information should be prepended with // and the shoul djust be displayed on the host and ignored, why don; tMarlin developers use // for the debug output...

http://reprap.org/wiki/G-code#Replies_from_the_RepRap_machine_to_the_host_computer

daid, 2015-05-04:
@wolfmanjm that "spec" was written after the fact. And does not even match the Marlin implementation used the last 3 years...

wolfmanjm, 2015-05-04:
@daid That maybe true but I think it is a good idea, and FWIW Smoothie is now using that concept.
// is supported by both pronterface and octoprint at the moment and seems to be a good way to display debug info that is not part of a reply to a specific command.

emartinez167, 2015-05-04:
+1

Regards/Saludos,

Ernesto

clefranc, 2015-05-04:
@foosel Can we take a look at your protocol?

How should it work? A simple protocol definition or a complete implementation, abstraction, supporting all processors, 3D printers, hosts? Should it be only for serial communications? A Web API or any other board connectivity? A separate repository or let firmware coders follow the guideline?

Still thinking out loud...

thinkyhead, 2015-05-05:
I love it when the system works. Good to see everyone too – hi!

linchunguang, 2015-05-05:
Power to continue playing solution how to write

foosel, 2015-05-05:
@clefranc As soon as it's in a presentable state, right now it's mostly creating an outline and nailing down some basics like ABNF syntax and basic communication specifics. 

The idea is to not create a new protocol, it is to finally formalize what is already in place today in regards to the communication that is taking place over our printer's serial interfaces. Of course, since we don't have an agreed upon standard (otherwise nothing would need to be done) that will mean that a compromise will need to be found on things that currently differ between the several implementations out there (e.g. format of `M105` replies, command output before or after `ok`, or even inline, etc). Also let me make clear, I'm not working on "my protocol". I'm really just trying to bring order into the chaos that is the current situation both firmware and host developers find themselves in and which is causing problems not only for the developers but also the users.

I wouldn't limit it to serial interfaces -- if a printer speaks TCP natively, that shouldn't matter (although things like checksuming wouldn't be necessary in such cases since thats already taken care off by the underlying transport layer).

ntoff, 2015-05-05:
+1,2,3,4, and 5. 

That recent G28 change debacle had me kind of angry to be honest since, out of the blue, my printer no longer raised up before printing which meant head crashes. That could be very costly on the Mendel since it's all solid mounted and a PEEK hotend, luckily I have a more flexible printer to test things on so head crashes just bounce off the springy bed.

Roxy-3DPrintBoard, 2015-05-05:
> That recent G28 change debacle had me kind of angry to be honest since, out of the blue, my printer no longer raised up before printing which meant head crashes. That could be very costly on the [...]

Changes should not be made to the code that is being distributed without having it tested first.  I don't understand the unwillingness to roll back the code to the point where everything worked.   I've been told my time would be more 'productively used' to help debug the sick and broken code than to complain about the practices that got us into this state.

Mean while, we see threads saying "Testers Needed!"     Good luck with that!  My printer can't even print using the current broken code.

foosel, 2015-05-05:
> To be fair, the code is being tested, it's just that the main branch you see when opening marlin on github happens to be the development version.

Given the amount of permutations due to the sheer number of `#ifdef` statements in the whole code base, I fear that the current "does it compile" tests on Travis don't even cover a double digit percentage of the actual possible variations.

gregrebholz, 2015-05-05:
I agree that having a "Release" branch which is always at the last tagged commit, 1.0.2 in this case, and marked as the default branch in github, would help a lot of users.  Developers will be familiar enough with git to change branches before working, and the act of doing so should put most users on notice that they're headed into the untested wild.   The tagged "releases" at the top still serve a purpose, to go back to prior releases, but people that just download from github and try compiling will be doing so against the last tagged release instead of whatever latest commit happens to be.  @boelle -- thoughts?

@Roxy-3DPrintBoard This may address what you referred to as a reluctance to roll back.  Unless a commit is wholesale "bad", it shouldn't be rolled back because there's something good in there worth keeping.  A new commit fixes the problem and continues to move the project forward.  But that's only true of a "Development branch" where this sort of thing is supposed to happen.  A "Release" branch would only ever contain the tested and vetted releases that Marlin developers want the world to be running.  The current Marlin project is a Development branch, and tagged releases which general users can download for use.  Our users don't seem to be catching on to that if they're getting angry about individual developer commits with limited testing (at most their own machine).  They're probably cloning the Marlin github repo ("Development branch") and then compiling whatever code is the latest today.  Head crashes ensue.

daid, 2015-05-05:
@gregrebholz if a single commit contains more then 1 change then the commit is flawed anyhow. See for example the change that triggered Gina this time. It was a hidden message change inside a change that added a debug flag...

daid, 2015-05-05:
@gregrebholz depends on who made the commit, what the commit is actually trying to accomplish. And if I want to be an asshole that day (which is usually a yes)

Roxy-3DPrintBoard, 2015-05-05:
> This may address what you referred to as a reluctance to roll back. Unless a commit is wholesale "bad", it shouldn't be rolled back because there's something good in there worth keeping. A new commit fixes the problem and continues to move the project forward.

Pretty much...  Everything with regard to the Broken G28 and Broken G29 has been with the attitude "Let's try this." or "Let's try that."   There isn't much of value.  Rolling it all back and starting fresh makes the most sense.

And really...  That is the whole reason for using a Revision Control System.  If there is some thing else of value, well...  It can be added back after we roll things back to the point they work.

boelle, 2015-05-05:
not any thoughts... i have just been trying to keep head and tails in the issues and not worried much about anything else

boelle, 2015-05-05:
but a +1 from me to give @foosel collab access. maybe even hope that she can help out getting a few bugs sorted

ntoff, 2015-05-06:
@gregrebholz 

> Our users don't seem to be catching on to that if they're getting angry about individual developer commits with limited testing (at most their own machine). They're probably cloning the Marlin [...]

My issue and "anger" came from the suggestion that they had "fixed" G28 to "work as it should" when in fact if you look at CNC references G28 doesn't work that way at all.

http://www.cnccookbook.com/CCCNCGCodeG28ReturntoReference.htm

> It's function is to return to the machine's reference position, sometimes called the zero position. That zero return position is where most progams begin, most machines will go to this position when you manually home or reference the machine, and it is the reference or zero position for calculating fixture offsets for mills and geometry offsets for lathes.

So the claims that it was now somehow fixed, were in fact even MORE broken.

Not to mention no consideration was ever given to every host program out there would need to be modified in order to work with the "fix".

>getting angry about individual developer commits with limited testing (at most their own machine).

Then perhaps they should ask for more testers before committing it to the main code repository. Just sayin'.

Wurstnase, 2015-05-06:
Because in a cnccookbook it is written that a G28 X5 should move, it is broken in Marlin?

CONSULitAS, 2015-05-06:
I think there is more than one interpretation of G28 if you read the page up to the end:
>Exercises
>1. Get out the programming manual for your machine and see how G28 is supposed to operate.

But: Shouldn't we stop this flame war and go back to work?

* We have a "Development" branch. This means "alpha state" of the software.
* Alpha Software **has** issues.
* If you use Software with issues, be carefully.
* Don't complain, but help.
* But it would be good to point simple users to which "stable" version to download.

foosel, 2015-05-06:
@CONSULitAS I hope that even in the alpha software case we agree on changes in behaviour and protocol interpretation need to be limited and at least be documented. Because once alpha software graduates to beta and later stable, without consideration these changes will possibly enter mainstream unnoticed (as has happened before).

boelle, 2015-05-06:
about host protocols... i agree, we have enough trouble as is... and i can only really think of a few:

1) Repetier
2) Pronterface

3) Octoprint if that one does not use one of the 2 above

foosel, 2015-05-06:
Hosts are not the ones defining the communication protocol - or let's say API here, since a protocol would imply something that is well defined, which we do not have here ;) Firmwares are. So Marlin, Repetier, RepRapFirmware, Smoothie, Sprinter and what else I might have forgotten there or just not know about yet.

boelle, 2015-05-06:
wish for xmas: one protocol to rule them all :-D

like it would be more simple if every country in the world had the same language

JustAnother1, 2015-05-06:
@boelle It must not be only one protocol. Supporting more than one protocol on the host is doable. The issue here is that with no protocol at all it is unclear if an issue is caused on the host or on the client. And a host can only be implemented to talk to the client as it is. Once a bug in the client is fixed this might then be also fixed on the host. So having a good specification of a protocol would be very good.

I can understand the comments about the reprap wiki page. But as IMHO that is the best we currently have wouldn't it make sense to work on that wiki page to bring it into a state where it is actually usable?

It is a wiki so modification is doable, right? Or is there something that makes this a bad idea?

CONSULitAS, 2015-05-06:
@foosel Hello Gina,

please don't misunderstand me. You are right complaining about the broken protocol. I am using the development branch for my K8200, too, and would be rather angry if it would be broken by a firmware bug...

But perhaps it it would be better to stop the discussion *here* because it is not really about "Stop changing serial commands and messages on a whim" any more.

I think we should break down the discussion to their different aspects and open issues like "Feature request: Cleanup host protocol and make it more reliable" or "update documentation for host protocol on reprap wiki".

What do you think?

Greetings Jochen

PS: It is not so easy to discuss in english, and not been misunderstood, if you are not a native speaker. :-)

PPS: Will try octoprint asap...

fmalpartida, 2015-05-06:
Hi folks,

I think we covered this topic two weeks ago on the hangout where I suggested to have a set of "branches" depending on the maturity of the releases and even a sand box to tryout experimental features before bringing them in.

We discussed about having a release branch with labels that then end up in the release repo, this in essence should be the user facing default branch with the latest stable code. Where people normally land, take the zip and use, in addition to the actual releases.

We then discussed about having an integration branch where things are tested out in preparation for the release. Changes are pulled here after testing by development and where tests can be run with different configurations. 

Finally, the development branch where things get changed and are unit tested.

The goal of the integration branch was to have an intermediate branch from the development branch and  the release (lets call it beta). This is where people more eager to engage early-on can give feedback on a "pre-release" and find a more or less reliable code base they can use.
We also discussed that it may be a little too over kill, but after reading this thread and several others I am still wondering if it is not the way to go. We have to recognise that the current code base is reasonably complex with a wide range of conditionally compiled options which make the code a bit "don't touch too much or you will break something else" in addition to a wide range of support for LCDs, boards, kinematics, features, ... with huge dependencies between source areas. To make things worse, there are more and more variants coming along which could make things even worse. Testers will and are limited to their configurations, but having a beta branch will give the chance to integrate and try things out before they are released.
I also suggest that the only effort that should drive us right now was to construct release 1.0.3 without introducing further changes to the code base. Minor isolated bug fixes, OK, but certainly not interface changes nor further code refactoring.

Finally, I would also like to mention interfacing. Wether we like it or not, Marlin has stablished a "standard de-facto" as far as host interfacing is concerned and we have to recognise that changing what the firmware is answering to a command may break something elsewhere (mainly on a host controller).
Personally, I am not a defendant of keep things as are, however, intentions to changing interfaces and behaviour have to be communicated adequately. At least give people to synch, touch basis on and have time to provide feedback and react.

I think we have a good opportunity to take action and take advantage of what we have (git) to introduce a little bit or order and meaning to what is expected from each branch.

Sorry for the lengthy post, I felt that sharing what we discussed and agreed on was on topic to avoid having lengthy entries like these and focus on features, enhancements and refactoring.

Cheers,

fm

thinkyhead, 2015-05-06:
Ideally, it would be nice if no bugs ever crept into the Development branch, but… it's the _Development_ branch! No one is forcing anyone to download and run the code that was checked-in today, which (in spite of best efforts) may introduce some new bug – and that could be fixed an hour later, for all anyone would know.

As for the resistance to "rolling back" the code, actually, yes, I have done the tedious work of rolling back the code. I have a branch that rolls back all the leveling and homing changes to mid-March. I _took the time_ and _made the effort_ to find all the effective changes that were made since the merge of Johann's fork occurred, before any complaints about `G29` began to appear. If you want the code from that point, you can get it with:

    git checkout 89fe774~1 -b before_delta_merge

I pulled that code and compared it to the current code. I applied all the new code which would have no effect on leveling or homing. I produced a new branch at that point and then applied the remaining changes in one merge. The differences between the old code and the new code are therefore embodied in this commit: https://github.com/thinkyhead/Marlin/commit/99977ef36e2f659ff2d35884fc7983616c81a410

I would like to get more help from developers running development code on their robots and providing feedback. Personally, I make more progress clearing up issues when people are generous enough to run the code I am providing, answer the questions I pose, who try things and figure out fixes.

From a distance it might seem that the code is just irredeemably broken and should just be rolled back. But when you run the code and observe the logs, it's clear there's just one silly error somewhere. I hope at least one or two persons in this thread will take the time to check out the commit https://github.com/thinkyhead/Marlin/commit/99977ef36e2f659ff2d35884fc7983616c81a410 and perhaps give my pre-delta branch https://github.com/thinkyhead/Marlin/tree/before_delta_merge and run it to see if it really does embody the code before the `G29` glitches appeared.

thinkyhead, 2015-05-06:
I'm all for updating the serial standard so that it can produce results that hosts can understand – but also be translatable. So, for example, you may still have `echo:` and `err:` as prefixes, similar to now, but then we also add an error (or status) number, which is either Marlin-specific, or shared with other firmwares. After the number there would be a translatable text message, which hosts simply display.

Roxy-3DPrintBoard, 2015-05-06:
> From a distance it might seem that the code is just irredeemably broken and should just be rolled back. But when you run the code and observe the logs, it's clear there's just one silly error somewhere. I hope at least one or two persons in this thread will take the time to check out the commit.

Thank you!   I'll load this code as soon as my print is done.

daid, 2015-05-07:
@thinkyhead No. No numbers. Numbers suck for a protocol like this. "keys" as a future, that might work.
```
ERROR: HOTEND_ON_FIRE: Your hotend might be on fire right now.
```
Everything in upper case is "fixed" the rest is translatable.

But change nothing without notifying the host software producers first. (Pronterface, Repetier, Octoprint, Cura are the major ones I think)

Also, your stance on the development branch is bad. Really bad. Yes, there could be bugs in there, but there should be made maximum effort to get no bugs in there. And people working on the Marlin code should be made clear that buggy code is not acceptable.
Why? **The code is safety critical. A small mistake will ruin your printer, a big mistake could burn your house down.**
I know of 4 printer fires, of in which once case an house got lost.

**As for the Wiki page**
As I stated before. The wiki page on GCode was filled after the fact. It's just a random mash of what different firmwares sort of are doing, in some cases actually quite inaccurate.

ntoff, 2015-05-07:
>To make things worse, there are more and more variants coming along which could make things even worse.

I wonder when it's time to say "ok, no more support"? I mean if everything is basically just an atmega2560 with different configurations for the pinouts, why is it up to Marlin devs to support hardware manufacturers who want to deviate? Every hardware designer and their dog trying to get a slice of the pie, it should be up to THOSE people to hand out the customised firmware. Velleman do it for the K8200 board + LCD panel they sell.

IMHO the firmware here should provide base level support for the hardware. Support for moving the steppers, displaying things on the screen, reading the SD card, etc. If someone wants to go and change the pinouts on their board just because they think they're clever then it should be up to them to supply a pins.h file for you to use.

I know it sounds shitty, but I guess that comes back to the "no standards" thing, since there's no real standard for pinouts.

JustAnother1, 2015-05-07:
@daid "*As for the Wiki page
As I stated before. The wiki page on GCode was filled after the fact. It's just a random mash of what different firmwares sort of are doing, in some cases actually quite inaccurate.*"

I understood you the first time. There are many standards that have been written after the fact. I just wanted to ask the question if it might make sense to improve the wiki page, or if it is doomed.

A first step could be to fix the inaccurate things. Maybe a second step would be to add a documentation of the "new standard" way. What I mean is:

**right now:**
G something
Marlin: does this
Teacup: does that

**new version:**
G something
Marlin: does this
Teacup: does that
standard : do exactly this

This is just a suggestion. I hope it makes sense. I just think that the easiest way to get to a common standard is to build on something that is already there and doesn't need to be changed too much.

daid, 2015-05-07:
Don't think it would be wise to try to put a standard on a page that everyone can edit as they want.

JustAnother1, 2015-05-07:
They have a discussion page and a history. So the changes are traceable. I don't see a problem. You migth want to talk to them. Maybe they can introduce the features that you think we will need.

The question where to put it if not on reprap? If you put it on ultimaker.com it will be the ultimaker-GCode standard. That might make it harder for non Ultimakers to contribute. Is there another central authority for open source / open hardware 3d printing?

foosel, 2015-05-07:
Fully agree with @daid. Everyone being able to edit around is what got us into this mess in the first place. A wiki is NOT a good place for having the definition of a standard. A history doesn't help you at all if you can't figure out which changes were agreed upon and which were people just trying to get their will into the standard without any proper discussion first. Plus you can't create proper versioning (like "protocol version 1.2") if everything is constantly shifting because everyone and their mother edits around. It just doesn't work. You need a selected few who curate the spec (as in, have actual edit access, *not* dictate it, just curate it), and in our case ideally some crowd sourced wisdom through public discussions about it.

My current approach:
* Wiki *with limited write access* but public discussions - *only* for the initial collaborative effort to get a definition done, the final spec would be read-only and static, I'd suggest something like hosting it on github pages ("reprapcode.github.io" or a custom (sub) domain even)
* Defined transport protocol as close to the current realities across the multiple firmwares as possible, basically a subset of all possible variants we have today - to minimize adaptation impact on both host as well as firmware developers
* Proper definition of requests *and responses* (this is something that is lacking nearly completely right now, leading to things like four or five different possible versions of a response to M105 alone)
* Properly defined points for vendor extensions (we need to end this "M4711 does X on Marlin and Y on Repetier" madness)
* Proper documentation of effects on the printer
* Responsibilities of the host & of the firmware in terms of request/response processing
* Long term: Proper test suites to allow hosts and firmwares to test their compliance with

Now... I'm fairly sure I'm going to regret this because I'm still working on it, there's no discussion platform in place yet and everything is still a bit in the flow while some thought sorting is taking place, but [here's what I'm currently working on as mentioned above](http://reprapcode.haz.wiki/). THIS IS STILL IN THE RAMP UP PHASE. So don't even think about starting to try to get me into any discussions about the scope or the contents or "chapter 6, section II, third paragraph, fourth sentence, fifth word". My goal with this right now is to create a basis for discussion first and foremost (with the discussion part still lacking, working on that). But if you want to get an idea of what I mean with "well defined" requests and responses, take a look the M20 spec in that page.

clefranc, 2015-05-07:
@foosel Thank you for sharing your preliminary work. Looking at it show great potential for open source firmwares (or even proprietary ones). All I can do right now is offering you my moral support and give encouragements. This is not small task, but I can see it's in good hands.

Please, let us know when you'll be ready to open discussion about it, it'll be a very interesting times.

thinkyhead, 2015-05-08:
@daid Quality is Job #1. But I do mean to be a little bit _ominous_ about the Development branch, since it is the forefront – tested, yes, but not _widely_ tested – and at the moment the processes that determine which code gets committed are only _mostly_ transparent. I think it would be Really Good™ to get at least one or two corroborating testers to agree that a PR is okay before it can be committed, or in some other manner get more regular QA to happen across the distributed Marlin Testing Labs.

repetier, 2015-05-08:
Just found this thread through @Wurstnase and since I'm developing both firmware and host software I guess I should also share my thoughts.

First, I fully agree with @foosel that is is VERY bad to change command behavior after first introduction. Host software ignores unknown commands, that is easy, but once we find them and start supporting them we need defined behavior or at a maximum additional behavior that does not interfere. It is not possible to follow all changes in all firmware derivatives so doing so will break the one or other host function.

Again, Gina is right that a backwards compatible standard to the current situation would be great as a starting point. There will be always lots of users never upgrading a firmware - I know from help request that they don't or perhaps every 2 years. So all hosts are more or less doomed to support old protocols anyway. The big question is how to proceed from here. Years ago I thought firmware development would come to an end, but there are always users with new ideas, so we can never expect that additions will stop. What may be possible are some standards that define on how new features get added without breaking old stuff. And if possible also in a way allowing fast parsing of responses. Things firmware develop often forget is that the host needs to understand the response and that means a test like a regular expression and that takes increasing time with increasing response variations.

So some things I'd like mentioned here:
Start a response line with // when it does not need to be interpreted by host e.g. debug stuff. I already had cases where my debug return x:7 and that was then interpreted as x position.

At least new responses should follow a simple scheme
RESPONSE_TYPE:\s*ResponseValue[\s]
If we want to add some text for readability we might introduce a ! or ; as response comment start, so we ignore everything after that like in error messages. A command can return any number of responses in one row. This is more or less as it already is, except maybe that uniqueness is not guaranteed and e.g. M105 repeats some types so ordering starts to matter.

Then we need some rules for command/ack system. I lately see that Marlin starts also sending line number together with ok. I'm only a bit confused with the additional N before the line number. My hope was that Marlin would adopt the response Repetier is using since the beginning to make it easier, but I can live with that.

Anyhow what I want to say is if we define such simple basic rules for responses, each response is unique and easy to filter. If a command adds new responses it is no problem for existing hosts. They interpret what they know and neglect the new. Still enough stuff for a little chaos, as there is and most probably never will be a norming committee.

A similar default rule should be stated for a valid g-code.
(Nlinenumber\s+)?([A-Z]value\s)+(\s+TextString)*checksum
With each letter only appearing at most once.

If we all could agree to at least some basic rules like this for future additions live would be easier for all. 

Haven't read Ginas bnf docs fully, but I guess it might propose something similar as this plus all the backwards  compatibility stuff. So it might be a good starting point to get a better base standard and hope other firmware developers would start keeping to some rules when they are there. At least I've seen here firmware and host developers. Maybe we should also ask Lars from Matterslicer as well to join the discussion and who ever is responsible for Pronterface. Then we would have all developers of popular hosts here to make it even more a standard. Did I forget a important host software not already present here?

@foosel Let me know if you need some background information to repetier firmware for the wiki. Otherwise when you say you think you are finished I will try to understand and check agains errors/missed cases I might see.

fmalpartida, 2015-05-08:
> To make things worse, there are more and more variants coming along which could make things even worse.
> I wonder when it's time to say "ok, no more support"? I mean if everything is basically just [...]

> IMHO the firmware here should provide base level support for the hardware. Support for moving the steppers, displaying things on the screen, reading the SD card, etc. If someone wants to go and [...]

> I know it sounds shitty, but I guess that comes back to the "no standards" thing, since there's no real standard for pinouts.

@ntoff That was another topic we covered in the hangout.
There are three different types of things that are currently supported:
1. Different boards based on the ATMega that have a specific pin mapping.
2. Different printers that use the same board with pin mapping variants.
3. Examples, which are now printers that people have included to be supported directly to the repo.

We discussed that:
For 1, whoever wants a board to be supported (as long as they don't fall into category 2), they are accountable to introduce it in the Marlin in the "board" section. They are also accountable for trying the board support on every release and report back. In this category fall boards that are evolutions, but not variants.
In 2 we have different things, printers and boards which are variants of a first boards or boards configured for a specify printer. Perhaps just remove them and maintain their configuration in a board specific branch.
3, examples printer specific configurations which have cropped in so that they can be updated and maintain here. Here we discussed in removing the examples and making them a bit more generic or have some a bit more significant for different mechanical variants.

If 2 and 3 continue to grow (which will, as Marlin is a good show case) the variants and examples are going to start adding a fair amount of overhead + end up being the call center.

thinkyhead, 2015-05-09:
@repetier One way to help development go forward is to add a "legacy" option that will send old messages, plus the new ones that hosts will ignore. If legacy-mode isnt compiled in then it uses just newer messages that the newer hosts will understand. We also have to option to do more major version numbering changes (more strongly indicating changes in protocols) so for instance you can say "Repetier 1.1 requires Marlin 1.1 or later." We're meeting tomorrow to discuss —among other things— methods and tools to achieve better coordination within and across these projects.

repetier, 2015-05-09:
A legacy mode might help older hosts to understand or waiting until hosts understand. After all I guess it is always the firmware starting with changes, so no host will understand them at the beginning. Newer hosts would still need to understand old stuff since they then need to support legacy answers plus new answers, so even more to understand and parse. And why should we need to change a working answer or answer part. That is something where you misunderstood me a bit.

Marlin has echo: and Error: at the beginning. That is not what I meant with response type. In fact Repetier only has Error:. If it not an error, it is a response so echo never was introduced (like it was before in Sprinter where we all come from :-) I would not mind adding "echo: //" as well as "//" as debug string introduction. I recently started making firmware configuration files to parse and create commands, so it is easier and faster to parse according to special firmware habits.

What I wanted to say with response types is that answers should be like
echo: T:188 @:255 B:80 @B:255

here T: , @: B: and @B: would be response types. If these are unique across all responses a host would only need to parse for these and could use a simple dictionary to map to parser functions to store new firmware state denoted by that response. It would not be important if this was a response to M105 or from M303 or what else could send a change on it. It could even be totally from from a command simply based maybe on a user action on the lcd like fan change, so hosts know fan has changed. That is something I'm missing for smoothieware and marlin - feedback for user changes so hosts can show new values changed on lcd (especially fan, speed, flow), or did I miss that?

ok response with N: I understand what you mean and you are right about additional parameters. To make them understandable. So if we are only interested in line number we could use this regular expression

^ok((?<line>\s+N?\d+)|(?<other>\s.*))*

for both cases, as long as there is no second pure number. So this would be a forward compatible version that covers no line number, pure line number, Nlinenumber plus any other addition.

I have send a invitation to Lars, so we will see if he joins.

wolfmanjm, 2015-05-09:
Smoothie does not send updates to a host when things change from an LCD basically because there may not be a host. Smoothie responses are sent back to the port they were requested from as Smoothie can have multiple connections to multiple hosts. So for the example of updating fan speed I'd prefer to see the host poll for that data as it currently does for temperatures.

I really like the prefix of // for any out-of-band data (data sent to the host that was not specifically requested) Smoothie already uses that in some cases. I also think all debug messages should be prefixed with //

I';d also like to point out there is an option documented 

```// action:pause```

this is currently implemented in pronterface and octoprint, and it allows the firmware to send a message to control the host. the command following the // action: currently can be pause resume and disconnect. Smoothie uses these to handle out-of-filament detection where the host needs to be paused until the filament is replaced.
By prefixing with // hosts that do not understand this will safely ignore it.

There are lots of different types of firmware with different architectures, so this really cannot be done from the perspective of one type of firmware (like Marlin or repetier).

wolfmanjm, 2015-05-09:
I'd also like to point out that the current reprap gcode specification (regardless of when it was written) says that firmware can send a !! instead of ok and the host should stop any print in progress on that response. (I presume that is what Marlin sends ERROR: instead) Smoothie sends !! on errors when it needs the host to abort the print. Octoprint and Pronterface currently support that correctly.

There seems to be a lot of stuff that is documented, but ignored. Hence the need for the protocol specification that @foosel is proposing.

repetier, 2015-05-09:
The //  exception is noted and will find it's way into repetier. Also I must note that it is not a clever decision saying // means ignore and then adding exceptions. That simply means hosts can NOT ignore // lines as they have to check for the exception. Sending action:pause without // should not be a problem for a host. We expect new answers to come and we ignore them anyway until we understand them. That is exactly what I meant with agreeing on a standard on how to parse responses. We will never unify the responses across all firmwares but we could at least agree on a solution how to safely parse responses and how to integrate new responses without breaking parsing rules we agreed upon. Regarding pause this is a backward issue we simply have to live with now.

Also you do not want to hear it, but I do not agree that a host should constantly pull all values. sending gcode is a bottleneck and constantly pulling fan, speed, flow just to get a value that rarely changes only increases it. And if you can send action:pause you can do the same with these values. Just send Fan:128 or whatever id you decide to choose. I even would like to remove the M105 polling that already pollutes all logs. Id prefer firmware sending it on it's own or for compatibility command to switch to autosend temperatures with a response like AUTOSEND_TEMP:1 so host would then stop polling and without response it assumes old firmware and continues polling.

wolfmanjm, 2015-05-09:
my reading of // handling is not that  it is ignored but that it is displayed in the log, it is the same or an alternative to the debug: or echo:, basically it is out of band data that should be displayed to the user in some form.

wolfmanjm, 2015-05-09:
I could live with the firmware not being polled if there is a decent specification for the protocol, and part of that specification is what channel/port the updates are sent to. As i said Smoothie has several possible ports that commands can come from and it only answers to the one the request came from. But then what happens when that port is disconnected? like the network connection is closed of the USB port pulled? I can probably figure it out, but as you see things are not as simple as you would like :)

repetier, 2015-05-09:
Yes, // should be logged. But is also says debug informations and these are things the host will not interpret (my interpretation) in contrast to responses that might be of interest like temperatures.

Regarding where to send value updates to I would normally say all open connections and all which ignore answers if not connected (serial port). I agree tcp is a bit difficult as you can not send without receiver, but on the other side you register if it was opened, so you know if sends would succeed. Not as simple as you need two kinds of print - one for sender and one for all. If you allow 2 parties to connect at a time the other would also be interested in changes I guess. If not you have serial if not tcp is connected reducing it to one default sender.

scotty1024, 2015-05-09:
Suggestion.

Link, whatever link it is, USB Serial, TCP Serial, Bluetooth Serial, Quantum Entanglement Serial... Comes up in "legacy mode".

Going forward there is a new Publisher/Subscriber capability. Hosts that support this initiate their capability down a given link with the new Publisher G-Code which will return the list of publishable data that can be subscribed to.

If a given firmware is in the dark about this new ability it simply errors the unrecognized Publisher command.

Getting Publisher data then enables the Host to select the data it recognizes and desires and craft a Subscriber command to be sent down the link to have the firmware publish to it the selected data on that link.

Link state is now elevated from legacy to published and legacy M105 data is canceled. You want thermal data: subscribe to it.

Link dissolution cancels active subscriptions and returns the link state to legacy.

If the host software is ever in doubt as to the legacy/published state of the link it can simply issue the Publisher command/Subscriber command sequence.

nophead, 2015-05-09:
Or put a simpler way: the host and slave exchange the protocol version numbers that they support and then use both use the lowest of the two. I proposed this to reprap many years ago and nobody could understand the concept or why it was needed.

foosel, 2015-05-09:
@nophead thank you for that, a proper handshake as first step of getting the printer connection going is actually on my wish list for the standardised protocol too, not only for exchanging version information (I'd even say that it was the responsibility of the host to adapt here instead of the firmware supporting multiple versions, simply due to reasons of practicality since firmware is usually running on very restrictive environments) but also to figure out if its even a printer you are talking to in the first place, for some reliable way of auto detection (there are users out there who don't know their ttyAMA0 from their ttyACM0, so getting the initial connection to the printer going can already be a challenge for some of our users and that needs to be improved IMHO). Already reserved a chapter for that on the reprapcode page, still brainstorming though (plus I just came back from a business trip and need to catch up on missed sleep ;))

About the // action triggers, @repetier I agree, I also always read // as being purely unparsed debug output, basically a FYI from the firmware without any consequences for the host. however, when the discussion for these triggers came up, // prefixing seemed like it would cause the least problems for existing hosts *exactly* due to this reason, there not being any protocol definition for data pushed from the firmware. I'd actually like to include a well defined method for this into the standard proposal (that was still far away back in I think October when we discussed this action trigger on irc), maybe with a custom prefix and the explicit addition that hosts must ignore unknown messages (thus removing the issue that made us use the // prefix in the first place).

Let me take this opportunity to say that I'm very glad we are all finally having this conversation :)

nophead, 2015-05-09:
The exchange and use lowest methods is the only way you can cope with firmware newer than host scenario / vice versa. Neither can see into the future but they can both see into the past.

nophead, 2015-05-09:
Note it is note their own version numbers (although that might be useful as well), it is the protocol version they support.

[22 more replies]

closed 2015-10-12 by thinkyhead