MarlinFirmware/Marlin #2023: AutoInit/Change sdcard

trikko, 2015-05-05:
Now when I change or insert a sdcard I have to select from menu "change sdcard" or "init sdcard". Can't this be done automatically when I select "print from sd card"? It seems a fast operation.

thinkyhead, 2015-05-06:
It sounds like you are inserting an uninitialized SD card. Does it have files on it, and is it using a standard file system?

trikko, 2015-05-06:
Yes. It's a fat32 sdcard with files. I can print from it using menu. But if I remove and reinsert card with printer turned on, I have to select "change SD card" item on menu. And then "print from sdcard". Can't " change as card" be triggered each time I select "print from SD card"?

The same happens if I start printer with no card inserted. When I insert it I have to select " init SD card"

thinkyhead, 2015-05-06:
What's your `SDCARDDETECT` pin? Did you have `SDCARDDETECTINVERTED` turned on?

trikko, 2015-05-06:
I have a megatronics v3. On pins_MEGATRONICS_3.h is defined -1.

Probably it is disabled then. Can we trigger "change SD card\init SD card" on "print from SD card" if is not defined or if is == -1?

thinkyhead, 2015-05-06:
According to [this thread](http://forums.reprap.org/read.php?4,322514), you need to do some wiring to get an SD card detect pin with Megatronics 2 – and it probably also applied to Megatronics 3. The instructions (German) are [posted here](http://forums.reprap.org/read.php?249,222157). Or, you can use one of the supported display accessories that Sainsmart sells on [their own store](http://www.sainsmart.com/3d-printing/3d-printing-modules.html).

trikko, 2015-05-06:
If you check pins.h  it says that megatronics has no sdcaddetect pin. That topic explains how to connect SD card, not how to use autodetect pin. Reading megatronics v3 datasheet it seems there's no that pin. No chance to trigger change of sdcard on menu print from sdcard?

trikko, 2015-05-06:
I mean pins_MEGATRONICS_3.h, not pins.h

thinkyhead, 2015-05-07:
@trikko Yes, well that is the issue. If you want to use a Megatronics board with certain SD card readers, you have to use some workaround, either by re-wiring some other pins to the SD card reader, or by wiring up some other switch (such as a spare endstop switch) to act as the `SDCARDDETECT` pin for the auto-detection feature. Which pin you use for `SDCARDDETECT` will depend on which pin is used by your hardware solution.

trikko, 2015-05-07:
The worst thing is that it has an on-board micro SD card reader and autodetect doesn't work neither for that

thinkyhead, 2015-05-08:
@trikko What do you think of the current option, where you skip __Init. SD Card__ and go straight to __Print from SD__ there is a menu item __Refresh__ (not initially selected). The only improvement I can see to make over that is:
- to automatically init the SD when you choose __Print from SD__, and
- remove the __Init. SD Card__ menu item completely.
- Optional: If __Print from SD__ can't read a card bounce back to __Main Menu__.

trikko, 2015-05-08:
Yes it's exactly my idea. I can't test it now. Anyway, did you remove "change sdcard" menu and call it on "print from sdcard" too? (I guess it's the same menu renamed when a card is inited)

thinkyhead, 2015-05-08:
I did leave __Change SD Card__ in place, but yeah, of course it would be redundant with this. The only thing left is, when __SD Card__ screen is visible and you swap the SD card, the electronics won't detect it. So the __Refresh__ item there will still need to be manually triggered. However, if you exit the __SD Card__ screen then __Refresh__ will happen on the next __Print from SD__.

I will make the rest of those changes and update the PR shortly….

thinkyhead, 2015-05-08:
Actually, it occurs to me that the __Init. SD Card__ menu item has a use, and it's a side-effect of `card.initsd` that making this a manual operation mitigates: Namely, the `auto0.g` file starts printing when `card.initsd` is called. So this will need some more work to deal with the special case where the card should be initialized but no automatic printing started. It may be important to some users that __Init. SD Card__ and __Change SD Card__ (potentially) start `auto0.g` so I need to put them back too…

Onward…

trikko, 2015-05-08:
I think it could be a good idea to use a #define for this
Something like: #define SDCARD_AUTO_INIT

So:
#if SDCARDDETECT >= 0
... current code ...
#elif defined(SDCARD_AUTO_INIT)
... here it automaGically call init/change sd card when you select "print from sdcard" menu ...
#endif

trikko, 2015-05-08:
Anyway SDCARD_AUTO_INIT it's not the best name I guess :)

closed 2016-05-04 by jbrazio