MarlinFirmware/Marlin #2024: Bed thermistor malfunction behavior

clefranc, 2015-05-05:
Doing some temperature protection test today!

I started with the bed:

1. Send a M140 S100 command, bed start heating...
2. Disconnect the bed thermistor...
3. The bed heater stop almost instantly.

The problem is that the target bed temperature is still at 100°C (LCD) and there is no message of a temperature malfunction, not on the LCD nor the host (Repetier), but the bed has nonetheless disappeared from the host.

I'm using the latest dev version.

thinkyhead, 2015-05-06:
@clefranc That sounds like a bug! I will try playing with that later too, and see if there's a simple fix. I assume you are testing thermal runaway protection and have that option enabled (?) but it sounds like this condition should throw an alarm regardless of whether there's thermal protection or not.

clefranc, 2015-05-07:
Updated the OP!

thinkyhead, 2015-05-08:
I see if the thermistor isn't connected at all, you probably don't get any alarm either.

clefranc, 2015-05-11:
@thinkyhead Is there a reason why there is not WATCH_TEMP_PERIOD for the bed? There is thermal runaway protection for both the hotend(s) and bed, but the WATCH_TEMP_PERIOD is not applied to bed.

https://github.com/MarlinFirmware/Marlin/pull/2055

clefranc, 2015-05-11:
@thinkyhead Here is some bed tests. Will do hotends later.

### TEST #1 - Bed heating pad disconnected before startup ###

The bed heating pad shows 0°C current temperature at startup.

**Host:**
* The bed doesn't appears in the Manual Control tab!
* Sending a M140 S100 set the bed target tempeature to 100°C on the LCD...
* The bed is not heating...
* No warning or error displays on the console.

**LCD:**
* Going to the Prepare/Preaheat ABS/Preheat ABS Bed set the bed target temperature to 100°C on the LCD...
* The bed is not heating...
* No warning or error displays on the LCD.

**Conclusion:**
When the bed's thermistor is disconnected before startup, 0°C current temperature is displayed on the LCD. When sending M140 command, or using LCD preheat, there is no warning or error displayed. The bed heating resume normally after reconnecting the thermistor.

### THERMAL RUNAWAY RISK: none ###
---
### TEST #2 - Bed heating pad disconnected after heating begins ###

The bed heating pad shows 25°C current temperature at startup.

**Host:**
* Sending a M140 S100 set the bed target tempeature to 100°C on the LCD...
* The bed start heating...
* The thermistor is disconnected...
* The bed stop heating immediately...
* No warning or error displays on the console.

**LCD:**
* Going to the Prepare/Preaheat ABS/Preheat ABS Bed set the bed target temperature to 100°C on the LCD...
* Idem to host.

**Conclusion:**
When the bed's thermistor is disconnected after heating begins, 0°C current temperature is displayed on the LCD and the bed heating stops. There is no warning or error displayed when the thermistor is disconnected. The bed heating resume normally after reconnecting the thermistor.

### THERMAL RUNAWAY RISK: none ###
---
### TEST #3 - Bed heating pad disconnected before startup when thermistor table begins at 25°C ###

The bed heating pad shows 25°C current temperature at startup (ADC read 1023).

**Host:**
* Sending a M140 S100 set the bed target tempeature to 100°C on the LCD...
* The bed start heating...
* Bed current temperature stays at 25°C...
* After 5 minutes, bed still heating...
  
**LCD:**
* Going to the Prepare/Preaheat ABS/Preheat ABS Bed set the bed target temperature to 100°C on the LCD...
* Idem to host.

**Conclusion:**
When the bed heating pad is disconnected before startup AND thermistor table begins at 25°C, 25°C current temperature is always displayed on the LCD. When sending M140 command, or using LCD preheat, the bed will heat indefinitively, or stop when proper command sent.

### THERMAL RUNAWAY RISK: HIGH ###
---
### TEST #4 - Bed heating pad disconnected after heating begins when thermistor table begins at 25°C ###

The bed heating pad shows 25°C current temperature at startup.

**Host:**
* Sending a M140 S100 set the bed target tempeature to 100°C on the LCD...
* The bed start heating and current temperature is rising...
* The thermistor is disconnected...
* The current temperature drops to 25°C (ADC read 1023)...
* The bed continue to heat...
* No warning or error displays on the console.

**LCD:**
* Going to the Prepare/Preaheat ABS/Preheat ABS Bed set the bed target temperature to 100°C on the LCD...
* Idem to host.

**Conclusion:**
When the bed's thermistor is disconnected after heating begins AND thermistor table begins at 25°C, the current temperature displayed on the LCD drops to 25°C. The bed will heat indefinitively, or stop when proper command sent.

### THERMAL RUNAWAY RISK: HIGH ###

thinkyhead, 2015-05-12:
@clefranc No bed watch period has been implemented yet, but it should be easy enough to clone the hotend code and apply it to the bed heater. I will do that soon.

clefranc, 2015-05-12:
Please continue discussion here: https://github.com/MarlinFirmware/Marlin/issues/2066

AnHardt, 2015-07-18:
Closing this. See #2066

closed 2015-07-18 by AnHardt