MarlinFirmware/Marlin #2026: Adding test print to eeprom

sochinho, 2015-05-06:
Hi!
I use Marlin firmware to my 3d printer, which works on ATMega2560. I considered about adding a test print to my device. Is the possibility to write simply text file with gcode to eeprom memory, read each command to cmdbuffer and finally execute it in my function?
Thanks for help!

Nprod, 2015-05-06:
I don't think you can fit any sort of gcode in 4kb of EEPROM... Just use an SD-card?

scotty1024, 2015-05-06:
The FLASH memory is technically EEPROM. The 256KB of that usually has some left over room depending on Marlin options used.

The XYZprinting DaVinci printers have three built in test prints. They also fully transfer USB print files before they begin printing.

A bit of built in bulk storage on hand at the controller would enable some useful features.

Nprod, 2015-05-06:
There are 128kb boards too like Melzi, Sanguinololu and Printrboard, and enabling LCD support on those leaves little room. I took a look at the Da Vinci electronics and sure enough, there's an SD card plugged in. I'm guessing that's where the test print files are located and where ongoing prints are stored, not the program memory. Just use an SD card...

fmalpartida, 2015-05-06:
The SAV MkI is also a 128KB board as it uses the AT90USB1286 chip. Having said this, I don't see much point in storing a demo file on board when you can have them on an SD card with FS access.
Micro-controllers have a limited amount of memory and in many cases we are optimising code such that it can fit in them!

There is a serious effort going on too about optimising code to free up the on-board memory to squeeze there more functionality, not data.

thinkyhead, 2015-05-06:
Final answer: No, there's just not enough free space in EEPROM for a full printable object. While it would be nice to include some kind of "test print" functionality in the firmware, actually the firmware is such a "thin client" that it isn't even aware of the nozzle size. For meaningful results you really need to use a slicer and pick test objects according to what you want to test.

closed 2015-05-06 by thinkyhead

scotty1024, 2015-05-08:
I'm repairing a DaVinci and got around to looking at the controller. They are using a taped in SDCard, however, it is not FAT. This is what lets it rapidly buffer the USB print transfer onto the card. There have been many negative comments about how long it takes to transfer GCode print files to SD Card in Marlin. If low overhead ATMEL oriented simple file layout could be implemented it could both store Demo files and USB transfered print jobs in a reasonable time period.

scotty1024, 2015-05-08:
Forgot to mention, folks that wanted accurate, really accurate print job progress? If we rapidly buffer USB print jobs to an ATMEL friendly captive SD Card: we can inspect the file as it transfers to collect the extruder movement data to provide the hyper accurate print progress. Many things could be tackled via introducing a captive SD card filesystem.

Nprod, 2015-05-08:
Sounds very handy, though it would make storing gcode files with a PC more problematic. Perhaps you'd need to have two partitions on the SD card? By the way, i think the DaVinci electronics used an ARM-based microcontroller, is this correct?