MarlinFirmware/Marlin #2028: The RepRap Standards Consortium

thinkyhead, 2015-05-06:
Does there need to be a Consortium for developing shared standards between open source firmware, host software, and 3d-printing middleware across the RepRap ecosystem?

boelle, 2015-05-07:
hmm... i would say no, but it would be a great advantage (i think). At least keep it an open inventation to say repetier etc.

one place where i for my own purposes could see an advantage in a common protocol as an example is that i like the idea of the repetier informer app on the phone and the repetier server app and then repetier host as they all can exhange info and status....

the big downside is that so far i have not been able to run repetier firmware as smooth as marlin...

with a common protocol i could mix between the 2 sides of the fence and it would still work.

foosel, 2015-05-07:
IMHO there doesn't need to be something formal like a consortium, but there does need to be communication. The current everything-goes approach is not going to scale anymore.

Also there IMHO doesn't need to be a standard for host interfaces, e.g. their REST APIs or something (what @boelle is implying) - those are after all usually only facing the user or already properly documented. The firmware interface however (so, the serial communication that the host is using to talk to the printer) is something that in desperate need of being standardized because it's not very well suited for direct user consumption (stop thinking in terms of the Maker community so far and start thinking in terms of children, teachers, your dog trying to operate a printer) - and the situation there has gotten to a point where I get nightmares every time a new firmware or even just a new fork of Marlin shows up because I know I will sooner or later have users opening new tickets in OctoPrint's tracker complaining that there are problems with the temperature curve of OctoPrint when they connected it to their printer running Awkward Penguin Firmware 3.4.

boelle, 2015-05-07:
i was thinking of the serial communitcation.... just me bad at picking the right wording

foosel, 2015-05-07:
Ok, I got mislead because you were talking about Repetier Informer, which is clearly using an API provided by a host, not the serial comm.

boelle, 2015-05-07:
i only mentioned repetier as they have a nice chain of stuff... the downside is getting support for my hardware.... thou i was given some instructions that might make everything work

Hope i will not get the big "Verräter" stamp on my back for trying out repetier :-D

But it never hurts to check if the grass is more green on the other side of the fence

gege2b, 2015-05-07:
Hi there

I agree for a standardized protocol for the serial comms 
And I think that the gcode itself needs to be harmonized across all the firmwares too 

Actually, based on the Gcode wiki on reprap.org, some (especially "M") codes are interpreted differently

Example : M206 is interpreted as "set home offset" for Marlin, Sprinter and Smoothie
Repetier interpret this as  "Set EEPROM value"

Another example : M117, interpreted as a "get zero position" by FiveD and "display message" by the others

This is not really a problem for host software, but I think it's a pain in the ass for the slicers to support all this flavors

foosel, 2015-05-07:
That in fact *is* a huge problem for host software that tries to do more than just stream files to the printer and maybe move the axes.

nophead, 2015-05-07:
The original RepRap 5D firmware is fundamentally different in the way it handles feed rates compared to modern firmwares. It treats F as another dimension and interpolates it between G1 endpoints, whereas normal CNC machines generate their own trapezoids. I don't know if any firmwares still do what the second example in the G1 notes describes.

thinkyhead, 2015-05-08:
Serial messages for the present just need to be coordinated with the main forks of the finite number of hosts that currently exist. Among these are Repetier, Cura, Printrun/Pronterface, and a few others.

At the moment, there is an "ok" extension (to include the current line number) being considered, but the syntax differs between Repetier and other hosts.

@daid suggests for serial messages to use a set of standard strings for status (e.g., "TEMP_ERROR" or "HALT") rather than status/error numbers, followed by message text, which can be fully translatable.

foosel, 2015-05-08:
Remember that Marlin is not the only firmware. So it's not just Marlin coordinating with the hosts, but the other firmwares too which would actually be necessary. To make this a bit less of a horrible hassle, a standard (but not necessarily a consortium, sounds like something that would become way too unwieldy instead of getting stuff done) makes sense.

JustAnother1, 2015-05-08:
How about calling it "round table" then. That implies that everybody is of equal importance, that everybody can join and walk away anytime and also stands for people coming together to salve an common issue. Then again it is just a name. The bigger issue is to somehow get such a thing organized, right? To get anywhere with this we have to move it somewhere else. A marlin Issue is not the place to discuss this, especially not if we want to also include other firmwares and other hosts into the discussion.

It would be good to have a forum for all parties involved in best case that would be on neutral grounds,..

suggestions?

thinkyhead, 2015-05-09:
So, perhaps there should be a separate Github project, specifically for posting the standards for "g-code", handshaking, serial messages, etc. Perhaps in time it could be developed into a standard set of libraries for firmware and hosts.

wijnen, 2015-05-16:
I think it would be extremely valuable to have an organization which has a mission of standardizing communication protocols in the 3-D printer world.  I have ideas about this which may differ quite a bit from most others.  In particular, I consider the serial interface an implementation detail that doesn't require documentation except inside projects.  However, given the current situation where every host software tries to communicate with every firmware, it makes sense to standardize it.  I'm just saying that if someone writes software that includes a host and firmware that talk some custom protocol with each other, that's fine with me.  (In fact, I wrote such software.)

What I'm more interested in is a protocol which allows end user applications like OpenSCAD, Blender and Firefox to implement a Print button that sends the file to the printer driver and prints it, without the user opening any "host program".

But regardless of what is going to be defined, we're not getting anywhere without a good place to discuss it.  A repository here on github sounds like a good idea; that would provide a wiki, version control, and a semi-forum in the form of issues like this one.

thinkyhead, 2015-05-16:
> host and firmware that talk some custom protocol with each other

@wijnen Absolutely. And they should open up the protocol to the community if it solves a problem.

>  a Print button

Well, of course that will differ by platform, and there are a lot of steps between a 3D model and the GCode. Microsoft is adding a 3D print option to their OS, so we'll probably end up adopting their standards, which they are [beginning to introduce](http://apps.microsoft.com/windows/en-us/app/3d-builder/75f3f766-13b3-45e9-a62f-29590d5781f2).

> A repository here on github sounds like a good idea

For current standards I'm posting the documentation at [marlinfirmware.org](http://marlinfirmware.org/) and once that's more complete it can be used as a reference document going forward.

daid, 2015-05-16:
@thinkyhead Microsoft isn't doing anything new or revolutionary. They all use already existing tech and pretty much bring nothing new to the table (3FM is essentially a fuck you against AMF)

thinkyhead, 2015-05-16:
@daid Agreed. If people use a Windows OS that "3D Print" button is going to be a Microsoft product, and it's not really going to be possible for third parties to override that in Adobe software, for example. Each system is going to have its own hooks for drivers and control panels, if people want to integrate with the system itself. I went through this making TabletMagic to support old Wacom tablets: every system has a totally different way to deal with control panels, daemons, etc. Thankfully, for the moment, they must still speak the "RepRap language" as a final output. The community still has leverage on this side of the equation….

wijnen, 2015-05-16:
@thinkyhead In my case, my software is open source and the protocol is available, but I'm not even trying to keep it constant between versions.  When you update, you need to flash new firmware to your printer (which is a single button operation from my host software).  Everyone is free to look at the protocol and learn from it, but it isn't intended to be a standard by any definition of the word.

As for the print button: I think it should be modeled after 2-D printer drivers.  I'm only familiar with the way it is done on Unix-based systems.  On those, there is a standard interface to the driver framework (CUPS), which allows programs to query settings (so they can present a "printer preferences" dialog) and to send a print job.  For applications, every printer is expected to understand Postscript.  The framework converts this into whatever protocol the printer really needs.

I haven't written any programs with a print button, but my guess is that for the entire process, there are libraries that create the preferences dialog and handle all the communication.  As an application developer, all you should need to do is add the button and write a function to create the Postscript.  Regardless of how it does work exactly for 2-D printers, I think we should define those things:
* A language to use for sending the data.  STL is too limited IMO, because it lacks support for non-straight segments and multiple materials.  AMF seems better, but I heard it was not an open standard?
* A method for connecting to the driver framework.  Most likely, this is simply a network port.
* A protocol to use on the connection to the driver framework.
* As part of that protocol, a method to register a printer to the driver framework.
* A standard language to use for well behaving printers.  In the 2-D world, almost every printer has its own proprietary protocol (though expensive printers actually use Postscript).  The framework should be able to handle that, but it would be nice to tell people "if you want to have your printer supported without writing a driver for it, you should understand this protocol".  I think that protocol should probably be G-Code, which then should be better defined than it is now.  This is not about what to send over the serial port; that's an implementation detail in this story.  It's about what the framework sends to the printer host program (which may or may not have a GUI, but must have a network interface).

It is probably a good idea to allow host software to be started by the driver framework, so that the printer driver is only running when there is an active print job.  For software written as a server, the program that is started by the framework is a simple proxy that passes all data to and from a network connection.

Anyway, I was hoping to write proposals like these in the new group which is meant for community-wide discussions.  I can register a project for this myself, but I feel it would be better if a better-known person would do this.

daid, 2015-05-16:
AMF is indeed not really open (yet). And contains a few flaws IMHO.
Microsoft came with 3MF: http://www.3mf.io/

The microsoft 3D printer driver stuff isn't that complex. And they have "example" drivers that do "3D model -> gcode", as well as the "gcode -> USB-Serial stream" stuff. Their slicer is quite basic as far as I understood. I have contact details at Microsoft for the 3D printing stuff, if you want to dive into that area.

wijnen, 2015-05-16:
Hmm, and we need a name of course, before we can register anything.  Shall we just follow the title of this issue, "RepRap Standards Consortium"?

daid, 2015-05-16:
I think something like "Open 3D printing Standard Consortium" might be better. Outside of the RepRap scene (which is only a part of 3D printing) the name RepRap isn't that well known and does not mean much for those people. "open" on the other hand, is something that people understand a bit better.

wijnen, 2015-05-16:
@daid I think it's important that this is cross platform, so driver writers don't need to write a separate driver for every OS like they do now for 2-D printers.  Is Microsoft's way of doing things portable to other systems, so that would be an option?  If so, then it may be a good idea to do that.  However, my experience with Microsoft is that they usually try their best to break things on other platforms, and break things from others on theirs.  I don't feel comfortable using anything they made, because of the risk of them screwing us over.

Looking at the 3MF format, it is very similar to AMF from what I saw.  In fact, it seems they simply renamed the XML tags...  It also doesn't support non-straight edges, which are an important feature IMO.  So I wouldn't want to use this format.

However, I'm not aware of any open format that does support them, and I don't think creating yet another format is a good idea.  Perhaps we can insert them into 3MF or AMF through their extension systems.

As for the slicing, I think it is important to have options there.  There are many things you can want from a slicer, so it doesn't make sense to have one slicer for all users.  Also, it may be good to have different slicers for different formats.  For example, I would like to have a Print button in KiCAD as well, which generates Gerber files that can be converted into G-Code and then milled on a RepRap.  So I think the framework should provide a method for registering and choosing different slicers as well.  If Microsoft doesn't support that, I think we should ignore their system and write our own.  Because of their malicious history, I would feel more comfortable with that anyway.

I'd happily do much of that writing.  However, before I do that, I think we should set up a group where we can make collective decisions.  Not to force people to do things, but to know that the community wants this, and so the work that is put into it is not wasted.  So again, can someone please set up such a group?

wijnen, 2015-05-16:
@daid That's probably true.  On the other hand, the consortium may grow to do more things than its original scope (for example, we may want to support PCB milling as well).  So it may be better to use a name that doesn't describe much of what we're doing.  "RepRap Consortium" would be even better from that point of view.  Then again, having "Open" in it is also good, because as you say it describes our identity, which isn't going to change.  Open RepRap makes no sense; that suggest that there is also a Closed RepRap.  "Open 3-D Consortium" perhaps?

daid, 2015-05-16:
No, you don't want milling. A 3D printer is not a mill. Don't pretend it's one. Mills already have a pretty good set of semi-standard GCode. Don't clutter a standard because you have some crazy workflow that 0.1% of the actual users might consider using. The whole idea of making a standard is making the default workflow easy for users, and trouble free.

A 3D printer is a 3D printer. It produces 3D dimensional objects. Stick with that.

2D milling/lasercutting/whatever should not be mixed into that. 3D milling, maybe.

wijnen, 2015-05-16:
I wasn't saying that it must be part of the same standard; I was saying that the consortium might also do other things than just 3-D printing, or just standards.  So I would like to keep those elements out of its name.

It's not desastrous to have an incorrect name: IBM (International Business Machines) sells to consumers, too, and Microsoft also sells hardware.  Both of them don't seem to suffer from their incorrect name.  But that doesn't mean we should choose a name that we can see coming is going to be incorrect in the near future.

thinkyhead, 2015-05-17:
> Open 3D printing Standard Consortium

:+1:

thinkyhead, 2015-05-17:
> writers don't need to write a separate driver for every OS like they do now

This will not be changing anytime soon. You can make a lot of the code shared between systems, but the _glue code_ and the _UI code_ will be totally unique to each system. This is where __user-space applications__ have the advantage. They can be written in Python and use Qt and run anywhere. Not so with integrated system components.

wijnen, 2015-05-17:
@thinkyhead I disagree with you, and I would like to explain why, but not here:

In order to get anywhere, I think we should really set up this new group.  If things go well, it will probably get its own website with more custom tools, but for now, to get things started, let's just open a repository here in github, put the standard(s) as files under version control and open issues for discussions.  As I wrote, I would prefer if someone else creates the project, but I might just do it myself.  I know I don't give people much time, but I do want to have these discussions (there are several combined in this issue) and I don't think it's productive to have them here.  Also, it really doesn't make a difference who creates the project.

As I wrote, I have an opinion about the name, but on the other hand I don't care much.  Let's get this thing started.

thinkyhead, 2015-05-17:
> They can be written in Python and use Qt, or be written in Java, and be packaged up to run anywhere.

Of course, this tend to mess with the user experience, because Mac users expect a certain interface behavior and style and Windows has its own style, and generally speaking it takes some expertise in each system to get it right.

foosel, 2015-05-17:
Just for the record (and since it apparently keeps getting forgotten after having been mentioned in another ticket), I already started an attempt to standardize the transport protocol between host and firmware at [reprapcode.haz.wiki](http://reprapcode.haz.wiki).

wijnen, 2015-05-17:
@foosel Yes, I was expecting that to be one of the standards that the consortium would help finish and maintain.  Do you have a preference for hosting?  Would the current plan of a github repository work for you?  Or would a different place be better?  I'm open to everything, as long as it has at least:
* A wiki for work in progress.
* A repository under version control for finished standards.
* A method for discussion.

Jnesselr, 2015-05-17:
Oh boy...

Okay, guys listen. Our current method is a free for all. You're literally suggesting we move from a free for all to another free for all with a recognized group that has no power but has a repo in hand. We already have a wiki as @foosel pointed out, that has limited write access. The coolest part about it is that it has a disqus module as well. I think if we do a straight wiki, then discussion will be either mixed with wiki pages (which would muddle the page and history of the page) or be in an issue which would have to be linked to the page. I get that you like github and using git, but it is so far from the correct tool for this job that it's not even funny. Now when we actually do a full standard doc, we may want to version control it in a repo, but this will not be an overnight operation. 

If you want to help and be part of this project, it's not through a consortium that defines the rules and hands them down to the lowly developers below (and note, I do mean developers and not end users), but rather everyone pitching in to give suggestions and ideas. If everyone has access to the same repo, then they'll change it how they want it. I mean just looking back through Marlin commits for a little while, there are a few commands that have been broken. Their implementation changed and so did their functionality.

If you want to call those with write access to the reprapcode wiki the 'consortium' then that's all well and good, but it helps absolutely nothing. @foosel, myself, and a few others are currently going through and designing a format. This is something already in progress. Please don't try to start yet another reprap protocol. Please just close this issue before we start trying to create conflicting standards.

Oh, and please voice your opinion on the reprapcode wiki page about specific commands and the protocol.

wijnen, 2015-05-17:
@Jnesselr You misunderstood my intention.  I agree with almost everything you write.

My goal is to have standards that people agree on, so that printers can work well.  The pipeline from idea to physical object contains many programs, and they need to communicate.  Currently, one program implements an interface (often without a proper definition even), and the other uses it.  The first program might then change the interface and things break.  This is bad.  That is what I want to fix.

I very much like "free for all" in terms of what programs do.  But your freedom ends where my freedom begins.  I'm writing host+firmware myself, and I'm very happy that there isn't some manager telling me what I can and cannot do.  However, I would really like to communicate with other programs.  And I would like to have some confidence that the other programs don't change their interface next week.  And if they do, I want to point at a standard and say "your program is broken, because you aren't following the standard" (or fix my own program, if the problem is on my side).

The fix for this is to have an authorative source for the definition of the interface.  I'm 100% with you that this does not mean that the interface must change.  The only thing that must change is that there must be exactly one location to find the definion, and the definition needs to be well done.  So your work on the definition is exactly what I want to see, and certainly not something I want to replace.

I wasn't aware of the discussion module; that means that perhaps this wiki is itself good enough as a home for the consortium (although disqus is blocked by ghostery; I haven't checked why, but it suggests there are privacy concerns).  I'm not at all married to github; I just want something set up, so we can discuss issues about interfaces in a place where people who are interested can subscribe, and where we can get some results in the form of a document that people will use as the authorative definition of the interface.  Github seems like the fastest way to get something.  We can always move to a more appropriate place when things get rolling.  But because you guys have been doing work on it already, I was asking for your opinion on where to host it.  If you want to use the place you're using now, and it has the features we need (for now anyway), I'm all for it.  As I keep saying: I just want to get it started.  I don't care much about the technical details for now.

But I'm confused why you think putting it under version control would be so hard.  Can you explain that?  The reason I would like it, and that I like it to be relatively hard to make changes, is that changes should be really rare and they should only be made after everyone involved knows they're coming.  A public wiki is great (and I think the right tool) for work in progress, but not for the final standard.  You don't want a single author saying "this would be a nice extension, I'll put it in the standard" without discussion.

The idea of a consortium is to have a group of people (the same that are writing the software, I would imagine) coordinating changes in those standards when needed.  This is not in any way meant to be limiting anyone in what they can make.  Also, it isn't meant to be an elitist club; anyone with an interest in those interfaces (that means any developer of printer related software) should be allowed to join.  I have been assuming that decisions are made through consensus whenever possible.  The consortium has no power over anyone.  It's very similar the the W3C: browser vendors and site authors aren't bound by them, but they all benefit from following the standards that are defined there.  What is also similar is that they didn't define something to replace HTML; they just wrote down what was implemented at the time.  (And now they are improving it through discussions with everyone.)

However, one more thing: the interface between hosts and firmware is not the only interface that needs a definition.  So if you want to use reprapcode, we probably want to reorganize the landing page to be a list of links to landing pages for specific interfaces, the host+firmware interface being just one of them.

Jnesselr, 2015-05-17:
@wijnen My problem is conflicts of interest. We have to be careful that someone doesn't change the protocol to be what they need and not what someone else needs. That's why I'm very concerned about getting user input, but limiting access.

A github commit is much more final in my opinion than a wiki page change. Plus, the history is per page on a wiki than full repo on a github repo.

As for you having your own boss or whatever, that's a personal issue not something any protocol will fix. If we defined the standard interface for communication between host and firmware, then if you want to use your own protocol or whatever, it only means that if you use a firmware with your host and the firmware follows the spec, but you do not then any issues are your fault. The same thing is true if the firmware doesn't handle the spec correctly but the host does. I know we agree on that topic. To me, it's not about having a manager telling you what to do. For me, it's just like implementing an interface in code. The host program uses the interface, the interface being the protocol in this case, and the firmware implements this interface.

It's already going to be hard enough to get the various firmwares and hosts to agree on this protocol, I don't want to throw other protocols into the mix. Everything in due time. Right now, host/firmware interface is definitely the most important.

wijnen, 2015-05-17:
@Jnesselr Conflict of interest is exactly the reason I want to start this consortium, which isn't owned by anyone.  But it must be part of the community, so (almost) everyone (including those with interests in the outcome) must be welcome to join.  But as I wrote, I think it's important that no single member is allowed to make changes to the standard once it is accepted; that must only be possible by the group as a whole.

A repository is indeed more final, which is exactly why I want it for the standard once it is complete.  It _should_ be hard to change it at that point.

My point about being my own boss was in reply to your statement of "a consortium that defines the rules and hands them down to the lowly developers below".  I was saying that I agree with you, and that your description does not fit the consortium I'm trying to set up here.

I don't agree that host+firmware interface is the most important; in fact, I don't care much about it at all.  I'm not saying you need to work on other interfaces, I just want to have a group where people can do that.  Your current work should definitely be a part of that group, but it shouldn't be limited to that.  I would really like to work on the interface between end user applications and hosts, and I would like to do that in the context of a group like this consortium.

But you didn't answer my question: given that your work should certainly be a part of the consortium, what is your preference for hosting?  Do you want to turn reprapcode into a more general page?  If not, do you want to move to some other place, such as github?

If you don't want to be a part of this, that's unfortunate.  However, it's not the end of the world.  We'll just set up a git repository (until we need more specialized things; then we'll migrate) where we make a list of all interfaces that need a definition, and for the firmware+host interface we'll just point to your work instead of hosting it.

Please let me know what you prefer.

thinkyhead, 2015-05-18:
> apparently keeps getting forgotten after having been mentioned in another ticket

I'm new here.

thinkyhead, 2015-05-18:
> are a few commands that have been broken

Apart from changing the meaning of `E` in `G29` I'm not aware of other examples.

thinkyhead, 2015-05-18:
> Please don't try to start yet another reprap protocol.

That was never the point. And the name "Consortium" is kind of tongue-in-cheek. What I am hoping for is just some way to coordinate changes across the various tools and projects. Github makes a lot of that easier, because most all of them are here, so I just have to add the author names to any comment and we can talk about it. This has just begun to happen – after posting this issue – and it is working out well to get good feedback.

thinkyhead, 2015-05-18:
> a consortium that defines the rules and hands them down to the lowly developers below

No, no, no, this is all about lowly developers working together to improve the pace of innovation.

daid, 2015-05-18:
@thinkyhead The G28 changes a while back is how this all started. And I'm glad we caught that one before it got out to a thousand printers.

thinkyhead, 2015-05-18:
> The G28 changes a while back

@daid You must be referring to that bad idea of having `G28` apply its XYZ values in the same way as `G92` before the author was informed that the G-Code standard has `X0` in this case also —counterintuitively— mean __Home to X Max__. Thankfully peer review and testing discovered the obvious bad side-effects. I am always gratified to see that the system works.

foosel, 2015-05-18:
@thinkyhead 

> I'm new here.

You participated in #2014 where it was mentioned and you proudly proclaimed you wanted to "liberally plagiarize" from it in #2090. 

> Apart from changing the meaning of E in G29 I'm not aware of other examples.

Sticking to only the very recent events, G28, resend messaging, buffer size for Witbox/Hephestos and just last week M110. That's enough to suspect there have been more things that just not have been detected yet and as you might remember from the above mentioned ticket #2014  Marlin's users are already becoming agitated.

But firmware deliberately interpreting the existing "protocol" on the reprap wiki quite liberally and thus causing issues with interoperability is nothing new and exactly what @Jnesselr is saying.

> to improve the pace of innovation.

I'm sorry, but if this is the goal of this there's already a conflict of interests here. The problem with the current state is NOT that there's not enough innovation happening. The problem is that it's happening completely unpredictable and without deeper thought for the consequences of decisions made in the progress. That caused this mess in the first place.

Just to be clear about that (again): The goal of something like RepRapCode is *not* to stiffle experimenting and innovation, but to create a "safe environment" for that where people can play around like mad without risking to break compatibility to existing solutions out there. Hence the big requirement to offer extension points.

@wijnen 

> I don't agree that host+firmware interface is the most important; in fact, I don't care much about it at all. 

Good for you. The general public of printer users out there however - although they may not in fact know it - do, because that specific interface is what makes them able to chose from a number of host software with varying feature sets. It is THE most important communication interface. If it's undefined, all people can do is print from SD or hope that whatever their printer vendor is providing them with in terms of software to talk with the printer doesn't suck. You can't tell a relatively dumb machine like a 3d printer what to do when you don't speak its language (and in some cases can't even figure out what language it speaks). I've been working on OctoPrint for the past 2.5 years, and full time for the last 10 months of that, and I can tell you that during that time the interoperability issues with the zoo of firmware out there have been the MOST common problems that I have to find workarounds for and that really cause huge pain for the users.

> Do you have a preference for hosting? Would the current plan of a github repository work for you? Or would a different place be better?

I'd rather continue working on it and actually get stuff done where we already started instead of going through the hassle of migrating at this time. The combination of limited-access-wiki and Disqus is working fine so far (there are some quite interesting discussions with @JustAnother1 for example). I also don't think that Github is the best platform to be honest, for the reasons already pointed out above (specific discussion for example). Also, let me be blunt... the word "Consortium" for me (thanks to personal experience) invokes images of many self-important people talking a lot, agreeing on nothing, discussing endlessly about the unimportant things by ignoring the important ones and hence also not getting anything done. I'll happily be convinced that you can pull this off differently, but I'd rather not make the success (or failiure) of RepRapCode depend on it, because for that I've already invested too much time and other resources on that.

thinkyhead, 2015-05-18:
> you proudly proclaimed you wanted to "liberally plagiarize" from it

Wow, things are a little uptight and not very up on sense of humor here. I'll leave this discussion and just go back to coding now.

JustAnother1, 2015-05-18:
@foosel I agree that the issue at hand is to get a relative stable common standard as a  "safe environment" to help making interoperability easier to achieve.

I fear a bit the risk of this: https://xkcd.com/927/

So the Idea of @wijnen to have some entity that takes care of the protocol might be a good think to avoid the perception that the great thinks happening at RepRapCode are just the Octoprint-G-Code Standard.

I'm also not sure if "consortium" is the right word. I'm sure that @wijnen does not want to create a old man grout that can not agree on anything. I think the name is just a [bike shed](https://en.wikipedia.org/wiki/Parkinson%27s_law_of_triviality) So I think it would be a good Idea to move the effort made on RepRapCode to this organization. @wijnen is probably happy if you become a member of that group. It would give this work a more neutral platform. And I don't think that moving it to github is a requirement.

foosel, 2015-05-18:
You do realize that I can define things all I want, if the firmwares don't agree on them I could as well just talk to a wall? ;) It's not like OctoPrint (or any host for that matter) can dictate anything here. If I make the software incompatible to reality, well, then I have a nice piece of software nobody uses. 

My goal is merely to provide a basis for discussions, and then collaboratively fine tune that. I just don't want to break off everything we have there now and physically move it over to some as of now uncertain new parent project.

I'm also happy to play along with anything as long as stuff moves forward to the better and it doesn't add even more work on top of everything I'm already doing without moving stuff actually forward.

Honestly, if this ticket here had come about a month earlier before I finally got my mind set on "well, if nobody else does it, I should give it a try, this way I've at least tried to make things I get worked up about so often better", you'd all saved me a lot of work ;)

wijnen, 2015-05-18:
@foosel I know many people care about this interface, and I think they should.  I wrote something which is in some ways similar to Octoprint, and exactly to avoid the problems with this interface that you're describing, I decided to write my own firmware which doesn't use G-Code at all.  My opinion is that G-Code is unusable for an unreliable link such as the serial port.  But it is what most people currently use, so I also think it's good that it is well defined.  My statement wasn't meant to imply that people shouldn't care; it was just to counter the argument that everyone should think that it is the only important interface at the moment.

Anyway, I get the impression that we agree on what is good, so I hope that I can work with you on this.  Here's my proposal:
* The landing page of reprapcode is changed to become a list of standards.  The current landing page is moved to behind one of those links.
* This being a wiki, a (trusted) group of contributors can edit the standards in progress, and create new ones.
* We add some rules for deciding on how to turn work-in-progress into official-standard, and how to make changes to a standard once it has become official (this should be possible, but not too easy; it requires support for the change from the community).
* Either you lead this project, or we do it as a group.  I'm not trying to take anything over.  I think decisions about standards would be best made by more people, but if you want to have control over procedures, that's fine with me.  If have some ideas about how we can run it as a group (without a leader), which I'm happy to share if you're interested.  But as I wrote, if you want to keep control, that's fine too.

To be specific about my own plans: I want to work on the interface between end user applications and host software.  With the proposal above, I would do this in the wiki, as one or more separate standards (in progress).

Please let me know what you think about this.

Finally: I don't care about the name.  My main goal at this point is to get this going, so that there is a place where we can discuss community-wide issues and work on standard interfaces.  Marlin's issue tracker just doesn't seem to be the proper place for that. ;-)

foosel, 2015-05-18:
@wijnen we definitely can do this. Also, I don't want to be the leader of anything, I'd rather have a proper group effort at work here (since that will definitely increase both quality and acceptance) and any input in regards how to achieve this is more than welcome.

That being said, I'll try to look into restructuring that wiki a bit like you propose in the coming days (ASAP) and give you write access (can you give me an email address for that?).

And I definitely agree that this issue tracker is the wrong place ;)

wijnen, 2015-05-18:
@foosel Great!  My address is wijnen@debian.org.  I'll post my ideas for the group over there as well.

fmalpartida, 2015-05-18:
Hi,
do count me in. Though I have little spare time: day job (that pays for the bill), family (kid take most of it and it will get worse when they are on vacation). It leaves little time for anything else. 

I do like very much the idea of having a common interface defined between firmware/host. 

As a matter of fact I wouldn't like to be on the host/slicer developer's side on an environment where interfaces are constantly changing, working against different interpretations of a particular feature or command. I think that we have reached a point in which running open-loop is by no means a way forward and is heavily impacting our limited contribution time.
I do think that the 3D arena is changing very fast and could change even faster if efforts were dedicated to set alignment. Time could then be dedicated to adding features, refactoring or even bug fixing as opposed to implementing or re-inventing the wheel (hope we don't end up with many wheels though) with different interpretations or variants. 

We have to recognise that no 3D printer will work if any of it components are not in synch (well any system really). We do have limited time, and it is time that most of us give generously to improve and/or share. It is great to see that efforts are going in this direction. 

How and where? I don't think a "bug tracking" system is the right place. A wiki? perhaps, the goal is to have  a place for team collaboration so why not. If there is something that has been initiated that can be used to start with lets use that, if at a later stage it feels that it is not working, it can be changed.

This brings back some echoes from the past reviving RFCs! These were finalised by email, back at the time so...

Very supportive of the initiative.

foosel, 2015-05-20:
@wijnen You should have gotten an email from reprapcode.haz.wiki (if not, check your spam folder and if it's not there drop me a mail at osd@foosel.net).

I moved the current reprapcode spec to its own namespace (called spec since that's where I put the subpages already, sorry for that, can't change that now or the pages will lose their existing disqus threads which I'd like to avoid for obvious reasons), feel free to open up new namespaces for further specs.

The wiki also allows rights management per namespace, so we can add independent subgroups of users if need be. You need admin rights for that though which I can't hand out - the wiki is provided by a good friend of mine and he trusts me to limit that kind of access. So just tell me if you need anything. I can look into Github/G+ logins though.

Sorry that took so long, still fighting off a nasty cold.

fmalpartida, 2015-05-20:
@foosel my email: fmalpartida@gmail.com

Take care of that cold!

thinkyhead, 2015-05-21:
It might help to know about some of the codes being proposed for the next version-tagged Marlin through their inclusion in the Development branch…

__M48__ - Z Probe repeatability test
__M111__ - Set the debug level (already in the standard, but new in Marlin, and doesn't do much yet.)
__M120__ - Enable endstops
__M121__ - Disable endstops
__M145__ - Set the heatup values for a material in the LCD menu (added for EEPROM parity)
__M380__ - Enable solenoid on the active extruder
__M381__ - Disable all solenoids
__M410__ - Quickstop, Abort all planned moves
__M420__ - Enable/Disable Mesh Bed Leveling (added for EEPROM parity)
__M421__ - Set a single Mesh Bed-Leveling XYZ coordinate (added for EEPROM parity)
__M428__ - Set `home_offset` based on the distance between `current_position` and the nearest "reference point." (exposes an action that already exists in the LCD menu)

alexrj, 2015-05-21:
If you start this, count me in.
But I'm not supporting this initiative because I don't think it would work. A standardization consortium is usually started by multiple competing companies or organizations agreeing on making their products compatible instead of building walled gardens. Since we're basically talking about the firmware protocol (aka G-code), a consortium makes sense if multiple competing firmware projects partecipate in it. Other stakeholders, like G-code generators, host software or individual users are useful but they're not the core of the agreement. So a consortium won't make sense if Marlin is the only involved firmware project (and by the way, is Marlin adhering officially as a project?).

For now I propose a 3D printing firmware manifesto with just [three points](https://github.com/MarlinFirmware/Marlin/issues/2090#issuecomment-104028108). If everyone agreed on that, we'd at least avoided bad changes like [what happened to `M204`](https://github.com/alexrj/Slic3r/issues/2745).

Jnesselr, 2015-05-21:
If anything, hosts should be heavily involved. Personally, I think the host/SD card holds the interface as opposed to the firmware. Looking at the firmwares currently, I wouldn't trust that they could present an agreeable set of gcode to communicate with anything. This definitely wouldn't bring up things like automatically giving the temperature or even the format the temperature comes back as. I'd like to standardize things like that. Things like that only really affect the host side. There are other things like machine auto detection and baud rate auto configuration that would be useful to put with something like this.

I'm just saying, it's very important to consider the host's perspective.

wijnen, 2015-05-21:
@alexrj I'm not sure how "count me in" and "I'm not supporting this" go together... Do you reserve your support for things that you are sure to work?

Anyway, you seem to be misled by the name "consortium".  I don't really have a better name, but what you describe is not what I want to set up.  I want a place for everyone involved in 3-D printer development to discuss things and organize things that nobody can do on their own.  The most obvious one is writing standards, but as far as I'm concerned we can do other things when the need arises.  And yes, I certainly want more than only Marlin to join.  A good start is that I myself have no relation at all with Marlin; I'm not even a user. ;-)

As I wrote in #2090, I like to keep the standards on reprapcode.haz.wiki, at least for now, because it'd be a waste of time to move them.  However, if we want some rules for how to decide on accepting or changing a standard, we will probably want something which isn't as easy to change for that.  So unless somebody objects, I'll set up a github project for the administrative side of this.

Jnesselr, 2015-05-21:
I object! Mostly because @foosel and I are already setting up a Github
group. Give us a couple of days to set some stuff up.

alexrj, 2015-05-21:
@wijnen, a place to discuss is not enough for making standards. We can discuss and agree on changing `G1` to `FOOBAR1`, and even writing a nice spec on a nice wiki, but if no firmware or just a single firmware implements it, then it's not a standard...

Jnesselr, 2015-05-21:
@alexrj Agreed, but we have to start somewhere.

daid, 2015-05-21:
@alexrj You control a major slicer. I do so as well. We control a major firmware here. I have no idea what the state of Sprinter is. @foosel has a major part with OctoPrint.
I think we're combined quite capable of making changes. But initially we don't have to make changes, initially we just have to agree on what actually is the current mess and call that the "legacy standard" or something.

alexrj, 2015-05-21:
Usually, the starting point for making a standard is when multiple established organizations or projects agree on sharing the protocol design process.  So, first step is making a list of what firmware projects are willing to share a common protocol. If there are at least two, and they have a good userbase share, then the process will have chances to succeed. Otherwise it will just be [another unimplemented proposal](https://xkcd.com/927/).

@daid, protocol adoption is generally controlled by the receiving part. In this case, firmware. In case of HTTP, HTTP servers (not browsers). In case of HTML, browsers (not web design tools)... etc.

wijnen, 2015-05-21:
@Jnesselr Ok, that's great.  I didn't know you were doing that; I'll just wait for you then.

@alexrj You keep implying that I'm going to write a standard and then tell you that you must use it.  That is not how this works.  We (that includes you) write this standard, and then we (including hopefully all available firmwares and hosts) will like to have a standard, and will use it.

So yes, multiple organizations must share the protocol design process.  Why do you think we continue telling everyone that we want their input?  What we're currently doing is meant to be a joint effort by multiple organizations, and by the looks of it we do indeed get input from several important players in the community.  So instead of explaining how things are supposed to work, look at what's happening and join the fun. ;-)

[9 more replies]

closed 2016-04-23 by jbrazio