MarlinFirmware/Marlin #2030: Marlin Google Hangout • Saturday 9 May 2015 at 22:00pm UTC • Topics TBD • Be there!

boelle, 2015-05-07:
```
Name             Zone    UTC-7 Offset               online weekdays      online weekend
@thinkyhead      UTC-7                       
@clefranc        UTC-4   + 3 hours           
@alexborro       UTC-3   + 4 hours           
@fmalpartida     UTC+1   + 8 hours           
@AnHardt         UTC+1   + 8 hours           
@Wurstnase       UTC+1   + 8 hours           
@boelle          UTC+1   + 8 hours                  13:00 - 20:00        ~8:00 - 22:00
@emartinez167    UTC+8   +15 hours = -9      
@ntoff           UTC+10  +17 hours = -7      
```
The above UTC+ values change at summer/winter time by one hour

"Official" Clock :  http://www.timeanddate.com/time/aboututc.html

<a href="https://plus.google.com/hangouts/_/gxn3wrea5gdhoo223yimsiforia" target="_blank">Hangout</a>

link to googledoc with topics:

<a href="https://docs.google.com/document/d/1GmKjDVOKLj6f6s76ZL99LvCPOwy_PkYK57M88Q2oNdM/edit?usp=sharing" target="_blank">Topics</a>

boelle, 2015-05-09:
me off for some hours so i can be somewhat sharp tonight

i have started moving documentation to http://www.marlinfirmware.org

fmalpartida, 2015-05-09:
Me too... Going to dozze off while seeing the F1 qualifiers.

thinkyhead, 2015-05-09:
Notes from the Hangout:
- Reduce `boards.h` and `example_configurations` down to a minimum
  - Hand machine Configurations back to vendors / builders
- `Integration` branch for testing
- `Release` branch as default
- `Development` branch for latest code before it goes to `Integration`
- `RC` branch for release candidates, or include them in `Releases`
- __Action Items__
  - Create these branches
  - Reduce Configurations
  - Message maintainers concerning configurations

closed 2015-05-11 by thinkyhead