MarlinFirmware/Marlin #2031: Step for second extruder

freezone23, 2015-05-07:
I do not understand where is error.
I have two identical extruders (MK8).
when I say to repetier extrude 10mm, 80mm comes out (the first extruder is ok).

is there anyone who can help me?

thanks

nophead, 2015-05-07:
Are your microstep jumpers set correctly?

freezone23, 2015-05-07:
I have all integrated into MKS BASE 1.2....
I can not understand!!!

nophead, 2015-05-07:
Try swapping the pin definitions for E0 and E1. If the fault remains with the extruder connected to E1 then it is a hardware fault. If it switches to E0 then a firmware problem.

freezone23, 2015-05-07:
please, can you tell me how and what to do ???
please ...

freezone23, 2015-05-07:
from:
#define E0_STEP_PIN        26
#define E0_DIR_PIN         28
#define E0_ENABLE_PIN      24

#define E1_STEP_PIN        36
#define E1_DIR_PIN         34
#define E1_ENABLE_PIN      30

to:
#define E1_STEP_PIN        26
#define E1_DIR_PIN         28
#define E1_ENABLE_PIN      24

#define E0_STEP_PIN        36
#define E0_DIR_PIN         34
#define E0_ENABLE_PIN      30

right???

nophead, 2015-05-07:
Yes if those are the pins for your board.

freezone23, 2015-05-07:
first of all thanks for your help......
Now the fault is on E0.

It depends on what?

nophead, 2015-05-07:
When you say E0 do you mean the extruder connector to the E0 port but now responding when the second extruder is told to move?

freezone23, 2015-05-07:
now the extruder E0 is connected on port E1...

nophead, 2015-05-07:
So the port E1 is faulty, most likely the pins of the chip that select the microsteps are not soldered correctly. As the board is not open source it hard to say what they should be connected to but if you look up the datasheet for the the stepper driver look at the MS0,MS1,MS2 pins. They should all be grounded for X16 mode but it looks like you are in halfstep mode.

freezone23, 2015-05-07:
thank you very much ... I'll try with a new card in days before closing the issue.

thanks again
ciao

p.s.
you think I can get over it with a soldering?

nophead, 2015-05-07:
That depends on you soldering skills.

The first thing to do is find the loose pins by gently pushing them sideways with a knife blade. If they move they are not soldered. If you do find one that moves then you need to press it down with a fine tip soldering iron and possibly apply a very small amount of solder.

freezone23, 2015-05-07:
ok nophead,,, thank you for all

freezone23, 2015-05-08:
sorry but I change the card with a new....

please re-open, the issue still persists ....

thinkyhead, 2015-05-08:
I wonder if this is related at all to #1793

nophead, 2015-05-08:
If swapping the pins causes the extruders to swap, but the fault remains on port E1 then that port must be faulty. It goes 8 times faster when stepped with the same number of pulses. I can't see how it can be firmware.

thinkyhead, 2015-05-08:
I assumed that @freezone23 meant he had changed the stepper driver board and was seeing some results confirming that it wasn't the driver board. But yes, it could still be the socket on the main board.

freezone23, 2015-05-08:
Hi Chris,
It is something incomprehensible to me.

I assure you that it has replaced the card with a new one.

They can not be  faulty all ....

there must be something that escapes us...

I should give it a try with a RAMPS?
(the problem are the connectors between the two cards)

or try if you give me the link with a new firmware tested ...
thanks always to support

freezone23, 2015-05-08:
please tell me a link for  a firmware tested for 2 extruders ....

thinkyhead, 2015-05-08:
By "card" I assume you mean you have replaced the entire controller board? And yet, you still see the same problem with the new board…? So, what kind of board is it?

nophead, 2015-05-08:
@thinkyhead,
   This is an all in one integrated board (closed source), the stepper drivers are soldered on Allegro chips so can't be swapped. However swapping the signals going to them proves one behaves differently to the other, so the hardware is faulty.

@freezone23,
   I assume you have tested two, which is not the same as all.

nophead, 2015-05-08:
@freezone23,
   You could also configure the firmware for one extruder on the second port as Marlin definitely works with one extruder.

freezone23, 2015-05-08:
I tried 2 MKS Base ...

I said if try a RAMPS even if they change the types of connectors

right now I tested the configuration with an extruder changing the PIN connecting it to the port E1 ....

the problem does not change ... this test could be significant ...so ???

nophead, 2015-05-08:
My conclusion is port E1 has the wrong microstep configuration on both boards. Is there any other possible explanation?

freezone23, 2015-05-08:
Dear nophead,
you're right!
I tested all over again with a RAMPS and everything works properly ....

certainly it is difficult to accept but it is.
I however think it's a matter of PIN ...

freezone23, 2015-05-08:
I think it was just luck ... by following these guidelines:
http://reprap.org/wiki/MKS_BASE_1.0

I forced the following lines at the end of the file pins_RAMPS_13.h:
(without losing too much time)

#define HEATER_1_PIN 7
#define FAN_PIN 9

the engine now works correctly .... besides recovering a connector (E0, E1, FAN, BED)

So there is something in the previous statements which prevented proper operation.

obviously not the correct way to make changes to a file so important but I would like someone to take charge of the necessary changes to make.

Chris really thank you for the time you gave me ...

thinkyhead, 2015-05-09:
I will re-open this while more investigation is done by @freezone23 to find some pattern.

thinkyhead, 2015-05-09:
@freezone23 Are those the only differences that you needed to apply to `pins_RAMPS_13.h` to get the MKS working as it should? If so, I will add formal support for the MKS board to `boards.h` and add a `pins_MKS_BASE_1.h` file with these overrides.

thinkyhead, 2015-05-09:
#2039 / https://github.com/MarlinFirmware/Marlin/pull/2039/files

freezone23, 2015-05-09:
They are the only changes made.
for the type of motherboard = 34, I get E0, E1, FAN AND BED.

thinkyhead, 2015-05-09:
@freezone23 Great. Then #2039 is all set, and you can use `BOARD_MKS_BASE` now.

nophead, 2015-05-09:
The only explanation I can think of is that some of the unused I/O pins are connected to the stepper driver microstep configuration pins. Hard to say without a schematic or documentation but I don't see any other way the fan and heater pin assignments can change how far the extruder moves.

closed 2015-10-13 by AnHardt