MarlinFirmware/Marlin #2046: Quickhome issue

ruggb, 2015-05-10:
When I set QUICK_HOME it will run BUT it crashes into the stops.
If I home each axis individually the endstops work fine.
I think I had one config that worked but I can't find it.
This is a CoreXY system. It does it from the display or from Repetier.
I have not been able to find anyone who has a clue.
Any ideas?

thinkyhead, 2015-05-11:
Do you get the same crashing results with `G28 X Y` as with `G28` (except Z not moving, obviously)?

Can you post your `Configuration.h` here, in a Gist, or on Pastebin?

ruggb, 2015-05-11:
I didn't try sending a G28, I assumed Repetier did that when I clicked the home all button.
I tried tripping the endstops when it was moving but it does not respond to any of them.
They all work if I home the axis individually.

===============================================================
//** changes make
//== changes considering

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.

//**
#define STRING_VERSION_CONFIG_H __DATE__ "05/10/2015" __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(Bill Ruggirello)" // Who made the changes.
//** original file from Feniak, Carl

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
// This determines the communication speed of the printer
#define BAUDRATE 250000

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

//// The following define selects which electronics board you have. Please choose the one that matches your setup
// 10 = Gen7 custom (Alfons3 Version) "https://github.com/Alfons3/Generation_7_Electronics"
// 11 = Gen7 v1.1, v1.2 = 11
// 12 = Gen7 v1.3
// 13 = Gen7 v1.4
// 2  = Cheaptronic v1.0
// 20 = Sethi 3D_1
// 3  = MEGA/RAMPS up to 1.2 = 3
// 33 = RAMPS 1.3 / 1.4 (Power outputs: Extruder, Fan, Bed)
// 34 = RAMPS 1.3 / 1.4 (Power outputs: Extruder0, Extruder1, Bed)
// 35 = RAMPS 1.3 / 1.4 (Power outputs: Extruder, Fan, Fan)
// 4  = Duemilanove w/ ATMega328P pin assignment
// 5  = Gen6
// 51 = Gen6 deluxe
// 6  = Sanguinololu < 1.2
// 62 = Sanguinololu 1.2 and above
// 63 = Melzi
// 64 = STB V1.1
// 65 = Azteeg X1
// 66 = Melzi with ATmega1284 (MaKr3d version)
// 67 = Azteeg X3
// 68 = Azteeg X3 Pro
// 7  = Ultimaker
// 71 = Ultimaker (Older electronics. Pre 1.5.4. This is rare)
// 72 = Ultimainboard 2.x (Uses TEMP_SENSOR 20)
// 77 = 3Drag Controller
// 8  = Teensylu
[... 667 lines cut ...]
* Sponsored by TrinityLabs, Reworked by codexmas
**********************************************************************/

// Number of servos
//
// If you select a configuration below, this will receive a default value and does not need to be set manually
// set it manually if you have more servos than extruders and wish to manually control some
// leaving it undefined or defining as 0 will disable the servo subsystem
// If unsure, leave commented / disabled
//
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Servo Endstops
//
// This allows for servo actuated endstops, primary usage is for the Z Axis to eliminate calibration or bed height changes.
// Use M206 command to correct for switch height offset to actual nozzle height. Store that setting with M500.
//
//#define SERVO_ENDSTOPS {-1, -1, 0} // Servo index for X, Y, Z. Disable with -1
//#define SERVO_ENDSTOP_ANGLES {0,0, 0,0, 70,0} // X,Y,Z Axis Extend and Retract angles

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //__CONFIGURATION_H
===========================================================================

ruggb, 2015-05-11:
and yes G28 x y gives the same result

thinkyhead, 2015-05-11:
Wow, that sure is an old `Configuration.h` file. Let me see about updating that for you to the latest. You should post your `Configuration_adv.h` too so I can update that for you as well.

thinkyhead, 2015-05-11:
One thing that jumps out is this line:

    #define EXTRUDER_OFFSET_X {30.0, 0.0}

It should probably be:

    #define EXTRUDER_OFFSET_X {0, 30.0}

The offset is relative to E0.

ruggb, 2015-05-11:
I want to define the1 farthest from 0 as the default extruder. I thought that would work?
I tried to find something later but couldn't figure out the version numbering.
It was originally from the guy who created this corexy printer.
It would be neat if the ver # was in the file.
What else might be out of date. The routine for quick_home is in marlin.main.h isn't it?
thx

==================================================
//** changes make
//== changes considering

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.

//**
#define STRING_VERSION_CONFIG_H __DATE__ "05/10/2015" __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(Bill Ruggirello)" // Who made the changes.
//** original file from Feniak, Carl

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
// This determines the communication speed of the printer
#define BAUDRATE 250000

// This enables the serial port associated to the Bluetooth interface
//#define BTENABLED              // Enable BT interface on AT90USB devices

//// The following define selects which electronics board you have. Please choose the one that matches your setup
// 10 = Gen7 custom (Alfons3 Version) "https://github.com/Alfons3/Generation_7_Electronics"
// 11 = Gen7 v1.1, v1.2 = 11
// 12 = Gen7 v1.3
// 13 = Gen7 v1.4
// 2  = Cheaptronic v1.0
// 20 = Sethi 3D_1
// 3  = MEGA/RAMPS up to 1.2 = 3
// 33 = RAMPS 1.3 / 1.4 (Power outputs: Extruder, Fan, Bed)
// 34 = RAMPS 1.3 / 1.4 (Power outputs: Extruder0, Extruder1, Bed)
// 35 = RAMPS 1.3 / 1.4 (Power outputs: Extruder, Fan, Fan)
// 4  = Duemilanove w/ ATMega328P pin assignment
// 5  = Gen6
// 51 = Gen6 deluxe
// 6  = Sanguinololu < 1.2
// 62 = Sanguinololu 1.2 and above
// 63 = Melzi
// 64 = STB V1.1
// 65 = Azteeg X1
// 66 = Melzi with ATmega1284 (MaKr3d version)
// 67 = Azteeg X3
// 68 = Azteeg X3 Pro
// 7  = Ultimaker
// 71 = Ultimaker (Older electronics. Pre 1.5.4. This is rare)
[... 670 lines cut ...]
* Sponsored by TrinityLabs, Reworked by codexmas
**********************************************************************/

// Number of servos
//
// If you select a configuration below, this will receive a default value and does not need to be set manually
// set it manually if you have more servos than extruders and wish to manually control some
// leaving it undefined or defining as 0 will disable the servo subsystem
// If unsure, leave commented / disabled
//
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Servo Endstops
//
// This allows for servo actuated endstops, primary usage is for the Z Axis to eliminate calibration or bed height changes.
// Use M206 command to correct for switch height offset to actual nozzle height. Store that setting with M500.
//
//#define SERVO_ENDSTOPS {-1, -1, 0} // Servo index for X, Y, Z. Disable with -1
//#define SERVO_ENDSTOP_ANGLES {0,0, 0,0, 70,0} // X,Y,Z Axis Extend and Retract angles

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //__CONFIGURATION_H
======================================================

ruggb, 2015-05-11:
I meant marlin.main.cpp

ruggb, 2015-05-11:
BTW is BABYSTEPPING implemented for corexy yet?

thinkyhead, 2015-05-11:
No, babystepping is not implemented for COREXY yet.

Your updated configuration is below. The only added bit that you probably don't need is the part that says:
```
// Disable max endstops for compatibility with endstop checking routine
#if defined(COREXY) && !defined(DISABLE_MAX_ENDSTOPS)
  #define DISABLE_MAX_ENDSTOPS
#endif
```

You can just uncomment `#define DISABLE_MAX_ENDSTOPS` instead, if you need it.

```
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

#include "boards.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

// This configuration file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Scara printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//

// @section info

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.3 dev"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2

// @section machine

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
// :[0,1,2,3,4,5,6,7]
#define SERIAL_PORT 0

// This determines the communication speed of the printer
[... 733 lines cut ...]
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_SENSOR

#define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
#define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.0 //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
#define MEASURED_UPPER_LIMIT         3.3 //upper limit factor used for sensor reading validation in mm
#define MEASURED_LOWER_LIMIT         1.9 //lower limit factor for sensor reading validation in mm
#define MAX_MEASUREMENT_DELAY       20   //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

//defines used in the code
#define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

//When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
//#define FILAMENT_LCD_DISPLAY

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H
```

thinkyhead, 2015-05-11:
If you aren't sure whether to leave `#define ENDSTOPPULLUPS` enabled, then you probably should. You can use the `M119` command to see the current state of your endstops.

ruggb, 2015-05-11:
the endstops that i have r the makebot version and they have a 10k in parallel with a 4.7k and a diode.
I figured that was enough and they work individually.

thinkyhead, 2015-05-11:
Please post your `Configuration_adv.h` so that I can update that for you too.

ruggb, 2015-05-11:
i thought i posted this but I guess I missed something in the process
this forum is a little strange to me.
how do i get the latest files for Marlin?
do I have to update each or can I dnld the whole thing?

=======================================================================
//**  changes made
//== changes to consider

#ifndef CONFIGURATION_ADV_H
#define CONFIGURATION_ADV_H

//===========================================================================
//=============================Thermal Settings  ============================
//===========================================================================

#ifdef BED_LIMIT_SWITCHING
  #define BED_HYSTERESIS 2 //only disable heating if T>target+BED_HYSTERESIS and enable heating if T>target-BED_HYSTERESIS
#endif
#define BED_CHECK_INTERVAL 5000 //ms between checks in bang-bang control

//// Heating sanity check:
// This waits for the watch period in milliseconds whenever an M104 or M109 increases the target temperature
// If the temperature has not increased at the end of that period, the target temperature is set to zero.
// It can be reset with another M104/M109. This check is also only triggered if the target temperature and the current temperature
//  differ by at least 2x WATCH_TEMP_INCREASE
//#define WATCH_TEMP_PERIOD 40000 //40 seconds
//#define WATCH_TEMP_INCREASE 10  //Heat up at least 10 degree in 20 seconds

#ifdef PIDTEMP
  // this adds an experimental additional term to the heating power, proportional to the extrusion speed.
  // if Kc is chosen well, the additional required power due to increased melting should be compensated.
  #define PID_ADD_EXTRUSION_RATE
  #ifdef PID_ADD_EXTRUSION_RATE
    #define  DEFAULT_Kc (1) //heating power=Kc*(e_speed)
  #endif
#endif

//automatic temperature: The hot end target temperature is calculated by all the buffered lines of gcode.
//The maximum buffered steps/sec of the extruder motor are called "se".
//You enter the autotemp mode by a M109 S<mintemp> B<maxtemp> F<factor>
// the target temperature is set to mintemp+factor*se[steps/sec] and limited by mintemp and maxtemp
// you exit the value by any M109 without F*
// Also, if the temperature is set to a value <mintemp, it is not changed by autotemp.
// on an Ultimaker, some initial testing worked with M109 S215 B260 F1 in the start.gcode
#define AUTOTEMP
#ifdef AUTOTEMP
  #define AUTOTEMP_OLDWEIGHT 0.98
#endif

//Show Temperature ADC value
//The M105 command return, besides traditional information, the ADC value read from temperature sensors.
//#define SHOW_TEMP_ADC_VALUES

//  extruder run-out prevention.
//if the machine is idle, and the temperature over MINTEMP, every couple of SECONDS some filament is extruded
//#define EXTRUDER_RUNOUT_PREVENT
#define EXTRUDER_RUNOUT_MINTEMP 190
#define EXTRUDER_RUNOUT_SECONDS 30.
#define EXTRUDER_RUNOUT_ESTEPS 14. //mm filament
#define EXTRUDER_RUNOUT_SPEED 1500.  //extrusion speed
#define EXTRUDER_RUNOUT_EXTRUDE 100

//These defines help to calibrate the AD595 sensor in case you get wrong temperature measurements.
//The measured temperature is defined as "actualTemp = (measuredTemp * TEMP_SENSOR_AD595_GAIN) + TEMP_SENSOR_AD595_OFFSET"
#define TEMP_SENSOR_AD595_OFFSET 0.0
#define TEMP_SENSOR_AD595_GAIN   1.0

//This is for controlling a fan to cool down the stepper drivers
//it will turn on when any driver is enabled
[... 407 lines cut ...]
#endif
#if TEMP_SENSOR_BED == -1
  #define BED_USES_AD595
#endif
#if TEMP_SENSOR_0 == -2
  #define HEATER_0_USES_MAX6675
#endif
#if TEMP_SENSOR_0 == 0
  #undef HEATER_0_MINTEMP
  #undef HEATER_0_MAXTEMP
#endif
#if TEMP_SENSOR_1 == 0
  #undef HEATER_1_MINTEMP
  #undef HEATER_1_MAXTEMP
#endif
#if TEMP_SENSOR_2 == 0
  #undef HEATER_2_MINTEMP
  #undef HEATER_2_MAXTEMP
#endif
#if TEMP_SENSOR_BED == 0
  #undef BED_MINTEMP
  #undef BED_MAXTEMP
#endif

#endif //__CONFIGURATION_ADV_H

AnHardt, 2015-05-11:
To get the 'latest' code of Marlin:
Go to 'https://github.com/MarlinFirmware/Marlin'.
Change from 'Release' to 'Development' branch. (Near the green symbol)
At the bottom of the right column click 'Download ZIP'

to get 'Marlin-Development.zip'.

Or:
git clone https://github.com/MarlinFirmware/Marlin.git
git checkout Development

ruggb, 2015-05-11:
Got it - got it - got it -- now I am confused
I found this == https://github.com/MarlinFirmware/Marlin/releases
and got this == Marlin-1.0.2.zip == problem with QUICK-HOME is still there.

then u sent me here == https://github.com/MarlinFirmware/Marlin
and got this == Marlin-Development.zip
and here == https://github.com/MarlinFirmware/Marlin.git
and got this == Marlin-Release.zip

at least "Marlin 1.0.2" gives me a version #
I assume Marlin-Release.zip is the same version as above since that came from a "releases" page and it was on the top of the list, but I don't know how to verify that, and why is it posted under a different name???
I assume Marlin-Development.zip would be >> 1.0.2 but it would be nice to see a version number like 1.0.2.198 or whatever.

AnHardt, 2015-05-11:
@ruggb 
Everything behind the "Or:" was mend literally to be used on the command line using the 'git' program.

Anyway - with  'Marlin-Development.zip' you got the 'latest' code.
When looking into 'Configuration.h' yol find in line 43 the actual version number.
When all went well you will read 
#define STRING_VERSION "1.0.3 dev"

ruggb, 2015-05-11:
thx
unfortunately that convention does not carry over to the releases - which is why I never saw it.
The worst thing in documentation is to have the same thing defined more than once. Compilers don't let u get away with that. BUT even worse is to list the same thing in 2 different places with 2 different names and worse yet without identification. Maybe there is a better way??

ruggb, 2015-05-11:
not to beat a dead horse but I just found this in the Marlin-1.0.2 and the Marlin-Release versions but nothing similar in the development version.
			u8g.drawStr(62,19,"V1.0.0 RC2-mm");

maybe u could use a defined value there that is stated in the config file. It would eliminate the multiple doc problem.

AnHardt, 2015-05-11:
Good observation.
So you can see - we did something to organize the chaos, Removed the double and centralized in Configuration.h.
Taking the buss and riding a bike is almost the same difference as, loading a .zip and checkout with git. So - if you want to investigate the way between the bus-stops (versions) - learn git.

ruggb, 2015-05-11:
ah yes - I have been consumed constantly for months on this 3D printer, Prior to that I thought I was going to get into Linux and git came up, but.............. u know how that goes.
toooooo much information.

AnHardt, 2015-05-11:
Back to the toppic.
How is ' QUICK_HOME' on a 'Core-XY' with Marlin '1.0.3 dev' working for You now?

ruggb, 2015-05-11:
have not tried it yet - soon

ruggb, 2015-05-11:
1.0.3 needs work
when I tell it to home it does the same thing for single or multi - it takes one step in the wrong direction on both axis and stops.

I found it part of it - your logic for the endstops is reversed - "false" should be "true" for negative logic == grd = active
the wrong direction step was the bounce.

QUICK_HOME . IS working
I discovered that I did have a dev copy -- that must have been the one it worked on previously but I became so confused with versions I lost it.

it still isn't stepping - it is like an issue with an open thermistor....................OK I made it work by chg to 1 extruder but I do have 2 so I am not sure why it appears like it isn't seeing E1 thermistor.

BABYSTEPPING does not have the gotcha for corexy now -- does that mean it is functional for corexy??

ruggb, 2015-05-11:
I just had a thought - maybe it is a definition thing u left out. I notice there are 4 heaters now instead of 3 + the bed - 
#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_1 5
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 0

Since redundancy it there also, does that infer that 0 & 1 r for E0 and 2 & 3 r for E1?
I'm checking.............NOPE
it does not work with 2 extruders defined and the above set = 5 for 0 and 2
It does work for 1 extruder with 0 set to 5.

Also, my extruders r E3Ds and the heater resistance is 3.5 ohms - that may be a better number than 6.7 in the watts calc line.

AnHardt, 2015-05-11:
Lets stay with the endstops for a while.
Could you please send a M119 from the host - to report the end-stop-status of your switches?
Maybe it will help you to play a bit with ENDSTOP_INVERTING.
https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Configuration.h#L334

Please note the difference between: 

    //#define TEMP_SENSOR_1_AS_REDUNDANT

and

    #define TEMP_SENSOR_1_AS_REDUNDANT

The first is deactivated - the second is active.

Redundancy is only possible for the first extruder.
The correct entry for two extruders with redundand thermistors for extruder1 are supposed to look like:

    #define TEMP_SENSOR_0 5
    #define TEMP_SENSOR_1 5
    #define TEMP_SENSOR_2 5
    #define TEMP_SENSOR_3 0
    #define TEMP_SENSOR_BED 0

Normally a E3D has no redundant thermistors.

If  TEMP_SENSOR_1_AS_REDUNDANT is active please deactivate.

@thinkyhead tried his best to port your old configuration to the new one. So don't complain about an entry you have not had in you old configuration. If you think 3.5 Ohm is the better value - just take it - it's your printer. 

BABYSTEPPING is currently NOT supposed to work with Core-XY!

ruggb, 2015-05-11:
1st - the endstop issue is fixed -- I set the inversion to true - as it was in ver 1.0.2 - ver 1.0.3 changed them incorrectly from what I see. M119 shows max endstops always triggered. min endstops work fine with them set as true - I am using a version of the Mendlemax endstop boards. If u change all this back to the way it was in ver 1.0.2 it will be correct - "false" must be "true".

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool X_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_PROBE_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.

I don't have redundant thermistors but I have no idea why there r 4 sensors listed - no explaination. I set 0 & 1 to 5 then 0 & 2 to 5 neither works with 2 extruders.
1 extruder with 0 = 5 works.

There was a compile trap to inform that babystepping did not work for corexy in 1.0.2. If it does not work in 1.0.3 u need to put it back in.

AnHardt, 2015-05-11:
The trap IS in '1.0.3 dev' - and works!
Usually on RAMPS the thermistor for the bed is connected to the board between the connectors for E0 and E1.  Could this be the problem?

AnHardt, 2015-05-11:
There are now 4 thermistors + 1 for the bed because we now want to support up to 4 extruders.

ruggb, 2015-05-12:
I am going to have to dump a bunch of these builds b/c I am totally confused now.
it appears that the // Mechanical endstop with COM to ground and NC to Signal uses "true" here (most common setup). is correct since that is what I am running with. Somehow I must have chgd them in the wrong config file then loaded the right one. my apologies!!!!!

I still can't get it to step with extruders = 2 and setting ALL or any combination of the heaters sensors to 5

so the trap apparently has been moved or chgd such that I don't recognize it - but I am not a programmer so forgive me.

thinkyhead, 2015-05-12:
Thanks, @ruggb – You had re-posted your `Configuration.h`.

thinkyhead, 2015-05-12:
Here's an updated version of `Configuration_adv.h` for you.
```
#ifndef CONFIGURATION_ADV_H
#define CONFIGURATION_ADV_H

#include "Conditionals.h"

// @section temperature

//===========================================================================
//=============================Thermal Settings  ============================
//===========================================================================

#ifdef BED_LIMIT_SWITCHING
  #define BED_HYSTERESIS 2 //only disable heating if T>target+BED_HYSTERESIS and enable heating if T>target-BED_HYSTERESIS
#endif
#define BED_CHECK_INTERVAL 5000 //ms between checks in bang-bang control

/**
 * Heating Sanity Check
 *
 * Whenever an M104 or M109 increases the target temperature this will wait for WATCH_TEMP_PERIOD milliseconds,
 * and if the temperature hasn't increased by WATCH_TEMP_INCREASE degrees, the machine is halted, requiring a
 * hard reset. This test restarts with any M104/M109, but only if the current temperature is below the target
 * by at least 2 * WATCH_TEMP_INCREASE degrees celsius.
 */
#define WATCH_TEMP_PERIOD 16000 // 16 seconds
#define WATCH_TEMP_INCREASE 4  // Heat up at least 4 degrees in 16 seconds

#ifdef PIDTEMP
  // this adds an experimental additional term to the heating power, proportional to the extrusion speed.
  // if Kc is chosen well, the additional required power due to increased melting should be compensated.
  #define PID_ADD_EXTRUSION_RATE
  #ifdef PID_ADD_EXTRUSION_RATE
    #define  DEFAULT_Kc (1) //heating power=Kc*(e_speed)
  #endif
#endif

//automatic temperature: The hot end target temperature is calculated by all the buffered lines of gcode.
//The maximum buffered steps/sec of the extruder motor are called "se".
//You enter the autotemp mode by a M109 S<mintemp> B<maxtemp> F<factor>
// the target temperature is set to mintemp+factor*se[steps/sec] and limited by mintemp and maxtemp
// you exit the value by any M109 without F*
// Also, if the temperature is set to a value <mintemp, it is not changed by autotemp.
// on an Ultimaker, some initial testing worked with M109 S215 B260 F1 in the start.gcode
#define AUTOTEMP
#ifdef AUTOTEMP
  #define AUTOTEMP_OLDWEIGHT 0.98
#endif

//Show Temperature ADC value
//The M105 command return, besides traditional information, the ADC value read from temperature sensors.
//#define SHOW_TEMP_ADC_VALUES

// @section extruder

//  extruder run-out prevention.
//if the machine is idle, and the temperature over MINTEMP, every couple of SECONDS some filament is extruded
//#define EXTRUDER_RUNOUT_PREVENT
#define EXTRUDER_RUNOUT_MINTEMP 190
#define EXTRUDER_RUNOUT_SECONDS 30.
#define EXTRUDER_RUNOUT_ESTEPS 14. //mm filament
#define EXTRUDER_RUNOUT_SPEED 1500.  //extrusion speed
#define EXTRUDER_RUNOUT_EXTRUDE 100

// @section temperature

//These defines help to calibrate the AD595 sensor in case you get wrong temperature measurements.
//The measured temperature is defined as "actualTemp = (measuredTemp * TEMP_SENSOR_AD595_GAIN) + TEMP_SENSOR_AD595_OFFSET"
#define TEMP_SENSOR_AD595_OFFSET 0.0
[... 468 lines cut ...]
  #define E1_K_VAL 50          // 0 - 255, Higher values, are higher power. Be carefull not to go too high    
  #define E1_OVERCURRENT 2000  //maxc current in mA. If the current goes over this value, the driver will switch off
  #define E1_STALLCURRENT 1500 //current in mA where the driver will detect a stall
  
//  #define E2_IS_L6470
  #define E2_MICROSTEPS 16     //number of microsteps 
  #define E2_MICROSTEPS 16     //number of microsteps
  #define E2_K_VAL 50          // 0 - 255, Higher values, are higher power. Be carefull not to go too high    
  #define E2_OVERCURRENT 2000  //maxc current in mA. If the current goes over this value, the driver will switch off
  #define E2_STALLCURRENT 1500 //current in mA where the driver will detect a stall
  
//  #define E3_IS_L6470
  #define E3_MICROSTEPS 16     //number of microsteps   
  #define E3_MICROSTEPS 16     //number of microsteps
  #define E3_K_VAL 50          // 0 - 255, Higher values, are higher power. Be carefull not to go too high    
  #define E3_OVERCURRENT 2000  //maxc current in mA. If the current goes over this value, the driver will switch off
  #define E3_STALLCURRENT 1500 //current in mA where the driver will detect a stall
  
#endif

#include "Conditionals.h"
#include "SanityCheck.h"

#endif //CONFIGURATION_ADV_H
```

thinkyhead, 2015-05-12:
__I want to define the1 farthest from 0 as the default extruder.__
In that case you want
    #define EXTRUDER_OFFSET_X {0, -30.0}

thinkyhead, 2015-05-12:
__how do i get the latest files for Marlin?__
Find the ".zip" link on the bottom-right of this page https://github.com/MarlinFirmware/Marlin

ruggb, 2015-05-12:
That configuration_adv.h appears to be the same as the one in 1.0.3 dev. Is that correct?
If so, the only issue appears to be when I define EXTRUDERS 2 it won't step. It will home.
I have all sensors defined as 5.
Did I miss something or is it a bug?

thinkyhead, 2015-05-12:
It is `Configuration_adv.h` modified with your settings.

That is curious that when EXTRUDERS is 2 your printer won't move.

ruggb, 2015-05-12:
Got it, thx.
well, I just tried 1.0.2 again and I have the same issue - Repetier does not show a sensor indication for E1 and I have chkd all the h/w up to the bd. I may have a bd problem. If Marlin does not see all the sensors it will not step. Looks like I need some serious digging. I think I will first try a config I thought worked previously. I guess I can get easily confused, ya think?.

ruggb, 2015-05-12:
??? I have a ramps 1.4 bd. It has 3 T inputs, 2 designate for Te0, Te1 and 1 for Tbed. electrically they r identical. If I connect Te1 to Tbed and set T0=5, T1=0, T2=5, T3=0, T4=0 will that match T2 with E1. If not, can I do that by designating redundancy. I am trying to figure out how to test that input.

AnHardt, 2015-05-12:
The pins for the analog inputs of the temp-sensors for RAMPS are defined in 'pins_RAMPS_13.h'.
Their names are:

    #define TEMP_0_PIN         13   // ANALOG NUMBERING
    #define TEMP_1_PIN         15   // ANALOG NUMBERING
    #define TEMP_2_PIN         -1   // ANALOG NUMBERING
    and
    #define TEMP_BED_PIN       14   // ANALOG NUMBERING

swap them to your taste.

ruggb, 2015-05-12:
OK the answer is MOTHERBOARD 34 no longer works -- it has to be = "RAMPS_13_EEB"
Maybe I had the wrong one in the first place.
thx for ur help.

ruggb, 2015-05-12:
if Z = 300 but it is really at 100 and I give it the command G1 z200 why doesn't it stop when it hits the endstop?

ruggb, 2015-05-12:
let me try that again - if Z = 300 but it is really at 100 and I give the cmd G1 Z100 why doesn't it stop when the endstop is triggered? IE when it is executing a G1 command it appears the endstops have no function. The work fine doing a home.

nophead, 2015-05-13:
By default they are only enabled during homing. After that soft endstops
are sufficient to prevent it hitting the end.

thinkyhead, 2015-05-13:
> // What is this for?
> #define ENDSTOPS_ONLY_FOR_HOMING

When this is enabled, end-stops will only be turned on during homing, and will be ignored during printing, if they happen to get triggered.

thinkyhead, 2015-05-13:
> OK the answer is MOTHERBOARD 34 no longer works -- it has to be = "RAMPS_13_EEB"

Yes, I put that into the `Configuration.h` above for you.

ruggb, 2015-05-13:
Yeah, I think I found the MB version about the time u sent that. Was that config for the dev version or the release version? If i comment out the endstops for homing only will that stop it if it gets lost and hits the stop before it thinks it should? My Z isn't functioning super reliably yet.

thinkyhead, 2015-05-13:
> Was that config for the dev version or the release version?

We only support the Development version around here, because the idea is to catch current bugs, not bugs that may not exist anymore.

ruggb, 2015-05-13:
got it, thx

ruggb, 2015-05-13:
when endstops for homing only is commented out it does not step except to home but the stepper is activated - Is that the way it is suppose to operate?

ruggb, 2015-05-13:
A point of confusion.
The Ramps bd documentation shows T0 T1 T2 on the board in that order. As listed in PINS for the Ramps 1.3 bd -- TEMP_0_PIN = 13, TEMP_1_PIN = 15, TEMP_2_PIN = -1 and  TEMP_BED_PIN =14
The config shows 4 Tx settings plus a bed. I suppose u should have a TEMP_3_PIN = -1 also.
OK, I define T0 and T1 b/c I have 2 extruders and when I install the bed, T-BED.
T0 ends up on T0 of the board, T1 ends up on T2 of the board and T-BED ends up on T1 of the board because T0 is attached to 13, T1 to 14 and T2 to 15.
U might want to consider changing that or adding some verbage. It took me an hour digging up docs to figure out why it wasn't working.

thinkyhead, 2015-05-13:
> when endstops for homing only is commented out it does not step except to home but the stepper is activated - Is that the way it is suppose to operate?

That is how it will operate if your endstops are showing up as triggered for some reason, such as having the wrong __inverting__ setting or the wrong __pullup__ setting.

AnHardt, 2015-05-13:
@ruggb 
Temperatures
We have 3 temperatures in 'pins_RAMPS_13.h' because RAMPS hase 3 dedicated pins fir temperatures. There are other boards with more than that. If you really need more than 3 temperatures - just find an other free analog pin and add it. 
The order of the pins we use is inspired by the RAMPS documentation - to avoid confusion. (Always have the documentation for your board at hand - especially when setting up a system)

Movement (for machines with only one endstop per axis)
At first test if your endstops are working ( M119 - movement by hand or trigger manually ).
Decide for the location of your 0,0,0. Usually this will be front-left-nozzle on the bed (but every oher decision is possible)
Decide for the axis names. Usually x to the right, y to the back, z upwards. (variations are possible - but follow the three finger rule of the right hand - x thumb, y intex, z middel - else you will get mirrored parts) Wire the motors accordingly. (It's a good idea to stay with your boards documentation)
Determine for each axis if you have min- or max- endstops. Set up your 'Configuration.h' accordingly.
Set up you travel limits (at this time less way - is more secure)
Try to HOME each axis separately (G28 X) (with the finger on the OFF-switch)- the movement must go towards the endstop. If not - change 'X_HOME_DIR'. 
After Homing all axes, try to move your axis with g-codes. (G91 for relative movement - than G1 x1 and G1 x-1) If the moves are NOT in the right direction change 'INVERT_X_DIR' AND 'X_HOME_DIR'. 
Repeat for the other axis.

Place the nozzle in the middle of nowhere and switch off your machine an on again - or simply reset.
Try again movements per g-gode. Movement should only be possible towards the endstop until you homed that axis Moving to the other direction is dangerous because we do not know where we are - and can't find out. Software endstops also can't work before homing - because we no not know where we are. Before homing, a movement towards the endstop should be stopped by the endstop, regardless of the setting of 'ENDSTOPS_ONLY_FOR_HOMING' (If not i'd see this as a bug in Marlin)
After homing the movement is limited by the software endstops if activated ('min_software_endstops'), by the hardware endstops if not deactivated by  'ENDSTOPS_ONLY_FOR_HOMING'. Additionally most hosts (Repetier, Pronterface, ...) limit the manual movement by using their hardware-setup.

This guide is far from complete and only mend to explain what is necessary for the endstops to work.

(Remark to myself: 
Display should show '???' for the positions before homing
Absolute movements by the menu should not be allowed
But the relative movements are neccesary
What do we report to the host before homing?)

thinkyhead, 2015-05-13:
@AnHardt Yes! The display _should_ show `???` or `---` for the coordinates it doesn't know.

ruggb, 2015-05-13:
M119 appears to be giving me the proper status. triggered when set, open when not set.
max endstops are intermittent sometimes showing triggered, sometimes open.
I am triggering them by hand - not stepping.
So the endstops r working correctly - does that mean the "endstops for homing only" is not working correctly? What is the designed intent?? Read below before answering.
=========================================================================
please reread my comment on the T0,T1,T2. U have set T0 to pin 13, T1 to pin 15, Tbed to 14.
The ramps docs have T0=13, T1=14, T2=15 so something is confused - it could be me!
With ur config Te0 is on the left, Te1 is on the right, Tbed is in the middle. I'm OK with that NOW, but lots of other ppl will expect T1 to be extruder E1 therm and T2 to be Bed therm.
=========================================================================
I went thru that procedure a long time ago. All movement is correct. AND it is setup as u describe as "normal". It required this change in 1.0.3 -- #define INVERT_Z_DIR false>>true, which is different than the 1.0.2 "release" which requires this change -- #define INVERT_X_DIR true>>false. Why r the defaults different now??
++++++++++++++++++++++++++++++++++++++++++
Homing always works as expected. - except that the BOUNCE speed does not appear to be reduced for X&Y as it is for Z. ______Is this correct_______??

min sw endstops r activated. 
HOMING_ONLY is not commented out -- which means the endstops r "deactivated" except for homing.

I do a reset or an Emergency stop, home each axis, move X&Y to 300mm, send G1 X0, trip X- ES, it just keeps moving to 0. Same for Y. So far, so good........This is what u said it is suppose to do, right?
NOW....no movement is possible, except home, until I do a reset (E-stop) again and homing. Why is that???
If the endstops r not tripped during a move it continues to move fine.
If I comment out HOMING_ONLY, only homing works, I can't get any other movement, but the steppers do activate. I don't understand why it would be designed like this. IS THIS A BUG????????

SO, After all that discussion, I am back to my original ?????????  I think there r a few things that may not be "design intent" -- IE, they r bugs. 
1. Ramps T0,T1,&T2 mixed up - at least make a comment about it in config file for others
2. endswitch operation with HOMING_ONLY commented out or not <<< this is a problem
3. and maybe homing bounce speed on X&Y

ruggb, 2015-05-13:
SORRY, I don't know why those paragraphs r large font - I wasn't yelling

ruggb, 2015-05-13:
guys
everytime I test this and comment, I go back to retest it. When I do things change and I get different results. I just went back and uncommented Homing only (that was all I did) and reloaded. When I trigger an endstop now, I still get movement afterwards. it even appears not to be lost. I have no idea what is happening at this point. My results were consistent and I changed things back and forth a number of times to be sure. It always compiled and loaded. but now it is functioning differently. I give up.

ruggb, 2015-05-13:
made another chg to lower Z travel from 305 to 295 and reloaded. It is back to the way it operated - not moving after interrupting a g1 command with an endstop. go figure.

AnHardt, 2015-05-13:
@ruggb 
Are you really shouting? Plonk!

AnHardt, 2015-05-13:
http://www.nextdayreprap.co.uk/images/nextday3dprinters/Reprap_Prusa_Mendel_Build_Manual/2500_2500__Reprap_Prusa_ramps_diagram.jpeg
http://reprap.org/mediawiki/images/archive/6/6d/20130212155330!Rampswire14.svg
https://farm8.staticflickr.com/7573/15685742014_772d025008_o.png
http://4.bp.blogspot.com/-r9egw5ddy8o/U2LOqPZLVBI/AAAAAAAAc5s/Ve9wUhBlF_I/s1600/REPRAP%20schematic.png
http://www.trustfm.net/hardware/Images/3Dprinter/electronics/big/24.png
Any further questions where to connect the bed?

If you use an external editor: Please save your changes - always. Close an reopen the IDE for EVERY compilation. Else the IDE will not reload the changed files. This helps to stabilize the build process.

>Got it - got it - got it -- now I am confused

I agree.

R u a native English speaker - v c. Your language is sloppy. I sense this as impolite.

If your problem with 'ENDSTOPS_ONLY_FOR_HOMING' persists i'll have a look on it - next week.

ruggb, 2015-05-13:
@AnHardt As I explained AS SOON as I posted it - I have no clue why those paragraphs r large font. - take it as that.
@AnHardt Only one of those links is definitive as it shows 2 extruders and a bed - and it is a Pololu which mine isn't and I never saw it. What is the problem with defining it so everyone know?
@AnHardt Yes En is my only language - if u think what I wrote was impolite - get over it. Ur remarks r very direct and could be considered inflammatory. My language may be sloppy but at least I can spell. I just don't like repeating myself and can't type long words repeatedly.
My purpose is to understand this sw and make it work as best I can on my machine. If I think there is a problem I research it as best I can. I have been a EE longer than u have lived and I have done programming but I don't classify myself as a programmer. Sometimes I get the answer, sometimes I don't. When I am VERY polite, I get dismissed with excuses that do not make sense. If I need to give u more info, then ask. But don't tell me to do things I already told u I did. Then I think u start qualifying to be a tech support person, and there r very few who know anything past the computer display.
I have explained my problem fully a couple of times. I restart my editor and I restart the compiler from scratch all the time except when I forget to close the previous one, but that isn't running inside a previous instance.
The problem is currently there. If u wish, ignore it as MY problem and me being confused. Maybe that is all it is, but I am not betting on it. Maybe it is a problem with my Arduino bd. I have no idea at this point. ANd it may be that u r just not sufficiently defining things.
U (or someone) solved the quick_home issue. I had to figure out why the sw didn't match the board temp pins. I solved my direction issues long before I came to this forum.
Now, r there anymore unanswered ??? 
thx
bill

AnHardt, 2015-05-13:
RepRap Arduino Mega Pololu Shield 1.4
-.............-............-.........-..........-

[42 more replies]

closed 2015-05-17 by thinkyhead