MarlinFirmware/Marlin #2047: Print not finishing

Diaoul, 2015-05-10:
I recently updated my firmware to latest (commit 598c7aea5dcf2883e69e353dfa0957d0273b6432) and facing an issue with my prints: they do not finish.
On last layer (top solid infill) it doesn't complete and the X homing starts while extruding resulting in a little area without infill.

See example GCode (I removed the middle part): http://pastebin.com/YtcHFNyJ maybe there's something wrong with it?

Could it be that the homing starts when received instead of be placed into the gcode queue?

I'm currently running more tests to see how the print behaves without G28.

thinkyhead, 2015-05-11:
`G28` isn't treated in any special way, so it should execute normally when it comes up in the queue.

Diaoul, 2015-05-11:
OK. I'll test with the Release branch this evening and get you some pictures too.
If that's not conclusive I'll try another host program (currently using OctoPrint).

nophead, 2015-05-11:
Yes but G1s are treated in a special way and go into the planner buffer for later execution, so subsequent commands can overtake them.

Try an M400 to wait for the motion to stop before homing.

thinkyhead, 2015-05-11:
@nophead Ouch, that seems wrong, at least for G28. It should wait for all planned moves to finish before executing.

nophead, 2015-05-11:
Well in gcode all the commands should be executed in the order they appear but Marlin doesn't do that.

It doesn't make sense for any command to queue jump apart from the reporting ones, like M105, perhaps.

foosel, 2015-05-11:
Uhm... from my understanding buffered commands are in fact allowed to be executed slightly after the fact. Acknowledgement should still happen in receiving order, but acknowleding buffered commands happens explicitely before their execution instead of after as for unbuffered commands. If you had a completely synchronous processing like you are saying you could not plan ahead and basically forget about any lookaheads. 

That being said I've never noticed the behaviour described by the OP (G28 basically overtaking a G0 or G1). Since G28 is entering the planner as well as far as I understand the code, that shouldn't happen, unless something has become broken in that part, and that recently (again, I've note seen that myself, and I'm fairly sure I'd noticed that kind of broken behaviour).

nophead, 2015-05-11:
Is there a notion of "buffered commands" in the Gcode spec? 

For commands that affect machine state it only makes sense to execute them in the order given, otherwise the meaning of the gcode is changed. For example my hot end can switch off and my bed cooling fan turn on before that last few lines have been printed since the M codes jump the queue. It's madness in my opinion. Basically every M code needs prefixing with M400 to get it to happen when it should.

foosel, 2015-05-11:
What 3D printers are talking is not gcode though. It's only influenced by it (at least now, might have been different when it all started) and has so much deviated that - in my humble opinion - we shouldn't call it like that anymore to avoid any confusion with what's specified in the NIST standard. 

Only move commands are buffered as far as I know (G0 to G3, and [in theory](http://reprap.org/wiki/Gcode#Buffering) also G28), so any other move commands should always make sure that the buffered commands are executed BEFORE executing themselves (and probably stall until then).

But again, I'm fairly sure that at least concerning G28 I've never seen the described behaviour, so my guess would be that the cause for it would be a fairly recent change. Just a guess though.

nophead, 2015-05-11:
Regardless of whether it is real gcode or not it is a sequence of commands which say mainly "do this", "do that". It is always going to cause surprises if it does "that" before "this". It isn't like reordering computer instructions that can be done without changing semantics. Nearly all codes change the state of the machine, or reports its state, and what the machine does, or what is reported will change if they are executed in a different order.

thinkyhead, 2015-05-12:
This isn't a case of commands re-ordering, so much. It's a case of a command coming up in the buffer in the right order, but superseding planned moves already in-progress. Since `M400` is _essentially required_ before `G28` I suggest just inserting a direct call to `gcode_M400()` in the first line of `gcode_G28()` (and in any other codes that directly address the planner).

Diaoul, 2015-05-12:
Didn't have time to try yesterday. I'll test what you suggest this evening.

Salandora, 2015-05-12:
+1 for thinkheads method.

oh and I can confirm this bug had it before updating my printers firmware. 
Just wasn't sure if it was a firmware thing or slicer. 
Never had the time to investigate.
Don't know if it still is in latest commits no chance to test it at the moment.

thinkyhead, 2015-05-12:
@foosel The moniker "G Code" is probably here to stay, since every slicer currently outputs files with a `.gcode` (or at least `.g`) extension. But I don't see any real problem with using the name __G Code__ to refer to the code that these firmwares use, because the term __G Code__ in the [NIST RS274NGC Interpreter - Version 3](http://www.nist.gov/customcf/get_pdf.cfm?pub_id=823374) document doesn't actually refer to the language standard (although CNC people may use that term informally). In fact, the term __G Code__ in that document only refers to _control codes that start with the letter G_.

The standard called "G Code" that RepRap adopted is influenced not by something else called __G Code__ (_per se_) but by the presence in the standard of these _codes starting with __G__._

We could get more creative about it, if we want, and decide on something else the "G" can stand for…. I propose __Guidance Code__.

foosel, 2015-05-12:
@thinkyhead My point is that right now we are mixing a file format used to describe instructions on how to produce a physical object (the `.gcode` files that slicers produce) with a communication protocol used to control the machines who use said files to produce a physical object (but need somewhat interactive commands for said controlling to take place, e.g. `M105`, `G29`, `M23` etc) by calling both GCode. That's misleading at best and causing problems for implementors at worst.

thinkyhead, 2015-05-12:
#2061

thinkyhead, 2015-05-12:
I've never heard anyone give a name to Marlin's communication protocol. I propose we call it __RepRap Firmware Handshaking__.

foosel, 2015-05-12:
*cough* -> [RepRapCode](http://reprapcode.haz.wiki/start) ;)

nophead, 2015-05-12:
>This isn't a case of commands re-ordering, so much. It's a case of a command coming up in the buffer in the right order, but superseding planned moves already in-progress.

Eh? The motion commands are transferred to the planner queue and wait there until the machine executes those moves. Meanwhile non-motion commands get executed as soon as they arrive at the head of the command queue. If that is not re-ordering, i.e. executing them in a different order than they appear in the file, then I don't know what is.

If you think of what the non-motion commands are: pausing, displaying messages, changing temperature, setting I/O ports, reading position, reading limit switches, turning motors off, etc. None of these give the intended result if they are executed before motion has finished. 

In a lot of cases people don't notice mid print but it shows up at the end. There are many cases where people do notice and have to be told to use M400. But really Marlin should always wait for motion to finish before executing anything except commands like M105 and M27.

thinkyhead, 2015-05-12:
> The motion commands are transferred to the planner queue and wait there until the machine executes those moves.
Well, if `G28` only added moves to the planner that would be ok, but apparently it immediately does other things.
> Marlin should always wait for motion to finish before executing anything except…
@nophead I suppose that's true, since it matters little whether the queue is blocked because it's full or because it's waiting for a move. I'm guessing the "lookahead" feature in Marlin doesn't depend on multiple moves, so that wouldn't be affected either…?

foosel, 2015-05-12:
> I'm guessing the "lookahead" feature in Marlin doesn't depend on multiple moves

Not sure if I understand your statement correctly, but from how I understood it so far the whole point of the lookahead feature is to evaluate multiple moves (the next three IIRC), hence the planner buffer's size of >1.

thinkyhead, 2015-05-12:
@nophead I just find myself repeating what you said in my reply. :sleeping: In fact, yes, it makes great sense to halt the command queue for all non-planner moves.

thinkyhead, 2015-05-12:
@foosel Well I guess what I feel is there ought to be an extension to the planner queue – or an intermediate symbolic queue – which includes or flags these other events, so they can stay in sync with the planner.

foosel, 2015-05-12:
Not another queue please. You already have the serial rx queue, the command queue and the planner queue. What you probably want is -- as you already stated above -- keep the command queue in sync with the planner queue but for that a simple check during command queue processing should probably be enough. Keep in mind that you are basically turning any commands that are not M105 and M27 into blocking commands by that, so this is a change that does impact host communications (long moves in the background causing previously fast commands to suddenly run into timeouts for example), so this is not a trivial change that should be done lightminded (although I do agree with the general notion that it would indeed make things to have for example M114 be in sync with moves out of the box).

thinkyhead, 2015-05-12:
But hmm, I can see that pausing the command queue to wait for the planner to empty (or other conditions) basically amounts to the same thing.

thinkyhead, 2015-05-12:
So @Diaoul is all looking well with the current code?

nophead, 2015-05-12:
@foosel,
  Yes the protocol is basically broken because you don't know how long a command will take and hence don't know the appropriate time out. It works fine when the hardware is sound because USB does the link level with no errors. Gcode was never meant to be a link level protocol with OK replies, checksums, timeouts and retries. It is just a command level protocol that is supposed to be sent over an error free link.

Reprap has tried to paper over broken hardware by adding link level features, but it is a mess with no clear separation of protocol levels and no way of catering with commands that block for significant time.

thinkyhead, 2015-05-12:
Hence, Marlin has just recently gained a "wait" message to be sent every 1s for the benefit of hosts. `ADVANCED_OK`.

foosel, 2015-05-12:
@nophead no doubt about that. I don't like the current situation either, but as already said in the other thread, I can't tell all my users to buy new controllers with properly designed hardware serials. We are now in this mess and have to see how we can mitigate it in the short and solve it in the long term (and if the latter involves basically optimizing for proper functioning hardware and only adding slow but working solutions for everyone else, that's also a way to slowly but steadily force people to migrate to working things).

@thinkyhead that message is to be sent though when the buffer runs empty (which one btw? RX, Command, Planner?), it won't help if there are still commands in the buffer but the acknowledgement of the last sent one is blocking and hence nothing is being sent from the firmware back to the host.

Hence a recent change in OctoPrint's code to recognize commands that tend to take as long as they take (e.g. G28, can take AGES) and to NOT assume the printer having gone AWOL in such cases because no meep from it has been received in the last `timeout` seconds. If you make all commands but a certain subset now possible long running & blocking ones, that means an impact here.

.oO( Note to self -- add some kind of dead man's switch to the protocol proposal and hope @nophead doesn't cringe too much about it ;) )

thinkyhead, 2015-05-12:
@foosel It will only be the command buffer that is full and waiting with `ADVANCED_OK`.

thinkyhead, 2015-05-12:
Incidentally, does the firmware have any way to distinguish between a paused host and hung host?

foosel, 2015-05-12:
> It will only be the command buffer that is full and waiting with ADVANCED_OK.

Full? I thought the "wait" message was for when the queue was empty (basically a "hello, I've got nothing to do over here!")

> But the firmware could do something economical like simply echo back a digit from 0 to 9 in an endless loop, each time it queues a command.

I think you'd need something more often. If you only get feedback from when a command is queued you win nothing if the queues are blocked. You'd basically have to not actually actively wait but instead still perform your main loop in order to be able to implement a proper dead man's switch. Otherwise the firmware might still not reply in certain situations when it's actually still alive and well but just very busy, and nothing would be gained.

> does the firmware have any way to distinguish between a paused host and hung host?

Honest question: Why would it need to? A web server also doesn't bother if a browser crashes (and although some people here keep calling the firmware the client, from a classical communication point of view it's rather the other way around). But in the case of OctoPrint at least you can depend on very regular pollings of M105 and during SD printing also of M27. Because OctoPrint will only stop sending those when it considers the printer dead and closes the connection.

Salandora, 2015-05-12:
@foosel it would be interesting to know if the host  crashed, because of the heaters, if you know host is crashed you could disable the heaters, power off motors and so on. just an example I guess there are much more situations where this could be interesting.

foosel, 2015-05-12:
@Salandora fair point

Wurstnase, 2015-05-12:
> Full? I thought the "wait" message was for when the queue was empty (basically a "hello, I've got nothing to do over here!")

Right! If and only if there is nothing in the queue.

Salandora, 2015-05-12:
Well just a guess but maybe the "ok" can be changed into a ping pong game between firmware and host. So firmware and host can be sure that the respective opposite is still alive.

Salandora, 2015-05-12:
Basic idea is instead of all the OKs, every n-Seconds the host sends a ping, firmware response with the status of the queues.

If firmware is not sd printing and don't receive a ping it first assumes a lost response after 1/2 timeout, resending queue status. After full timeout it assumes a crashed host.

Same for the host.

This could be an optional feature to be backwards compatible.

nophead, 2015-05-12:
The way I have handled this sort of thing in the past is to have all
commands reply within a short specified timeout. If they need to take
longer they are split into a start doing something and then a second
command to poll for it finishing and possibly returning a result. That way
the link level protocol does not need to know about how long commands take.

thinkyhead, 2015-05-12:
Oops, I should have been looking more at [NGC](https://code.google.com/p/rs274ngc/downloads/detail?name=RS274NGC_3.pdf&can=2&q=), of course...

Diaoul, 2015-05-12:
I can confirm the fix with latest Developement branch. Thanks!

closed 2015-05-12 by Diaoul

thinkyhead, 2015-05-13:
> Full? I thought the "wait" message was for when the queue was empty (basically a "hello, I've got nothing to do over here!")

Quite right! For some reason I thought it was meant to be a pingback to the host saying "I'm still here but my buffer is full - keep waiting." That wouldn't be such a bad idea, though it only really only needs to ping back every 10 seconds, or any period below the host timeout.

foosel, 2015-05-13:
The latter would be pretty much what I meant with a dead man's switch.

scotty1024, 2015-05-13:
> I recently updated my firmware to latest (commit 598c7ae) and facing an issue with my prints: they do not finish.

You should add an M400 to the beginning of your ending G-Code in your slicer configuration.

Diaoul, 2015-05-13:
@scotty1024: this workaround shouldn't be necessary and the fix is already implemented.

scotty1024, 2015-05-13:
Apparently not.

I thought it was agreed that Development wasn't going to tolerate broken pulls anymore? Code needs to be tested prior to being submitted and if it proves to not work after integration, rolled back out?

I've been working to implement code for the HPX2-MAX I have and I'm very frustrated with the state of the Development branch.

thinkyhead, 2015-05-14:
@scotty1024 You may have more luck focusing on a single feature starting from a base of 1.0.2 instead of Development HEAD. Once you have the code worked out it shouldn't be too hard to adapt it to the current code base. I am here to help with integration when you get to the point of wanting to roll a pull request.