MarlinFirmware/Marlin #2049: Bed twist compensation

alvinavr, 2015-05-10:
If the Y-bars are not parallel it appears as if the print bed is twisted and it is difficult to get the first layer to stick. 

I have a working prototype solution for this, for my setup it makes a vast improvement. The bed is "level" across the entire surface to better than 0.025mm, the measurement limit of my dial gauge.

The solution rides on top of the AUTO_BED_LEVELING feature and does not add a lot of code. If the auto bed levelling probes four or more points there is enough information available to solve the problem. 

How is it best to submit this new feature? Could someone explain how to create a git pull request. Is their a guide for submitting pull requests that I missed?

thinkyhead, 2015-05-11:
- Fork Marlin
- Make a working copy of your fork on your computer
- Create a new branch (based on `Development`) using the command line or the Github client
- Make changes to your branch
- Commit your changes to your branch
- Submit a PR using the Github client or from your fork's Github page

AnHardt, 2015-05-11:
I'm exited about that new feature.
Additionally to what @thinkyhead wrote:
http://www.marlinfirmware.org/index.php/Codingstandards
and not totally up to date but with the right tendency:
http://www.marlinfirmware.org/index.php/Contributing

CptanPanic, 2015-05-13:
@alvinavr that would be a great feature.  Can't wait to check it out.

alvinavr, 2015-05-16:
Found some bugs that need fixing and my printer broke. I think I now understand the bug and can see a fix. My printer has a slightly twisted frame and I baked that assumption into my code. That assumption is not always correct.
My printer is now repaired and testing can start again.

boelle, 2015-06-01:
i will close this one.... we dont want to support bad adjusted printers.... if we do that we can quickly make more of a mess for ourselfes software wise

AnHardt, 2015-06-01:
@boelle 

> we dont want to support bad adjusted printers

So you think we should remove ABL? :-)

boelle, 2015-06-01:
no no... but twisted beds? a tilted bed yes....

AnHardt, 2015-06-01:
> If the Y-bars are not parallel it appears as if the print bed is twisted.

I think this is a relative common problem and for most printers hard to adjust.

CptanPanic, 2015-06-01:
I agree this should remain opened. This is just solving the more general problem of ABL in more dimensions.

alvinavr, 2015-06-01:
I am on holiday at the moment. I will get back to work on this issue next week.

I ran into some problems with my solution. I had hard coded some assumptions that are very specific to my printer. In particular the centre of rotation.

I can see how to solve it, I JUST need to do the algebra for a two factor linear regression.

CptanPanic <notifications@github.com> wrote:

>Iagreethisshouldremainopened.ThisisjustsolvingthemoregeneralproblemofABLinmoredimensions.ReplytothisemaildirectlyorviewitonGitHub.

closed 2016-04-23 by jbrazio