MarlinFirmware/Marlin #2056: Marlin Google Hangout • Saturday 23 May 2015 at 22:00pm UTC • Topics TBD • Be there!

boelle, 2015-05-11:


CONSULitAS, 2015-05-11:
@CONSULitAS       UTC+1   + 8 hours

will try to be there, but can't promise

paclema, 2015-05-11:
@paclema UTC+1 + 8 hours
I'll try to be there too, this time with connection fixed.

boelle, 2015-05-11:
oki both are now added to the table

btw... if anyone knows how to create a bootloder hexfile for a atmega 1284p with 20mhz crystal let me know....

fmalpartida, 2015-05-17:
I'll be there!
One more thing, could we change the time to something a bit like 21:00 UTC, the proposed time is like 12:00 PM here with DLS going on.

thinkyhead, 2015-05-17:
21:00 works for me.

AnHardt, 2015-05-19:
Sorry, will be away. 
Kites still have priority.

Wackerbarth, 2015-05-20:
Thats 16:00 for me (UTC-5). See you then.

thinkyhead, 2015-05-21:
Should we invite the Marlin Documentation thread participants? I feel there is still some need for folks to express their concerns.

boelle, 2015-05-21:
everyone can join, i would not mind

Den torsdag den 21. maj 2015 skrev Scott Lahteine <notifications@github.com

Jnesselr, 2015-05-21:
I think I can make that. UTC - 5 I think.

scotty1024, 2015-05-23:
Topic list suggestion.

New Pololu stamp.

3A drive, 1/128 micro stepping, built in SPI digipot for current limit and SPI status register read back for various fault conditions. I'm thinking of it as the Arduino Due of steppers. :-)

https://www.pololu.com/product/2970

foosel, 2015-05-23:
I can't make it, still not back on my feet (and if I were I'd be at a friend's wedding eve celebrations)

fmalpartida, 2015-05-23:
Take good care foosel. This year's flu is as nasty as I remember.

boelle, 2015-05-23:
i will be there.... maybe early... but our topics google doc is totally empty....

EVERONE can make suggestions as comments and we are 3 that can edit...

So please add something or we will end up watching each other pick nose

boelle, 2015-05-23:
@foosel Gute Besserung

boelle, 2015-05-23:
i'm in hangout but cam and mic disabled... will be there 5 min before we plan to start

REMEMBER to add something to the google doc.... everyone should be able to add comments... please do at bottom of doc if possible

fmalpartida, 2015-05-23:
Don't worry Scott and I are waffle masters ;-) 
There are some topics we have been discussing in threads which are worth bringing forward.

fmalpartida, 2015-05-23:
@boelle are you in the hangout?
We are in the HO that is linked here.

boelle, 2015-05-24:
hangout closed for now... we might skip the normal 14 day period if we dont have any topics

thinkyhead, 2015-05-24:
Some of the active projects to be considered when going forward on new features.

Firmware:
  - Sprinter
  - Repetier
  - LinuxCNC
  - Grbl
  - BotQueue
  - Print2Peer
  - AstroPrint
  - RepRapWeb
  - Teacup

Hosts:
  - Octoprint
  - Repetier
  - Printrun / Pronterface
  - Replicator G
  - Cura

 Slicers:
  - Slic3r
  - Cura
  - KiSSLicer
  - Skeinforge
  - Simplify3D

Jnesselr, 2015-05-24:
BotQueue, Print2Peer, AstroPrint and RepRapWeb are all hosts. Sorry for the confusion.

Wackerbarth, 2015-05-24:
Add "Repeater Host - Mac"

thinkyhead, 2015-05-24:
Other notes:

- `M240` noted. Triggers a camera pin on Marlin. Has another function in Teacup.
- `M333` noted as not being host-friendly due to blocking.
- `PR Coding Standards` label: A tag for PRs that improve code quality, maintainability, encapsulation, etc.
- Noted: Cura may be bailing early on prints with recent Marlin. Needs more corroboration.
- Topic: How can Marlin help hosts?
 - Do hosts use `M115` at all? How can it be improved?
 - Consistent temperature reporting among firmwares.

Wackerbarth, 2015-05-24:
Repeater Host - Mac Issues M115 shortly after connecting to the printer.

whosawhatsis, 2015-05-24:
Repetier Host Mac has been on version 0.56 for how long now? Over a year and a half? I wouldn't classify it as active.  

Though seeing the direction the windows version has gone since then, I guess that's a blessing.

Wackerbarth, 2015-05-24:
"active" -- In terms of development, no.  But, so far, I have gotten the best results using it on my laptop to drive my printers.

Jnesselr, 2015-05-24:
It's my go to when I hook the printer up to my Mac as well.

thinkyhead, 2015-05-24:
Adding MachinKit to the list, I suppose… http://blog.machinekit.io/

thinkyhead, 2015-05-24:
I use Repetier Host Mac to examine my gcode, sometimes to combine or modify STL scaling, etc. Never used it to slice or print…

fmalpartida, 2015-05-24:
I scrapped repetier on Mac a year and a bit ago.
Here in Spain, the use base is very low and almost everybody uses Cura now. Its simple and it works very well.

Wackerbarth, 2015-05-24:
Well, I just tried the latest version. It crashed.  :(
Also, feature-wise, I would like to see a better stored "settings" mechanism.
From a user's perspective, I like the slic3r scheme of having separate stored settings for machines, filaments, and operational modes.

thinkyhead, 2015-05-25:
> I would like to see a better stored "settings" mechanism.

I would like to have an application that lets you organize all your STLs by category (leaving them where they are in the filesystem), to be able to associate STL objects with configuration (.ini) files that are tuned for them, to be able to do batch slicing whenever you update a configuration, and to be able to direct the .gcode files that it generates to a chosen destination. I started on the general idea by making a [set of droplets](https://github.com/thinkyhead/Slic3r-Droplet-for-OSX) that you can use to batch-slice objects with a selected configuration, but a proper GUI – not too different from an organizer like iTunes – would really help.

closed 2015-05-31 by boelle