MarlinFirmware/Marlin #2058: configuration.h contains no version at all;;;

lrpirlet, 2015-05-11:
I talk about the so said released version, one of the 3 branches in the first marlin page that is not in the released list (yet?).
The configruration.h does NOT include any definition for the version...
```C++
// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
```
I would suggest using something like 
```C++
// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.3 preRel"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2
```

In the same vein, I would suggest using for the development version

```C++
// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.3 dev"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2
```
Finaly I would suggest using for the integration version

```C++
// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "1.0.3 int"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION // will be shown during bootup in line 1
//#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H // will be shown during bootup in line2
```

IMHO this would reduce misunderstanding about what poeple are using when reporting issue... 
For instance, in #2009 , 3machine is reporting using **1.0.3 dev** last updated 10-May... It cannot be release (no version), it cannot be integration last update 9-may... I should be development version that has been overwritten 11-may...

boelle, 2015-05-11:
use dev branch... we are still bugfixing

closed 2015-05-11 by boelle

3machine, 2015-05-11:
I copied the marlin folder from my gihub directory, after a sync with the github client, so what version is it? dev?

boelle, 2015-05-11:
#define STRING_VERSION "1.0.3 dev"

fmalpartida, 2015-05-11:
@3machine, 3 main branches have been created.
Branch 1 marked as release is the default branch now which has release 1.0.2 which is the last version that was tested and marked as being good enough for use. It has bugs but they are well known and they are all in the works. That release didn't have a version tag on it.

Branch 2 marked as integration, the goal of this branch is to gradually build the release branch integrating and testing what has been tested/partially tested by developers. Bugs are more controlled and "broken" or pull requests that cause collateral damage can be fixed. Main users are early adopters and testers.

Branch 3 targeted for developers, they should fork from it, make their developments and try them out in their fork prior to a PR. Then they will have to test it prior to being promoted.

I was currently writing a wiki entry so that we can all see what the intention of each branch is and its purpose. In the entry I was also going to write down the versioning scheme and labelling (which is very much in line with what @lrpirlet is suggesting).

In principle, it should allow to gain a bit of pace and may need to sort out some corner cases before trying to work on parallel releases.

In a nutshell:
If you want the greatest and latests as it is cooking and you can live with a good range of bugs, a marginally tested code base or would like to contribute to the  firmware use the "development branch".
If you want the latest with a more or less beta quality or you would like to test what is being prepared for release use the integration branch.
If you are just interested in the last release, simply go to the "release" branch.

This is trying to overcome a little bit of the confusion that has been going on around the firmware and organise the environment.

3machine, 2015-05-11:
thank you, it's clear.

AnHardt, 2015-05-11:
@3machine 
If you can't find a STRING_VERSION  in 'Configuration.h' you have 1.0.2 - the release-branch.
In 1.0.2 you have to other places where a Version is defined - in:
https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/Marlin_main.cpp#L64
and in 
https://github.com/MarlinFirmware/Marlin/blob/Release/Marlin/dogm_lcd_implementation.h#L129.
During the further development we centralized this to 'Configuration.h'.

AnHardt, 2015-05-11:
Right now added

//#define STRING_VERSION "1.0.2"

to the 'Configuration.h' in the Release-Branch.

thinkyhead, 2015-05-12:
Careful everyone! The default branch is now __Release__ and it's easy to make a PR against __Release__ from the Github client application by mistake.