MarlinFirmware/Marlin #2059: Who can make a bootloader for me?

boelle, 2015-05-11:
Hi

I have replaced my 644a chip with a 1284p

https://github.com/MarlinFirmware/Marlin/tree/Release/ArduinoAddons/Arduino_1.0.x/hardware/OMC_tkj

is my old hardware files...

of course they are no good and i cant figure how to make a new hex file

who can help out a windows noob ?

foosel, 2015-05-11:
*A summerly evening. The mobile vibrates. What's this, a new ticket on the Marlin bug tracker? But.... Huh?!* - Over 350 watchers of this repository just now ;)

Just a friendly suggestion by one of them: maybe the rep rap forums or irc would be better suited for issues like this? :) Because for people who watch this repo to stay up to date on pending issues with the code base or pull requests this just added some more noise...

Just my 2ct

boelle, 2015-05-11:
well i'm just beginning to be desperate.... at least i flagged it as "not a bug"

epatel, 2015-05-11:
This? https://code.google.com/p/sanguino/source/browse/sanguino/bootloaders/atmega/ATmegaBOOT_168_atmega1284p.hex?r=8e37b15844b11cf20cb8ca8604f44d562170815d

boelle, 2015-05-11:
is that for a 20 Mhz crystal? will try tomorrow anyways

epatel, 2015-05-11:
From the makefile it looks like it's 16Mhz
https://code.google.com/p/sanguino/source/browse/sanguino/bootloaders/atmega/Makefile?r=8e37b15844b11cf20cb8ca8604f44d562170815d

https://code.google.com/p/sanguino/source/browse/sanguino/bootloaders/atmega/?r=8e37b15844b11cf20cb8ca8604f44d562170815d

boelle, 2015-05-11:
bummer, my crystal is 20 Mhz

ntoff, 2015-05-12:
Don't you need a separate ICSP programmer, plug the board in via that and in the arduino IDE you change your uploader to the ICSP one and click "burn bootloader" ?

boelle, 2015-05-12:
i have the programmer.... BUT i miss the bootloader... a bootloader is a HEX file...

i only have the hex file for a 644a chip.... i now have a 1284p chip which is the same pin mapping but it needs a different bootloader hex file

nophead, 2015-05-12:
Maybe a gen7 one will work as they also use 20MHz I believe.

boelle, 2015-05-12:
will try, but also experimenting with compiling my own... using optiboot thou

assume it does not matter which flavour i use as it its only there to allow programming

nophead, 2015-05-12:
It needs to be the same baudrate and protocol version as that is defined in the boards.txt file you are using.

boelle, 2015-05-12:
yes... or change the boards.txt to match the chip... but i have kept that the same thou

boelle, 2015-05-12:
managed to compile the bootloader and build a boards.txt but burning the bootloader failed:

avrdude: verification error, first mismatch at byte 0x1fc00
         0x11 != 0xff
avrdude: verification error; content mismatch

boards.txt:

##############################################################

omc2.name=OMC with Atmega1284p at 20Mhz (Optiboot)(Brownout Disabled)

omc2.upload.maximum_size=130048
omc2.upload.maximum_data_size=16384

omc2.upload.protocol=arduino
omc2.upload.speed=115200
omc2.bootloader.path=OMC
omc2.bootloader.file=optiboot_bootloader-1284-20MHz.hex

omc2.bootloader.low_fuses=0xF7
omc2.bootloader.high_fuses=0xD6
omc2.bootloader.extended_fuses=0xFF
omc2.bootloader.unlock_bits=0x3F
omc2.bootloader.lock_bits=0x0F

omc2.build.mcu=atmega1284p
omc2.build.f_cpu=20000000L
omc2.build.board=AVR_OMC
omc2.build.core=arduino:arduino
omc2.build.variant=sanguino

output while compiling:

C:\Users\Bo\Desktop\Arduino IDE 1.0.6\hardware\arduino\bootloaders\optiboot1>omake.bat AVR_FREQ=20000000 atmega1284p

C:\Users\Bo\Desktop\Arduino IDE 1.0.6\hardware\arduino\bootloaders\optiboot1>..\..\..\tools\avr\utils\bin\make OS=windows ENV=arduino AVR_FREQ=20000000 atmega1284p
avr-gcc (WinAVR 20081205) 4.3.2
Copyright (C) 2008 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 113636, UBRRL = 21, Error=1.3%
../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega1284p -DF_CPU=20000000  -DBAUD_RATE=115200 -DLED_START_FLASHES=3      -DBIGBOOT    -c -o optiboot.o optiboot.c
../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega1284p -DF_CPU=20000000  -DBAUD_RATE=115200 -DLED_START_FLASHES=3      -DBIGBOOT  -Wl,--section-start=.text=0x1fc00 -Wl,--section-start=.version=0x1fffe -Wl,--relax -nostartfiles -nostdlib -o optiboot_atmega1284p.elf optiboot.o -lc
../../../tools/avr/bin/avr-size optiboot_atmega1284p.elf
   text    data     bss     dec     hex filename
    600       0       0     600     258 optiboot_atmega1284p.elf
../../../tools/avr/bin/avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega1284p.elf optiboot_atmega1284p.hex
..\..\..\tools\avr\bin\avr-objdump -h -S optiboot_atmega1284p.elf > optiboot_atmega1284p.lst
rm optiboot_atmega1284p.elf optiboot.o

C:\Users\Bo\Desktop\Arduino IDE 1.0.6\hardware\arduino\bootloaders\optiboot1>

boelle, 2015-05-13:
btw thanks to @AnHardt for sending me an hexfile to try out

AnHardt, 2015-05-13:
@boelle 
Have you recently flashed something else successfully?

boelle, 2015-05-13:
no, the only thing i had sucess with are setting the fuses

and of course i can flash the 644 chip that came with the board so
programmer should be ok

nophead, 2015-05-13:
What programmer have you got. I had to get a new one when I switched from 644 to 1284.

boelle, 2015-05-13:
usbtiny

nophead, 2015-05-13:
Yes that was the one I had that was too small for 128K. The clue is in the name :disappointed:
I got a Pololu USB AVR Programmer to replace it.

boelle, 2015-05-13:
oh... thought it did not matter... but we learn every day

nophead, 2015-05-13:
From here: https://learn.adafruit.com/usbtinyisp
>Works with any AVR ISP chip with 64K of flash (or less) - does not work with Atmega1281/1280/2561/2560

I don't know why because I would not expect it to have to load all the code into itself. It should just be a firmware upgrade to handle larger addresses.

boelle, 2015-05-13:
will order one after dinner: http://www.hobbytronics.co.uk/pololu-avr-usb-prog

boelle, 2015-05-13:
well i think there is some truth to the usbtiny is not up to the job... i just compiled a bootloader for the 644 and flashed it without any issues.... the command line is the same, i just replaced the part no from atmega1284p to atmega644p

someone have told me that optiboot should be uploaded with the arduino protocol... but i just used the stk500v2 protocol... how much does this matter?

boelle, 2015-05-13:
anyways.... i thought i would load marlin on the 1284 with "upload via programmer"

only one odd notice compared to on the 644.... when i issue a G28 X the x carriage moves away from the endstop. everything else works fine

when i use pronterface to move left and right on X i can do so... but i cant move closer to endstop than the pos where i last issued a G28 X

boelle, 2015-05-13:
oh well X issue might be because it thinks that X is allways triggered... could be because uploads with the 1284 fails....

boelle, 2015-05-13:
narrowed it down to the bootloader... its the same issue on the 644... so maybe the makefile?

nophead, 2015-05-13:
You can get problems with the limit switches if you have the fuses set wrong, such that JTAG is enabled. I can't remember if it is X or Y or both.

boelle, 2015-05-13:
Jtag not enabled....

nophead, 2015-05-13:
Well the bootloader has no influence once it passes control to the firmware.

boelle, 2015-05-13:
at least back on 644 with the bootloader that came with the electronics

boelle, 2015-05-13:
spoke to soon... had to replace the 644 with another one and flash bootloader on that one and load marlin... now X homes like it should....

so something has happened to the chip.... i guess there is no way i can blank it more or less to factory conditions before i mark it a bad one?

scotty1024, 2015-05-13:
Have you been working with the board while wearing anti static wrist strap?

Even when you hook up the programmer you can drive static into the chip.

boelle, 2015-05-13:
almost... my PSU is the 30A kind from ebay which is metal and i touch that one which again gets its ground from a 3 prong wall outlet...

boelle, 2015-05-13:
but at least i get a better programmer in the next few days and i still have an unused 1284p chip

they are not that expensive but if there are a command i could run to get it back to factory conditions that would be great as i could mark them as used but ok and put them in as spares

scotty1024, 2015-05-14:
Bo,

I have that exact programmer myself. The TTL serial interface is a useful bonus as well when working with 3D printer boards. You may use that more than the programming feature. 

ATMEL AVR fuses are called fuses after the original household electrical box fuse. They can only be blown one time. They can not be reset. One of the things the fuses do is protect the onboard FLASH memory so people can not read it out of the chip e.g. protect the code from being copied.

That wouldn’t work too well if you could reset *that* fuse.

If you examine the other fuses they control other features that you don’t want to be reset. Chris mentioned the X or Y end stop getting screwed up. Some pins are dual use and need to be set to match the wiring of the PCB the chip is running on. If the pin got flipped then the board could malfunction, so they are controlled by a fuse so they can’t be accidentally flipped back by electrical things like brown outs etc…

Once the fuses are set for a particular board you get to keep using the chip in the same kind of board.

A possibly useful reference: http://www.ladyada.net/learn/avr/fuses.html <http://www.ladyada.net/learn/avr/fuses.html>

boelle, 2015-05-14:
oh well

the only fuse i was able to change on the 644 was the extended one that
controls brown out detection. i forgot to turn that off. i just changed it
in boards.txt and burned the bootloader again. then marlin

the chips with the screwed x pin will go to the electronics graveyard

for the 1284 chip i'm just worried that i did something wrong, ie wrong
fuse settings thou i checked them many times with a fuse calculator

could really use some expert help to check everything

boelle, 2015-05-14:
will close this one again

i can end up eating more chips in value compared to a ramps board

closed 2015-05-14 by boelle

scotty1024, 2015-05-14:
You've learned something you didn't know and seems to me you've got enough here for a nice addition to the wiki to help others with boot loaders, fuses and in circuit programmers.

nophead, 2015-05-14:
@scotty1024,
  It isn't true that you can't reset the fuses. From the link you posted
> What I did not mention is that there are also 3 bytes of permanent (by permanent I mean that they stick around after power goes out, but that you can change them as many times as you'd like) storage called the fuses. The fuses determine how the chip will act, whether it has a bootloader, what speed and voltage it likes to run at, etc. Note that despite being called 'fuses' they are re-settable and dont have anything to do with protection from overpowering (like the fuses in a home).

boelle, 2015-05-14:
true, but only if what i did is right

will post full steps when home but i guess compile part is right as i just
run a command and spec chip

scotty1024, 2015-05-14:
Chris,

I said they use fuse name based on fuses from household fuses. And yes, at the mechanical level inside the chip, they are a conductor that gets melted by an electrical overload just like a household fuse.

They used to take pictures and put them in the chip manuals to re-assure engineers that their fuse design was stable before and after melting out. There is a layer of chemicals applied to the silicon chip inside the epoxy package to protect the silicon chip from moisture/oxygen that penetrates the epoxy. Popping a fuse could damage this layer so they would show you how their clever design addressed that risk. I will see if I can find one such photo from Atmel.

Intel CPU chips have fuses too but instead of using an electrical overload to melt them they use a laser at their factory to melt the fuses. One use for their fuses is to set the maximum clock speed to prevent over clocking the CPU. This allows them to then setup some of them as "Black Edition" that can be heavily over clocked and sold for extra $$$ to the over clocker community.

scotty1024, 2015-05-14:
And you can let people know not to waste money on 644's and TinyISP's because Marlin is hard to fit into it's high school jeans anymore. :-)

nophead, 2015-05-14:
@scotty1024,
  Yes some chips had fuses that melted and that is where the name comes from but the "fuses" in AVRs are just programmable flash that can only be changed with the ISP programmer. You can change the setting many times and I have done so myself.

nophead, 2015-05-14:
I should say modern AVRs, as the architecture as been around so long now they may have had one time programmable fuses in the past.

scotty1024, 2015-05-14:
Chris,

So why are Bo's X end stop pins not coming back? Damage to the transistors on those pin's from the fuse being incorrectly set?

Does Atmel rig the IP fuse to the block erase cycle for the code FLASH thus forcing the code to be blanked along with the IP fuse?

scotty1024, 2015-05-14:
Chris,

I was making money with 8051's and 80188's back when Atmel was born. I only used their various serial storage devices in designs e.g. store the MAC for a NIC. It wasn't until I built my first 3D printer that I started looking at their CPU's.

nophead, 2015-05-14:
Yes I similarly worked with microprocessors and microcontrollers before the birth of AVR, but only started using them recently because of RepRap, so I am not an expert.

I don't see any "IP" fuse in the ATmega1284P datasheet. SPIEN is perhaps what you are referring to: Enable Serial Program and Data
> The SPIEN Fuse is not accessible in serial programming mode. 

I think it only gives protection against reading the chip via SPI. I don't think there is any protection against reading the firmware with JTAG and presumably you can only set SPIEN with JTAG.

With the JTAGEN fuse set to 0 all the endstop pins and X dir stop working. X-STOP becomes TCK, which is still an input, so I can't see how it could be damaged. Certainly other people have made this error and recovered from it by clearing JTAGEN.

So in short I can't explain Bo's problem.