MarlinFirmware/Marlin #2060: Error compiling on current build related to Z-Probe

Natealus, 2015-05-12:
Hey guys, sorry I've been off for a while I had a few other things to take care of in business and life lol. I've recently merged all the recent changes and I got a compiling error so I decided to do a fresh download and configure and I still seem to be getting the same compilation error.

Arduino: 1.6.1 (Windows 8.1), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

stepper.cpp: In function 'void __vector_17()':

stepper.cpp:582:13: error: 'endstop_z_probe_hit' was not declared in this scope

             endstop_z_probe_hit=true;

             ^

stepper.cpp:641:13: error: 'endstop_z_probe_hit' was not declared in this scope

             endstop_z_probe_hit=true;

             ^

Error compiling.

Does this ring any bells as to what the issue is? I tried adding a space after = but it didn't fix it.

Wurstnase, 2015-05-12:
Sorry, my fault. Fix is on the way.

Wurstnase, 2015-05-12:
#2062 should solve this.

Natealus, 2015-05-12:
Groovy @Wurstnase :) I'll give it a roll and see if it fixes the compilation issue.

Natealus, 2015-05-12:
It did fix it thanks @Wurstnase! :+1:  :+1:

closed 2015-05-12 by Natealus

Wurstnase, 2015-05-12:
I made the problem, I fixed it ;)