MarlinFirmware/Marlin #2066: Thermistors malfunction tests results

clefranc, 2015-05-12:
Here are the bed and hotend thermistors malfunction tests results for PR https://github.com/MarlinFirmware/Marlin/pull/2055.

Tests done using full thermal protection.

Please report errors, omissions, ask for more tests or information about results here.

### TEST 1 - Bed's thermistor disconnected before startup ###
The bed shows 0°C current temperature at startup.

**Host:**
* The bed doesn't appears in the Manual Control tab!
* Sending a M140 S100 sets the bed's target tempeature to 100°C on the LCD...
* The bed is not heating...
* No warning or error displays on the console.

**LCD:**
* Going to the Prepare/Preaheat ABS/Preheat ABS Bed set the bed target temperature to 100°C on the LCD...
* Idem to host

**Conclusion:**
When the bed's thermistor is disconnected before startup, the bed's current temperature displayed on the LCD is always 0°C. When sending M140 command, or using LCD preheat, the bed's target temperature displayed on the LCD is 100°C. The bed is not heating and there is no warning or error displayed. The bed heating resumes normally after reconnecting the thermistor.

### THERMAL RUNAWAY RISK: none ###
---
### TEST 2 - Bed's thermistor disconnected before startup AND thermistor table begins at 25°C ###
The bed shows 25°C current temperature at startup.

**Host:**
* Sending a M140 S100 sets the bed's target tempeature to 100°C on the LCD...
* The bed starts heating...
* The bed's current temperature stays at 25°C...
* The bed continues to heat indefinitively...

**LCD:**
* Going to the Prepare/Preaheat ABS/Preheat ABS Bed set the bed target temperature to 100°C on the LCD...
* Idem to host.

**Conclusion:**
When the bed's thermistor is disconnected before startup AND thermistor table begins at 25°C, the bed's current temperature displayed on the LCD is always 25°C. When sending M140 command, or using LCD preheat, the bed's target temperature displayed on the LCD is 100°C. The bed will heat indefinitively, or stops when proper command sent.

### THERMAL RUNAWAY RISK: HIGH ###
---
### TEST 3 - Bed's thermistor disconnected after heating begins ###
The bed shows 25°C current temperature at startup.

**Host:**
* Sending a M140 S100 sets the bed's target tempeature to 100°C on the LCD...
* The bed starts heating...
* The bed's thermistor is disconnected...
* The bed's current temperature drops to 0°C...
* The bed's target temperature stays at 100°C...
* The bed stops heating immediately...
* No warning or error displays on the console.

**LCD:**
* Going to the Prepare/Preaheat ABS/Preheat ABS Bed set the bed target temperature to 100°C on the LCD...
* Idem to host.

**Conclusion:**
When the bed's thermistor is disconnected after heating begins, the bed's current temperature displayed on the LCD drops to 0°C and its target temperature remains at 100°C. The bed stops heating and there is no warning or error displayed. The bed heating resumes normally after reconnecting the thermistor.

### THERMAL RUNAWAY RISK: none ###
---
### TEST 4 - Bed's thermistor disconnected after heating begins AND thermistor table begins at 25°C ###
The bed shows 25°C current temperature at startup.

**Host:**
* Sending a M140 S100 sets the bed's target tempeature to 100°C on the LCD...
* The bed starts heating and current temperature is rising...
* The bed's thermistor is disconnected...
[... 206 lines cut ...]

### THERMAL RUNAWAY RISK: none ###
---
### TEST 14 - Hotend's thermistor disconnected after target temperature reached AND thermistor table begins at 25°C ###
The hotend shows 25°C current temperature at startup.

**Host:**
* Sending a M109 S240 set the hotend's target tempeature to 240°C on the LCD...
* The hotend start heating and current temperature is rising...
* The hotend's current temperature reaches 240°C...
* The hotend's thermistor is disconnected...
* The hotend's current temperature drops to 25°C...
* The hotend's target temperature remains at 240°C...
* The hotend continues to heat indefinitively...

**LCD:**
* Going to the Prepare/Preaheat ABS/Preheat ABS Bed set the hotend's target temperature to 240°C and bed's to 100°C on the LCD...
* Idem to host.

**Conclusion:**
When the hotend's thermistor is disconnected after its target temperature is reached AND thermistor table begins at 25°C, the hotend's current temperature displayed on the LCD drops to 25°C and its target temperature remains at 240°C. The hotend will heat indefinitively, or stop when proper command sent.

### THERMAL RUNAWAY RISK: HIGH ###
---
See https://github.com/MarlinFirmware/Marlin/issues/2024 and https://github.com/MarlinFirmware/Marlin/issues/2025 for original issues.

nophead, 2015-05-12:
Tl;DR but basically thermistor tables should start at 0 or at the very least below MINTEMP.

Natealus, 2015-05-12:
Talk about through testing procedure. ;) Nice work there. :+1:  :+1:

clefranc, 2015-05-12:
@nophead I totally agree.
Unfortunately, I have a 500°C thermistor which have a very high resistance around room temperature. So the "official" thermistor table from the hotend vendor start at 25°C. Someone have already redid the temp table using a thermocouple and all the stuff: same result. The ADC will read 1023 occasionnaly, enough to trigger the MINTEMP and annoy the user, hence the vendor provides a table that start at 25°C.

Is there a workaround? Anyway, I think the next Marlin release should address such issues.

@Natealus Thanks, it took me half a day to document, but I'm willing to redo it whenever new PR are available for this, because I believe safety should be priority one for the next release.

nophead, 2015-05-13:
A long time ago all the Marlin thermistor tables where changed to make sure they started at 0 so that they would always give MINTEMP when disconnected.

Which thermistor is it? Presumably it isn't 100K, that the circuit was designed for. I can see why it might not be because in order to have any resolution at 500C it would need to be a higher resistance version and then you can't measure room temp accurately enough with only a 10bit ADC.

thinkyhead, 2015-05-13:
> The LCD status message parameters: is then displayed.

#2067

thinkyhead, 2015-05-13:
> The bed doesn't appears in the Manual Control tab!

So that's the host, correct? I see that it does still show up in the LCD menu under Control > Temperature > Bed

thinkyhead, 2015-05-13:
> TEST 12

This should be slightly improved by #2067, as it keeps the test alive while the temperature is still low enough.

thinkyhead, 2015-05-13:
> TEST 4

By my reading of the code this should throw a Runaway error after `THERMAL_PROTECTION_BED_PERIOD` seconds transpires. The test is as follows:

    // If the temperature is over the target (-hysteresis) restart the timer
    if (temperature >= tr_target_temperature[heater_index] - hysteresis_degc)
      *timer = millis();
      // If the timer goes too long without a reset, trigger shutdown
    else if (millis() > *timer + period_seconds * 1000UL)
      *state = TRRunaway;

clefranc, 2015-05-13:
@thinkyhead Do I update the OP with the new PR or post a new comment?

thinkyhead, 2015-05-13:
@clefranc You should continue to comment on the differences, but since the OP is very long, it will be better to update it with your new results rather than post another long comment.

AnHardt, 2015-05-13:
> Is there a workaround?

a) Set MINTEMP well above 25 degree. Preheat the nozzle with a lighter to avoid MINTEMP-ERRORS during startup.
b) Use an other thermistor.
c) if you really need the higher temperatures - how about a thermocoupler (MAX6675 or AD595 )

thinkyhead, 2015-05-13:
The ultimate workaround is to keep augmenting the heater sanity check. The current routines to catch thermal problems are:

- Temp Wait Period routine observes the heating rate any time a new, higher temperature is set. If the temperature reading doesn't go up fast enough it will halt the machine. (This error will also be triggered by a board that is powered by USB but which hasn't any power to the RAMPS shield.)
- Thermal Runaway Protection catches a situation where the heater falls too far below the target temperature for too long, indicating that the thermistor probably fell out.
- The MIN/MAXTEMP routine catches a situation where the thermistor raw value registers outside of a fixed range, indicating a totally disconnected thermistor or a stuck relay/FET.

What is missing?
- Watch for any sudden change in the temperature reading, (such as 5°C in 1 second) to instantly catch a disconnected or bad thermistor.

nophead, 2015-05-13:
Well the algorithm I proposed earlier is simpler and catches all those simply by insisting it should get hotter when full on and cooler when full off.  Sudden jumps will get caught indirectly if they violate those basic rules. Catching them immediately will get lots of false alarms due to noisy wiring, poor grounds and loose connections. People would rather their print finished with a loose connection than abort and as long as the temperature is still under control that is OK.

Wurstnase, 2015-05-13:
I agree with Chris. My thermistor for my heatbed is really noisy while heat up. Also a lot of different checks will give it too much complexity also later for debugging issues.

AnHardt, 2015-05-13:
'Workaround' implies to me - a not so perfect solution to make it work - somehow.
Currently working on a PR what checks for sudden temperature jumps.

clefranc, 2015-05-13:
@thinkyhead Updated the OP with https://github.com/MarlinFirmware/Marlin/pull/2067

KiteLab, 2015-05-13:
@clefranc 
Does this mean - nothing has changed? Or didn't you uses ~~siekeout~~ everywhere?

clefranc, 2015-05-13:
@KiteLab Updated the strikeout so you can see what the new PR brings new.

At this time, only Test 12 goes from HIGH to none.

AnHardt, 2015-05-14:
Got 'sudden temperature jumps' to work on raw-temperature level. Tried this to avoid floating point compares.  But it's ugly, because of a lot of cases to integrate MAX5564 - with no oversampling.
I will have an other try later on the floating point level. (too difficult to concentrate when angry)

thinkyhead, 2015-05-14:
@AnHardt To the extent that things can be done at the raw temperature level, I think that will perform the best and provide the best encapsulation. We'll just have to pre-convert any temperature-based stuff into sensor values. And I like the suggestion from @nophead, if we could use an algorithm that simply checks that when the heater is on the temp goes up, when it's off the temp goes down, and _Bob's your uncle_.

nophead, 2015-05-14:
Yes it doesn't care about set point changes and start up, etc. You can rely on the fact the control loop will apply full power or full off when the temperature is outside the PID band.

grob6000, 2015-05-15:
I did a wee table, and only issue I can see with @nophead's proposal is when the printer is sitting at ambient, the heater is off, the temperature will stay constant (won't decrease) - would this trigger the fault?
Maybe a couple of options:
(a) only check when heater is enabled (not ideal - worth identifying errors even when they're off!)
(b) have a threshold (say 60C, yet another config parameter!) above which you trigger (if heater off and temp not down then kill), but below this only (if heater off and temp up then kill).
I'm a fan of doing temperature stuff in temperature space - would make the whole MAX5564 thing not an issue, not to mention any other temperature sources included later down the track...
Is it worth doing an FMEA table for this stuff, make sure we've chosen a sane solution?

thinkyhead, 2015-05-15:
#2077 looks good - and simple. I'm almost inspired enough to add BED MINTEMP and round out the whole thing...

nophead, 2015-05-15:
@grob6000 the pseudo code I posted caters for that condition explicitly with this line:
```
              initial_temp = max(current_temp(), MAX_ROOM_TEMP);
```

AnHardt, 2015-05-15:
When i tried to implement it yesterday, ma firs note was:
        // MAX_ROOM_TEMP is critical here. For sure we don't want false positives. 
        // One thermometer more, for the 'chamber temperature', would be fine.

Some ideas pop on simultaneously , for different reasons.

Any good ideas for a default value?

nophead, 2015-05-15:
Should really be called MAX_AMBIENT_TEMP I suppose to make more sense when you have a heated chamber.

50C should be safe for most cases.

grob6000, 2015-05-15:
Ah yes, thanks @nophead. Yup, 50C sounds good to me - I run my chamber at 40C at the moment, and besides, somebody who cares enough to add a chamber heater can probably configure the value in config_adv. :)

thinkyhead, 2015-05-15:
> when the printer is sitting at ambient, the heater is off, the temperature will stay constant (won't decrease) - would this trigger the fault?

Only if it's asserted that the heater must decrease when the heater is off. If you think about it, it's enough to just check that it doesn't increase when the heater is off. Standing still should be allowed in all cases, because it's a nonsense case for a heater that's on.

AnHardt, 2015-05-15:
@thinkyhead 
Good point. This way we don't need a configurable parameter for `MAX_AMBIENT_TEMP`.

During tests a had overshoot. Heater was on, is now off, but temperature rises for a while. It's not as simple as hoped for.

AnHardt, 2015-05-15:
Experimenting now with a delay.

clefranc, 2015-05-15:
@thinkyhead @AnHardt Sorry for the delay. I've just expanded the hotend test to use LCD Prepare/Preheat ABS/*Preheat ABS All*. Previous tests were done only with Preheat ABS 1. I've then discovered that some tests that were at thermal runaway risk none are indeed *HIGH* when using *Preheat ABS All*.

The thermal runaway protection seems to be ignored when one of the hotend's temperature is rising and the other is not.

Very interesting...

grob6000, 2015-05-15:
With the heater off & temp going up check: as long as we still pickup mintemp/maxtemp for the thermistor electrical faults, this should be fine I think.

nophead, 2015-05-15:
@AnHardt,
  The algorithm I proposed has a delay because even if it doesn't overshoot it takes time to start rising or falling when the heater state changes. It it didn't, control would be trivial.

@thinkyhead,
>Only if it's asserted that the heater must decrease when the heater is off. If you think about it, it's enough to just check that it doesn't increase when the heater is off. 

No that isn't sufficient because the room temperature can fluctuate up and down a little while the heater is off. That is why I had MAX_ROOM_TEMP to not require it to fall when switched off below that temperature. In normal use it will be much hotter when switched off but if somebody aborted heating immediately you could get a false trigger.

grob6000, 2015-05-15:
Good point - hadn't thought of that. Something like my heated chamber would trip pretty quickly - the hotend temperature would gradually increase as the chamber ambient increases, even though the hotend heater is off. I would like that this didn't trigger a temp fault...

Perhaps MAX_ROOM_TEMP threshold for the (heater off & not dropping --> kill) logic is just the best idea.

And could we use something like "MAX_ENCLOSURE_TEMP" or "MAX_CHAMBER_TEMP" instead of "ROOM" or "AMBIENT": these imply *outside* the printer in my mind...

nophead, 2015-05-15:
A lot of printers have no inside or outside if they don't have an enclosure or chamber. Perhaps quiescent?

thinkyhead, 2015-05-15:
> the room temperature can fluctuate up and down a little while the heater is off

Well, of course you would ignore normal slow fluctuations.

nophead, 2015-05-15:
Yes but if you insisted it goes down when switched off near room temp just as your central heating switches on you might get a false alarm as it might slowly increase for a long time. Hence the reason for not requiring it to go down when switched off below MAX_ROOM_TEMP.

thinkyhead, 2015-05-15:
That's not a bad idea to do also. The chances of a hot-end sitting below —say 80°C — for any period of time in normal usage are very slim. If a heater was left on at that temperature for a week it probably would be ok, but possibly messy, certainly costly.

nophead, 2015-05-15:
Well that is why I suggested 50C as nobody has a room that hot and a hot end at 50C indefinitely would cause no problems at all, you could touch it.

clefranc, 2015-05-15:
When I do a **Preheat ABS All** before 16 seconds after a board reset, I get an **Err: MAXTEMP BED**.

AnHardt, 2015-05-21:
Not watching all heaters with`Preheat_X_All` is addressed by #2134.

AnHardt, 2015-05-21:
@clefranc 
>When I do a Preheat ABS All before 16 seconds after a board reset, I get an Err: MAXTEMP BED.

In witch patch combination, with witch sensors? Can't reproduce it.

clefranc, 2015-05-21:
@AnHardt
> In witch patch combination
It do this with the latests dev.
> with witch sensors?
Both my printer have 2 hotends, 1 bed.
> Immediately or after a time?
Like I said, just after a reset and before 16 seconds mark. After 16 seconds, all is going well.

thinkyhead, 2015-05-22:
@clefranc How weird. The bed doesn't use the 16 second timer.

clefranc, 2015-05-23:
@thinkyhead Is there a reason the bed's thermistor doesn't use the WATCH_TEMP_PERIOD?

It'll will prevent #2066 test 2.

thinkyhead, 2015-05-23:
> Is there a reason the bed's thermistor doesn't use the WATCH_TEMP_PERIOD?

@clefranc So far it just hasn't been implemented. But when it is (maybe I'll try it now) the bed should have its own watch parameters.

AnHardt, 2015-05-23:
Try of an analysis.
1. Not connected bed sensor could be caught by BED_MINTEMP during startup if it would be implemented.
Reconnect could be caught by 'jump'.
MIN and MAX temp are caught in the interrupt. kill() is not perfect. Currently until kill() has shut down the interrupt we have enough calls to killl() to get out of memory, see a reset and may lose communication..
2. Currently can not be caught by the implemented methods. Maybe the modified NopHead algorithm I proposed in  #2066 could catch this. I'm still working on that. Rising-constant-falling seems to work now - but i'm not ready to publish for now.
3. BED_MINTEMP should work. Jump too.
4. Jump will work if temperature has rised enough before disconnecting. Modified NopHed could detect not rising temperature while heating.
5. Jump or Bed_MINTEMP will work.
6. Can be caught by jump.
7. MINTEMP. See 1.
8. THERMAL_PROTECTION or mod. NopHead. Preheat_X_All now checks all heaters (only E3 before). I have no explanation for the resets.
9. MINTEMP.
10. THERMAL_PROTECTION works - but did not call kill() before.
11. MINTEMP
12. THERMAL_PROTECTION and jump will work.
13. MINTEMP
14 Jump will work.

Conclusion:
BED_MINTEMP would be good.
'Jump' catches a lot when implemented. Maybe to much - when adjusted too sensitive.
Calling kill() from interrupt is problematic.
'Modified NopHead' isn't ready now. Maybe I'll get it to work.
25°C sensors are not a good idea - now.

I'm Away for the weekend. More during the next week.

thinkyhead, 2015-05-23:
So, set a "kill" flag, and call it from the main loop instead?

AnHardt, 2015-05-25:
Test 7: is the multiple calls of kill(). Interrupts work before display is initialized. My tests say - it's ok with #2156 #2157
Test 10: Was the not calling kill(). Fixed with #2115.

thinkyhead, 2015-05-26:
Test 2 is still flagged. Do the latest changes (bed mintemp, etc.) help with that one?

clefranc, 2015-05-26:
@AnHardt @thinkyhead Will try to test #2156 #2157 tonight...

AnHardt, 2015-05-27:
@thinkyhead 
Test 2 can not be caught by BED_MINTEMP but by #2164. Will fail then with `HEATING_FAILED`.

clefranc, 2015-05-27:
@AnHardt @thinkyhead See OP for PR #2164 test 8,10,12 & 14.

Almost there, but the test 14 failed because after M109, there is no **Heating failed**.

Almost there!!!

AnHardt, 2015-05-27:
@clefranc 
May i ask about your settings for #2164? Do you have activated MAX_THERMO_JUMP_AMOUNT?
Maybe you downloaded to early to have the "jump-test" at all.
Let's see why test 14 failed.

The thermometer shows 240°C. The heater_state is ether `OFF` or `ON` but probably not `FULL_ON`. The temperature_state is probably `constant`.
The sensor is disconnected. The displayed temperature jumps to 25°C. (should have been detected by 'jump' - downloaded to early or not activated or has an error) The temperature_state switches to `falling`, for a few readings - until the average drops below 25+TEMP_RAW_NOISE - after that, the temperature_state will be and remains to be `constant`. The PID sees a much to low temperature and switches the power to FULL_ON. The heater_state changes to `FULL_ON` and will stay there. The temperature_state (probably `falling`) is stored, the MAX_TEMP_OVERSTOOT_TIME begins to run.
The next few tests will see a `falling` temperature_state - no change - no test.
The average temperature reaches  25+TEMP_RAW_NOISE. The temperature_state changes from `falling` to `constant`. `Constant` is no reason to fail.
MAX_TEMP_OVERSTOOT_TIME is reached somewhen. The test will see a temperature_state of `constant` - no reason to fail. The test is rescheduled. Therefore the heater_state is set to `ON`.
The next test sees a temperature_state of `constant`, will switch heater_state to `FULL_ON`, stores `constant`.
The temperature_state will not change again - so no further test until MAX_TEMP_OVERSTOOT_TIME is reached again. Here the test will see a temperature_state of `constant`. That's no reason for an error. 
Every further retest will have the same result.

Conclusion:
Even this implementation of the (nophead-)test can not catch case 14.
The additional "jump-test" should have caught this.

clefranc, 2015-05-27:
@AnHardt Your're right, I've used the default parameters. Using these catch test 14 with a **Heating failed**:

```
#define MAX_TEMP_OVERSTOOT_TIME 10
#define TEMP_RAW_NOISE 6
#define TEMP_CONSEC_COUNT 6
#define MAX_AMBIENT_TEMPERATURE 50
#define MAX_THERMO_JUMP_AMOUNT 30
#define HEATER_STATE_DEBUG
```
I wrongly assumed I should use the default protection for the test. Sorry for the confusion.

I'll redo all test tomorrow. Is my values above correct?

AnHardt, 2015-05-27:
Strategy to tune (set up) the new test.

1.) Turn off THERMAL_PROTECTION_HOTENDS and THERMAL_PROTECTION_BED in Configuration.h.
We don't want to see what they may detect - we want to se what the new function does.

2.) In Configuration_adv.h turn on the new functions by setting MAX_TEMP_OVERSTOOT_TIME to value different from 0.
The correct value you can find by setting a high target value to the heater but turning it off when MAX_AMBIENT_TEMPERATURE is reached.
The time between turning off and changing direction of the measured temperature is MAX_TEMP_OVERSTOOT_TIME. A smaller value is saver but may give false positives.

3.) Turn on HEATER_STATE_DEBUG to see what's going on.

4.) Don't heat any heater but observe the output of HEATER_STATE_DEBUG on the serial console.

    ...
    echo:SD card ok
    Temperature: 0 count: 127 Temp.: 984 Temp.A.:0 Heater: 0 T_State: 0 Time:4294965691 H_State:1
    Temperature: 1 count: 127 Temp.: 108 Temp.A.:0 Heater: 1 T_State: 0 Time:4294965687 H_State:1
    Temperature: 0 count: 0 Temp.: 984 Temp.A.:984 Heater: 0 T_State: 0 Time:29916 H_State:2
    Temperature: 1 count: 0 Temp.: 108 Temp.A.:108 Heater: 1 T_State: 0 Time:29917 H_State:2
    ...
    Temperature: 1 count: 0 Temp.: 99 Temp.A.:107 Heater: 1 T_State: 0 Time:27630 H_State:2
    Temperature: 0 count: 0 Temp.: 983 Temp.A.:983 Heater: 0 T_State: 0 Time:27433 H_State:2
    Temperature: 1 count: -1 Temp.: 99 Temp.A.:107 Heater: 1 T_State: 0 Time:27434 H_State:2
    Temperature: 0 count: 0 Temp.: 984 Temp.A.:983 Heater: 0 T_State: 0 Time:27236 H_State:2
    Temperature: 1 count: -2 Temp.: 108 Temp.A.:106 Heater: 1 T_State: 0 Time:27237 H_State:2
    Temperature: 0 count: 0 Temp.: 983 Temp.A.:984 Heater: 0 T_State: 0 Time:27039 H_State:2
    Temperature: 1 count: 0 Temp.: 108 Temp.A.:106 Heater: 1 T_State: 0 Time:27040 H_State:2
    ...
    
Temperature: 0  -> bed
Temperature: 1  -> heater 0
count: 127 -> not initialized
count: 0 -> initialized
Temp.: 984 -> readout of a normal thermistor at room temperature
Temp.: 108 -> readout of a MAX 6675 at room temperature
Temp.A.:0 -> Average temperature not initialized
Temp.A.:984 -> Average temperature readout for the last 16 readouts - thermistor
Temp.A.:108 -> Average temperature readout for the last 16 readouts - MAX6675

Now some noise:

    Temperature: 1 count: -1 Temp.: 99 Temp.A.:107 -> at this moment Temp. (99) was smaller than Temp.A.-TEMP_RAW_NOISE (106-6) so the counter is decreased.
    Temperature: 1 count: -2 Temp.: 108 Temp.A.:106 -> the temperatures are close together again - the counter will be reset.
    Temperature: 1 count: 0 Temp.: 108 Temp.A.:106 -> ok again

5.) Find a value for TEMP_RAW_NOISE where the count stays between -2 and 2, than add a bit.
Now try to produce noise by switching on/off lights, fans, play with the display, machines close by, maybe an other heater. 
Count should not grow above +-2.
(For a 25° thermistor test above 25° - else you'll have no noise at all.)

6.) Set TEMP_CONSEC_COUNT to a value the count NEVER can reach by noise. If set to low, you may get false positives.

7.) For MAX_AMBIENT_TEMPERATURE 50° is a good value. A extruder heater may not fall below this, placed above a working heated bed.

8.) To adjust MAX_THERMO_JUMP_AMOUNT take a look on your `Temp.:` readings at room temperature. Take the difference to 1024 and to 0. The smaller difference is the maximum jump the value can make in this direction. To detect an error in worse conditions make MAX_THERMO_JUMP_AMOUNT the half of the smaller difference. The value must be higher than TEMP_RAW_NOISE. If the difference/2 is smaller than TEMP_RAW_NOISE take TEMP_RAW_NOISE+2. The fastest "normal" change is about full_range in one minute with about 5 readings per second (1024/60/5 = 3). Due to non-linearitys of the thermistor table the change my be a bit faster.

9.) Tests
Extruders and the bed are handled the same for this tests. So the results should be the same.
In various states,
 * Disconnect the sensor,
 * Short the sensor,
 * Disconnect the heater or the main power supply.
 
10.) Disable HEATER_STATE_DEBUG
11.) Reenable THERMAL_PROTECTION_HOTENDS and THERMAL_PROTECTION_BED in Configuration.h.

clefranc, 2015-05-30:
@AnHardt Here my parameters:
```
#define MAX_TEMP_OVERSTOOT_TIME 10
#define TEMP_RAW_NOISE 6
#define TEMP_CONSEC_COUNT 6
#define MAX_AMBIENT_TEMPERATURE 50
#define MAX_THERMO_JUMP_AMOUNT 30
#define HEATER_STATE_DEBUG
```
Doing regular heating, I can't get it to work without error:

* When heating the nozzle to target temperature 240°C, I get **Err:TERM. JUMP** at around 150°C, then many more before reaching target. Sometimes getting **THERMAL RUNAWAY** right after a reset of the board.Even got a **Heating failed**.
* When cooling the nozzle from 240°C, I get **THERMAL RUNAWAY** at around 210°C, then more before reaching room temprature.

AnHardt, 2015-05-31:
@clefranc 
Thank you for the tests.
Identified 2 stupid errors. Update will follow.

clefranc, 2015-06-03:
@AnHardt See result for https://github.com/MarlinFirmware/Marlin/pull/2231.

Look at test 2 & 8.

```
#define MAX_TEMP_OVERSHOOT_TIME 16
#define TEMP_RAW_NOISE 7
#define TEMP_CONSEC_COUNT 7
#define MAX_AMBIENT_TEMPERATURE 50
#define MAX_THERMO_JUMP_AMOUNT 30
#define HEATER_STATE_DEBUG
```

thinkyhead, 2015-06-10:
@clefranc @AnHardt Just looking at __TEST 2__. The bed temperature should be seen to rise, but it does not rise. Is that condition supposed to be caught in #2231? What needs fixing to catch it?

[11 more replies]

closed 2016-04-23 by thinkyhead