MarlinFirmware/Marlin #2068: PID tuning has a hard time with slow heaters

thinkyhead, 2015-05-13:
Tom Sanladerer @imitation points out issues with PID in cases of badly-designed hot ends. Is it worth tweaking this code to improve results in such situations?

https://www.youtube.com/watch?v=icLKTOCH9VQ&t=270

Wurstnase, 2015-05-13:
For such slow heaters we could implement something like this
https://github.com/Wurstnase/Marlin4Due/blob/0ba35412ff6e3e961e3a9c4f0157f046dfdd7ce1/Marlin/temperature.cpp#L493
Well, i need some better variable names and some more comments in this code.

ntoff, 2015-05-13:
I think people building the hot ends should build them to comply with how the firmware works, not the other way around. If your heater and thermistor are on opposite sides of a non thermally conductive material then I don't think it's up to the firmware to attempt to rectify a major design flaw. It should be up to the designers to fully test their hardware against existing solutions and if they don't work, they change the design. Just look at every other hot end out there, nice thermally conductive heater block and thermistor nice and close to the heater. Why cater for back yard clowns with bad designs?

thinkyhead, 2015-05-13:
@ntoff I am inclined to agree!

AnHardt, 2015-05-13:
Hard coded values are always a bad thing.
On the other hand - such, supposed to be, rarely used values are polluting the configurations.
How about a nice, well named #define in the file the value is used?

20 degree is a reasonable value for a well designed hotend.
Is this really needed with a well working MAXTEMP - besides autotuning may than potentially can last close to ever?

ntoff, 2015-05-15:
Sort of on topic I guess, I just got the latest version from github and every time I get "heating failed" where previously I had no issues. My hot end goes up by maybe 1 degree every second, it's not exactly fast but it also has no trouble PID tuning. But when I go to actually print, "heating failed" even though at the time it happens the temperature is still rising. Have there been any significant changes in heating safeguards I should go poke around in?

I would put the E3D hot end on, it heats up nice & fast but hey.. you know.. can't print a mount because "heating failed" :p

clefranc, 2015-05-15:
@ntoff Thermal protection is on by default now, try to set higher values in Configuration_adv.cfg:

'''
/**
 * Thermal Protection parameters
 */
#ifdef THERMAL_PROTECTION_HOTENDS
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius
'''

ntoff, 2015-05-15:
I did see that part and changed it for the next attempt but decided to quickly go back to the old working version of Marlin just to test the PID values and see if maybe the  old version would error out. Unfortunately it didn't so then I had to wait for the print to finish. HA.

thinkyhead, 2015-05-16:
@ntoff @clefranc Actually, the thermal options have now been reduced to 2, `THERMAL_PROTECTION_HOTENDS` and `THERMAL_PROTECTION_BED`, with tunable parameters moved to `Configuration_adv.h`. By default it will watch for thermal runaway after heating, and it will look for slow heating while still below the target.

ntoff, 2015-05-16:
well mine is still going up when it fails https://www.youtube.com/watch?v=kQ6gb_lOuik goes up just as slowly as it does with the previously working firmware.

thinkyhead, 2015-05-16:
Try changing the parameters from 4 degrees in 16 seconds to 3 in 16, or 4 in 25. It looks like the heater is just very slow. We can raise the defaults if this is common.

ntoff, 2015-05-16:
I'll hook up the E3D without actually mounting it and compare how long it takes to heat up. When does the timer reset? Every time the temperature increases? or whenever the command to change the temperature is issued?

thinkyhead, 2015-05-16:
@ntoff If `WATCH_TEMP_PERIOD` is 16 and `WATCH_TEMP_INCREASE` is 4, then it will check the heater every 16 seconds to see if it has gone up 4 degrees since the last check. Once it reaches target, then thermal runaway takes over.

ntoff, 2015-05-16:
does that apply to the bed as well? because the bed is even slower and that never gives me any errors.

Am waiting for the bed to warm up, have attached the E3d and will let you know what happens using default settings.

Admittedly I made the hot end myself but the heater is aluminium and the thermistor is right next to the heater so it shouldn't give too much of a lag between heating and measuring. http://i.imgur.com/kOhSJnu.jpg

I'm not too sure on the heater cartridge though, it's out of a K8200 that ran on 15v and I'm running this on 12v so the heater might just not be all that great, but still I've had no issues before.

thinkyhead, 2015-05-17:
> does that apply to the bed as well?

@ntoff No, there is no checking of the bed as it heats.

> lag between heating and measuring

That's not important to this feature. But it can make PID tuning a challenge.

>  make it "foolproof" for less than perfect heaters?

A rise of 4 degrees in 16 seconds is "good" but not "great" performance. Below that is "poor" and I think it might actually be a good thing that Marlin throws a temperature error for such hot-ends.

ntoff, 2015-05-17:
> I think it might actually be a good thing that Marlin throws a temperature error for such hot-ends.

Perhaps. I guess at least now I know something's wrong with that heater cartridge since the other printer heats up twice as fast. I've never actually timed either of them or sat around watching it. I just hit print and wander off. That's how reliable they (usually) are.

thinkyhead, 2015-05-17:
@ntoff I guess it comes down to whether we should accommodate unpredictable tolerances. In the golden heyday of RepRap, Nichrome wire-wrapped machined bolt hot-ends were all the rage. We can widen the tolerance at least for cheaper hot-ends, I think. Does 3 degrees in 20 seconds seem ok? The length of time is the main thing – it's the minimum time to failure, and the interval to each test, so really, a very good setting might be: 1 degree in 10 seconds

nophead, 2015-05-17:
Its might be better to detect which way it is moving rather than look for a
temperature change over a defined time. That way if it continues to rise
before falling when switched off the amount of time to start falling is
about half of the time to be a lower temperature than the point is switched
off.

thinkyhead, 2015-05-18:
So…
- __Within target range__: No checking
- __Above target range__: Should be turned off, and falling within (and/or after) an interval
- __Heating, below target range__: Needs only to be rising within (and/or after) an interval
- __Not heating__: Needs only to be falling within (and/or after) an interval

nophead, 2015-05-18:
Yes but the problem seems to be some heaters have such crazy overshoot that you have to wait so long for it to be falling the test becomes worthless as during the same period it could become dangerously hot.

ntoff, 2015-05-18:
This is what was on configuration.h

#define DEFAULT_Kp 41.23 	
#define DEFAULT_Ki 2.89	
#define DEFAULT_Kd 146.92	

though I'm not 100% sure if that's what I was using or whether I'd auto tuned and just saved to EEPROM

this is what I use on the Mendel, takes about 2 & a half minutes to reach full temp:
    #define  DEFAULT_Kp 22.2
    #define  DEFAULT_Ki 1.08
    #define  DEFAULT_Kd 114

thinkyhead, 2015-05-18:
@nophead I always thought that PID tuning dealt with overshoot too. Or would that require a whole new kind of tuning? I'm sure it's possible to predict overshoot using some kind of pre-analysis.

AnHardt, 2015-05-18:
@thinkyhead 
Here we deal with a 'badly designed hotend', with a lot of fairly unconductive  material on a long way between resistor and thermistor. (what tom did not cement in as intended - what makes the problem worse). And we talk here about the PID_autotune(), what by purpose produces a overshoot to measure it. The problem here is (if there is one at all), that the designer of the routine did not suppose a heater could ever have such a high overshoot. If we agree that the tested hotend is 'bad designd' than the value is ok. If we think the hotend is 'ok', we have to increase the value. In ether case 'naming' the value would be a good thing. #2111

In #2092 i produced a long lasting overshoot by purpose, carefully taking care the extruder will cool down slowly, and avoiding PID to produce a high overshoot, to stress the test. Under normal conditions, with well tuned PID, the same tested hotend will basically not react different from @nophead 's.

So yes. PID deals with overshoot and works well.

thinkyhead, 2015-05-18:
> what tom did not cement in as intended - what makes the problem worse

An important point!

ntoff, 2015-05-21:
um, ok. So I just got the latest version but kept my configs from the version that worked with the e3d but NOT with my own hot end and now even with the e3d I'm getting heating failed all the while the temperature is shooting up like a rocket so it's definitely not failing to heat.

thinkyhead, 2015-05-21:
@ntoff Did you run into any sort of pitfall that we should warn users about when they upgrade?

ntoff, 2015-05-22:
No, I just messed up a setting without realizing when I brought the github Marlin over to the copy that I mess with to upload. I replaced a file I shouldn't have.

boelle, 2015-06-01:
i agree on that we should not start to support bad setups....

closed 2015-06-01 by AnHardt