MarlinFirmware/Marlin #2071: X-Endstop/Homing not working

gegiarmo, 2015-05-13:
I'm trying to setup my new Azteeg X3 Pro, printer was previously working well on RAMPS 1.4. As a note, it's a custom printer, but works as a standard Cartesian. I've downloaded the release version of Marlin, updated feedrates, endstop inversion, etc. as necessary.

The issue I'm running into is that it has weird X-endstop behavior. Y-axis is set up exactly the same and behaves as it should. When the X-endstop is not inverted, homing only makes the axis progress a bit, similar to what @emartinez167 ran into with Issue #1725. When the endstop is inverted, it completely ignores it, whether it is triggered or not.

Here is a pastebin of the configuration.h: [here](http://pastebin.com/4bB56pU1).

thinkyhead, 2015-05-14:
@gegiarmo There have been enough changes in the configuration files that you probably need to doa refresh. If you have some tool that can do a diff between your `Configuration.h` and `Configuration_adv.h` files and the latest versions, that will help you to update them to be compliant with the latest code. Or, just start with the fresh files and then apply your customizations. After that, try again, and if the issue still exists we can look further.

gegiarmo, 2015-05-14:
Started fresh with the latest development build. Still having the same behavior.

thinkyhead, 2015-05-15:
@gegiarmo Do you see the same behavior with `ENDSTOPS_ONLY_FOR_HOMING` enabled and also disabled? (Trying to see if, or how, that setting might be involved.)

gegiarmo, 2015-05-15:
Yes, same behavior with ENDSTOPS_ONLY_FOR_HOMING both enabled and disabled.

gegiarmo, 2015-05-15:
So, fun story. Per the Azteeg X3 Pro pin definitions [here](http://files.panucatt.com/datasheets/x3pro_pins.pdf), the X-min is pin 2 and max is pin 3, the opposite of RAMPS and as defined in Marlin. So this is what I set it to, because I thought Marlin was wrong. The punch line: The Azteeg documentation is wrong, pin 3 is correct for X-min. It works now. Thanks for the input though @thinkyhead .

closed 2015-05-15 by gegiarmo

thinkyhead, 2015-05-15:
> Panucatt, I see that the PDF you have posted at http://files.panucatt.com/datasheets/x3pro_pins.pdf has an error, which is causing some users to become confused. The PDF shows X MIN ENDSTOP as Pin 2 and X MAX ENDSTOP as Pin 3. This is incorrect, and is not aligned with RAMPS. The pin numbers should be reversed, with X MIN ENDSTOP as Pin 3 and X MAX ENDSTOP as Pin 2. Please correct this and post a new PDF.

gegiarmo, 2015-05-15:
Yup, I already did the same, hopefully they do update it.

dvseliteus, 2016-07-26:
I am having the same exact issue with my Azteeg X3 and the X-Axis End stop not working.
I have changes the pin assignment on the pin.h tab in the Marlin firmware and when I press the limit switch nothing happens?
#if MOTHERBOARD == 99
#define KNOWN_BOARD 1

#define X_STEP_PIN          3
#define X_DIR_PIN           2
#define X_ENABLE_PIN        -1
#define X_STOP_PIN          16
I use simplify3d and when I hit the home button it triggers X to zero out even though it has not hit the switch?
Anyone have an idea what's wrong here?

thinkyhead, 2016-07-27:
@dvseliteus What version of Marlin are you using?

dvseliteus, 2016-07-27:
Hi @thinkyhead thank you for the quick reply.
I'm using Marlin-1.0.2 here are my [pin.h](http://pastebin.com/CGYfbg7w) and [configuration.h](http://pastebin.com/suMwxPD5) files.
I reached out to panucatt for support but have not gotten any reply. I'm stuck at this point.
Any help would be greatly appreciated. Thank you!

thinkyhead, 2016-07-27:
@dvseliteus

With the line `#if MOTHERBOARD == 99` it looks like an older version of Marlin. I recommend trying 1.0.2-1 or 1.1.0-RC7 and see if it works better for you.

The Azteeg X3 has most pins identical to RAMPS 1.4 (Extruder, Fan, Bed). The pins file with 1.0.2-1 seems correct. Anyway, here are the proper pins for the ones you mentioned above:
```cpp
#define X_STEP_PIN         54
#define X_DIR_PIN          55
#define X_ENABLE_PIN       38
#define X_MIN_PIN           3
#define X_MAX_PIN         2
```

dvseliteus, 2016-07-28:
Thanks again for your help @thinkyhead 
I have the motherboard set as 67?
I have entered the pin numbners under the #67 pins and I am still having the same issue.
I am hesitant to try 1.1.0-RC7 as I will have to re-program everything and I don't really know what I am doing. Of all the things I thought would get me stumped I can;t believe it's something like this!

dvseliteus, 2016-07-28:
Here is what panucatt has provided for support;
"it is possible that the endstop pin on the Atmel chip is not working anymore. It can be caused by static getting into the pin from the endstop wiring, Ive seen a few of these happen on X3s and X3 Pros. What you can do is to use the MAX endstop pins for the X endstop by defining the X endstop using the pin of the MAX endstop. Say if you will be using the Xmax as your Xmin endstop you can use D2 instead of D3 which is what the Xmin uses. Actually you can use any free pin and just make sure its pulled up in firmware."

Now here is where I get confused. Where in the firmware exactly am I making these changes in correspondence with the pins on the board? I just can't seem to wrap my head around this. Can someone please dumb it down for me?
[http://pastebin.com/CGYfbg7w](url) [http://pastebin.com/suMwxPD5](url)

I have tried all 3 pins for the X axis on the board it must be something in the firmware. I'd like to stick to Marlin 1.0.2-1 I tried working with 1.1.0-RC7 and I'm getting weird compile errors like "Error compiling for board Arduino/Genuino Mega or Mega 2560." I feel like I'm just going in endless circles of frustration.

thinkyhead, 2016-07-29:
> I am hesitant to try 1.1.0-RC7 as I will have to re-program everything

You'll be better off using `RCBugFix` or `1.1.0-RC7` as the newer Marlin places the pins for each board in its own file. Anyway, to switch the endstop pin for your board in `1.0.2-1` all you need to do is change [lines 599-600 in `pins.h`](https://github.com/MarlinFirmware/Marlin/blob/1.0.x/Marlin/pins.h#L599-L600):
```diff
-   #define X_MIN_PIN           3
-   #define X_MAX_PIN           2
+   #define X_MIN_PIN           2
+   #define X_MAX_PIN           3

```

dvseliteus, 2016-07-29:
Thanks again @thinkyhead I have 1.1.0-RC7 all setup but I am getting a strange "Error compiling for board Arduino/Genuino Mega or Mega 2560." and I'm not able to upload to the board.

I'll try the fix above on 1.0.2-1 and see if I can at least get the x-stop triggering and then focus back to the newer version.

thinkyhead, 2016-07-29:
@dvseliteus Which version of Arduino IDE are you using?

dvseliteus, 2016-07-29:
@thinkyhead arduino-1.6.10, I need to use 1.6.0 or later (earlier) don't I...

thinkyhead, 2016-07-29:
Version 1.6.10 seems to be acting up. Try 1.6.9 or 1.6.8.

dvseliteus, 2016-07-29:
Ok you got me working with the 1.1.0-RC7 version and I quite like the format over 1.0.2-1. I have tried Arduino version 1.6.10, 1.6.0, installed U8glib library, 1.6.8, and finally 1.6.9.
I keep getting a compile error; 
SanityCheck.cpp:554: error: #error "HEATER_1_PIN not defined for this board."
     #error "HEATER_1_PIN not defined for this board."
I looked at your thread https://github.com/MarlinFirmware/Marlin/issues/1741 but couldn't figure it out.
[http://pastebin.com/BMk2qkff](url) Line 554

Blue-Marlin, 2016-07-29:
Please append the Configuration.h causing the error.

dvseliteus, 2016-07-29:
@Blue-Marlin http://pastebin.com/U2ffANSJ here you go :+1:

Blue-Marlin, 2016-07-29:
pins_AZTEEG_X3.h was not prepared for two extruders. Please try the folowing patch.
Feedback please.
```diff
@@ -24,11 +24,18 @@
  * AZTEEG_X3 Arduino Mega with RAMPS v1.4 pin assignments
  */
 
 #define BOARD_NAME "Azteeg X3"
 
-#include "pins_RAMPS_14_EFB.h"
+#if EXTRUDERS == 1
+  #define IS_RAMPS_EFB
+#endif
+#if EXTRUDERS == 2
+  #define IS_RAMPS_EEB
+#endif
+
+#include "pins_RAMPS_14.h"
 
 #if ENABLED(VIKI2) || ENABLED(miniVIKI)
 
   #undef DOGLCD_A0
   #undef DOGLCD_CS
```

thinkyhead, 2016-07-29:
Is 2 heaters the maximum for Azteeg X3? If so, I'll add a check in its pins file for >= 3.

Blue-Marlin, 2016-07-29:
Yes.
http://reprap.org/mediawiki/images/a/a5/X3_wiring_diagram.jpg

thinkyhead, 2016-07-29:
Feel free to just post a link to http://LMGTFY.com for these kinds of questions. 😜

Blue-Marlin, 2016-07-29:
All RAMPS combinations could work. But because of the extra low power mosfets there are enough extra connectors for fans. So these two make most sense.

dvseliteus, 2016-08-01:
@Blue-Marlin thank you for pointing this out.
I went ahead and added the code to the pins_AZTEEG_X3.h tab but I am receiving the following error;
[http://pastebin.com/4JY1a6KY](url)
Here is my revised pins_AZTEEG_X3.h file: http://pastebin.com/pgHi0czA
Feels like we are getting close here, thanks for the support all!

Blue-Marlin, 2016-08-01:
@dvseliteus 
Applying a diff means removing the lines beginning with '-' and adding that ones, beginning with '+' - without the '+'.
Don't worry. The changes are already in RCBugFix.

dvseliteus, 2016-08-01:
@Blue-Marlin 
Thanks, I have downloaded the RCBugFix and tried my best to make the modifications needed to run the Azteeg X3 and Viki LCD (I'm a begginer)
I am running into an error: #error "HEATER_1_PIN not defined for this board."
I tired to implement some fixes from this thread: [https://github.com/MarlinFirmware/Marlin/issues/3497](url) but was unsuccessful.
Here is a download link to my Marlin Folder: [image]<br /><div style=font-size:9px;font-family:Arial, Helvetica, sans-serif;width:127px;font-color:#44a854;> <a href=http://www.filedropper.com >upload files free</a></div>
I'm not sure how to properly set the Z_MIN_PROBE correctly?

thinkyhead, 2016-08-02:
@dvseliteus I just submitted a patch to `RCBugFix` that should fix your `HEATER_1_PIN` error. It adds code to assign the power outputs to different uses for RAMPS derivative boards, which were not being auto-selected previously. On RAMPS the pin labeled D10 is for the first extruder, D9 is for a fan or a second extruder, and D8 is used for the heated bed or a second fan. The same order will apply on your Azteeg X3, although the plug labels may differ.

dvseliteus, 2016-08-02:
Thanks @thinkyhead  I will continue to keep at it and see if I can get the code to eventually compile.
I am able to get the firmware to compile with 1 Extruder.
2 Extruders I am getting #error "HEATER_1_PIN not defined for this board." no matter what I configure the pins_RAMPS.h pins. (Waiting for submitted a patch to RCBugFix that should fix your HEATER_1_PIN error.)

When I compile with 1 extruder I am getting a weird error with the #define LCD_I2C_VIKI Panucatt VIKI LCD with status LEDs
'DIOBEEPER_PIN_RPORT' was not declared in this scope

I am going to buy a Ramps 1.4 Kit and also try hooking that up. That compiles with 2 extruders and #define ULTRA_LCD with no issues. I still want to use the Azteeg but need to push for a quick temp solution.

@thinkyhead @Blue-Marlin Thank you for your great support and troubleshooting. Hopefully we can get this fixed and post it for others if they want to use this hardware.

thinkyhead, 2016-08-02:
> When I compile with 1 extruder I am getting a weird error with the #define LCD_I2C_VIKI Panucatt VIKI LCD with status LEDs 'DIOBEEPER_PIN_RPORT' was not declared in this scope

#4506 should fix this.

dvseliteus, 2016-08-02:
[#4506](https://github.com/MarlinFirmware/Marlin/pull/4506) did the trick! Thanks @thinkyhead

dvseliteus, 2016-08-02:
Viki menu does not work when button is pressed and the X Y and Z fields blink question marks?

thinkyhead, 2016-08-02:
Blinking question marks is normal behavior. When you home an axis, that axis will stop blinking.

dvseliteus, 2016-08-02:
Ok great that makes sense. I still have the original issue that first brought me to this forum X-Endstop/Homing not working #2071 with the RCBugFix. I will have to try to see what's going on here again as I cannot home X properly.

Blue-Marlin, 2016-08-02:
Blinking
'X' <-> '?' means axis not homed.
'X' <-> ' ' means stepper has been shut down since last homing. If you don't like it turn it off via `#define DISABLE_REDUCED_ACCURACY_WARNING`

thinkyhead, 2016-08-02:
I believe the PR #4506 should fix the encoder button issue. It will be merged shortly.

dvseliteus, 2016-08-02:
@Blue-Marlin thank you for the info I quite like it actually now that I know what it means. Thanks @thinkyhead I got the beeper working but not when I press the button it shows infoscreen text on the first line for a half second and then reverts back to the main screen display.

thinkyhead, 2016-08-02:
Test after #4506 is merged and see if it helps.

Blue-Marlin, 2016-08-02:
Basic harware endstop setup:
* Send M119 in mid air and  observe the output.
* No endstop shall be triggered.
* If one shows as triggered try to switch it by hand. If it alters the state revert it (*_ENDSTOP_INVERTING true)
* If it does not trigger, alter the related ENDSTOPPULLUP and retest.
* disable unused endstops (//#define USE_XMAX_PLUG)

Probe:
If you have a unused z-min-pin, use that.
If you have a z-min-endstop in use, define 'Z_MIN_PROBE_PIN' with the number of an unused endstop. Also define `Z_MIN_PROBE_ENDSTOP` and undefine `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`.

Always retest with M119 if all makes sense now.

dvseliteus, 2016-08-03:
@thinkyhead I am having the same result after applying the updated code to the 6 tabs on #4506 fix.
Here is what it looks like when I try to access the menu: [](https://youtu.be/mE4ppKlIf-U)
@Blue-Marlin I am using simplify 3D to send an M119 command and nothing happens, I am not sure how to proceed next?
Thanks again for sticking with me on these issues.

Blue-Marlin, 2016-08-03:
@dvseliteus 
Don't use non free software.
I'm quite sure there is an option to make Marlins output visible in the log.

dvseliteus, 2016-08-03:
@Blue-Marlin I have purchased Simplfy 3D and use it with my prusa printer.
I can copy a log file from the software if you like or I can test with a software you recommend?

Blue-Marlin, 2016-08-03:
There is an option to make Marlins output visible. I suppose that is off by default.

The output should look like 
```
23:55:44.754 : N5129 M119*21
23:55:44.759 : Reporting endstop status
23:55:44.759 : x_min: open
23:55:44.759 : y_min: open
23:55:44.759 : z_min: open
23:55:44.759 : ok
```
> Don't use non free software.

That's more a political statement. Regardless if you paid for it or not.

thinkyhead, 2016-08-03:
> Don't use non free software

(By "non free" he doesn't mean not costing $¢£ money. He means "non free" in the sense of not coming with the freedoms given to you by the GPL, such as having the source code.)

thinkyhead, 2016-08-03:
> Here is what it looks like when I try to access the menu

Very strange. It's as though the button is showing up as pressed when you press it. And then it's showing up again as pressed when you release it. What happens if you simply press-and-hold the button? Does it act as though it is being pressed repeatedly?

dvseliteus, 2016-08-03:
@thinkyhead Yes when I  press-and-hold the button it just acts as it's pressed re-repeatedly. I'm starting to think as a beginner I picked the most difficult parts and their hardware versions to make work.
@thinkyhead @Blue-Marlin Thanks for that info I now understand it was in reference to an open source software. I am still unsure how to implement blue-marlins instructions.

thinkyhead, 2016-08-03:
@dvseliteus Do you have the "`SPEAKER`" setting enabled? I'm wondering if the "debounce" that we use for the controller button is working properly with `SPEAKER` both enabled and disabled.

dvseliteus, 2016-08-03:
@thinkyhead 
        if (this->state.tone.frequency > 0) {
          #if ENABLED(SPEAKER)
            ::tone(BEEPER_PIN, this->state.tone.frequency, this->state.tone.duration);
          #else
            this->on();
          #endif
Is there where we see if it is enabled on the buzzer.h file?

thinkyhead, 2016-08-03:
In `Configuration.h` you will see:
```cpp
//
// SPEAKER/BUZZER
//
// If you have a speaker that can produce tones, enable it here.
// By default Marlin assumes you have a buzzer with a fixed frequency.
//
//#define SPEAKER
```

dvseliteus, 2016-08-03:
Speaker appears to be disabled:
// SPEAKER/BUZZER
//
// If you have a speaker that can produce tones, enable it here.
// By default Marlin assumes you have a buzzer with a fixed frequency.
//
//#define SPEAKER

thinkyhead, 2016-08-03:
Fabulous. Try enabling it to see if it makes any difference. Then we will take a look at the `lcd_quick_feedback` function to make sure it has an adequate delay after playing the tone.

dvseliteus, 2016-08-03:
Okay
//
// SPEAKER/BUZZER
//
// If you have a speaker that can produce tones, enable it here.
// By default Marlin assumes you have a buzzer with a fixed frequency.
//
#define SPEAKER
Yielded the same issue with the menu button

thinkyhead, 2016-08-04:
@dvseliteus On most controllers the controller button triggers once, then we have a delay and some logic to "debounce" the button, so it won't trigger again right away. Possibly the current method isn't reliable for your Viki. So, I will see about making a more robust button handler for you to test soon.

dvseliteus, 2016-08-04:
Thank you @thinkyhead the Viki worked great in the older version of Marlin 1.0.2-1

Marlin 1.0.2-1 - X-Axis had weird X-endstop behavior but viki worked
Marlin RCBugFix - Viki button issue and same weird X-endstop behavior (won't trigger)

thinkyhead, 2016-08-04:
Have you updated your configuration since http://pastebin.com/U2ffANSJ? I'll have a look at that now.

thinkyhead, 2016-08-04:
@dvseliteus Use the Pronterface app instead of Simplify3D. Use `M119` from the console to test the endstops and see how they are behaving.

dvseliteus, 2016-08-04:
@thinkyhead that last pastebin was of 1.1.0-RC7 version.
Here is the current version http://pastebin.com/rTb5e6YT RCBugFix

thinkyhead, 2016-08-05:
Your configuration seems fine. How does the output from `M119` look?

[23 more replies]