MarlinFirmware/Marlin #2080: SD Card use FAT32

rdacomp, 2015-05-15:
Use FAT32 when using the SD Card. If there are difficulties I can help.

ntoff, 2015-05-15:
What do you mean? I'm already using FAT32 on my SD card

rdacomp, 2015-05-15:
The fact that it is not working.

rdacomp, 2015-05-15:
The fact that it is not working.
RAMPS 1.4 no sd card format fat32!!!

thinkyhead, 2015-05-15:
@rdacomp http://www.marlinfirmware.org/index.php/Reporting_Bugs

ntoff, 2015-05-15:
I have 2 RAMPS 1.4 boards, one with a full graphics LCD and one with just a regular 4 line LCD, both with SD readers and both read FAT32 just fine.

Are you using an SD reader built into a LCD display or an addon SD reader module?

thinkyhead, 2015-05-15:
@rdacomp The output of `M20` can be misleading. There is some rudimentary long filename support, so for example, when you use an LCD controller you will see long filenames listed on the screen. However, you are correct that the full interface to the long filenames has not been implemented for the G-Code to select an SD file, `M23`. And hosts also do not have access to long filenames through the firmware. I would like to see that fixed also.

rdacomp, 2015-05-15:
I use kit sainsmart.
http://www.sainsmart.com/mega2560-smart-lcd-2004-controller-a4988-ramps-1-4-3d-printer-kit-for-reprap.html
Later configuration.h and attach files that you have requested.
I have tried several types of cards but no one sees it.
Run and look SanDisk absurd. My proposal to solve all problems in software.

thinkyhead, 2015-05-15:
Did you read this? http://forums.reprap.org/read.php?151,255472

rdacomp, 2015-05-15:
And not only that. I've already the entire Internet rummaged.
Many people use USB and simply do not use the SD Card!
Therefore, the problem is still not solved.

ntoff, 2015-05-15:
@thinkyhead That looks like it's more to do with the pins just being wrong for a specific motherboard (the megatronics) but rdacomp said he bought a kit containing a RAMPS board which has no SD card so as long as he's using the correct (and by now default) ramps_13_efb pinout then it *should* "just work" shouldn't it?

The issues I see on that forum look like they come from the pins just being wrong. Unless I'm missing something.

thinkyhead, 2015-05-15:
Yes I see it [also mentioned here](http://www.fabric8r.com/forums/showthread.php?1510-So-I-just-received-my-SainSmart-LCD-Smart-controller-12864) trying to get it to work with a Printrbot board.

> the smart controller's onboard SD slot would not work without at least one connection to the motherboard's EXP1 header…

I use the very same controller with RAMPS and it performs well. I recommend using my case for it too.

rdacomp, 2015-05-15:
But I also expect that ramps_13_efb 33 should still work. But it is not. I checked all the connections and contacts. All OK. But the SD Card is not working.

ntoff, 2015-05-15:
I already have the Velleman equivalent on the frankenprinter with box and a full graphics controller on the mendel90 (embedded into the frame)

rdacomp, 2015-05-15:
I try this option
http://www.fabric8r.com/forums/showthread.php?1510-So-I-just-received-my-SainSmart-LCD-Smart-controller-12864/page2
Maybe you want to disable JTAG?

ntoff, 2015-05-15:
There doesn't seem to be an easy to find schematic of the LCD board, what you could do is trace the pins that go from the SD card, figure out their function from the SD card spec sheet and make sure they go to the right pins on the RAMPS board. I had to do this when I converted my 3drag / k8200 controller over to work on a RAMPS board.

AnHardt, 2015-05-15:
schematic is here: http://reprap.org/wiki/RepRapDiscount_Smart_Controller

Had problems with the provided cables before. Check the cables by swapping them. If then the display fails, you know why.

rdacomp, 2015-05-15:
Сhanged.

AnHardt, 2015-05-15:
@rdacomp 
Result?

rdacomp, 2015-05-15:
LCD operates. SD Card is not!

rdacomp, 2015-05-15:
pins.h
Line 1474 change pin from 26 to 40:
#define SDSS 40 //QU-BD_Ups SD Chip Select

sd2card.cpp
Line 303 add new line:
pinMode(PB_SS_PIN, OUTPUT);

Lines 305-308 comment out "IF" lines, change SS_PIN to low and set PB_SS_PIN to high:
//#if SET_SPI_SS_HIGH
digitalWrite(SS_PIN, LOW);
digitalWrite(PB_SS_PIN, HIGH);
//#endif // SET_SPI_SS_HIGH

sd2pinmap.h
LINE 225-226 change 20 to 40 and add new line:
uint8_t const SS_PIN = 40; // F2
uint8_t const PB_SS_PIN = 26; // B6

AnHardt, 2015-05-15:
@rdacomp 
pins.h Line 1474 is somewhere in the ulimaker part when using Marlin 1.0.2. This is for sure irrelevant.

In later Marlins there is no Line 1474 at all.

Please update.

thinkyhead, 2015-05-16:
@rdacomp We only support the latest Development code. Are you describing how you got it working with Marlin 1.0.2?

thinkyhead, 2015-05-16:
Ah, I see, you got it here http://www.fabric8r.com/forums/showthread.php?1510-So-I-just-received-my-SainSmart-LCD-Smart-controller-12864&p=19292&viewfull=1#post19292

boelle, 2015-06-01:
closing this one as several people are using fat 32 without any issue... myself included

closed 2015-06-01 by boelle